espressif/arduino-esp32 #6126: UART Loopback doesn't work on 2.0.2

SuGlider, 2022-01-11:
### Board

ESP32, ESP32-S2 and ESP32-C3

### Device Description

DevKit

### Hardware Configuration

ESP32 - loopback on UART2 : TX2 <--> RX2
ESP32-S2 and ESP32-C3 - loopback on UART1 : TX1 <--> RX1

### Version

latest master

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

When trying to send data in a UART loopback configuration, it fails and nothing is read.
Using Arduino Core 2.0.0 and 2.0.1 - it works fine. The issue is with v2.0.2
Most problably something related to last change on support to MODBUS.

### Sketch

```cpp
HardwareSerial LOOPBACK_UART(2);  // UART2 on ESP32 and UART1 on S2/C3

void setup() {
  Serial.begin(115200);
  LOOPBACK_UART.begin(115200);

  Serial.println("Serial Port 0 - Testing");
}

void loop() {
  static uint8_t count = 0;

  //Shorted Pins TX<->RX using ESP32/UART2 or S2/C3/UART1
  LOOPBACK_UART.write('0' + count);
  if (LOOPBACK_UART.available()) {
    Serial.write(LOOPBACK_UART.read());
  }
  count++;
  if ('0' + count > 'Z') {
    count = 0;
    LOOPBACK_UART.println();
  }
  delay(100);
}
```

### Debug Message

```plain
Correct Output should be:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1420
ho 0 tail 12 room 4
load:0x40078000,len:13540
load:0x40080400,len:3604
entry 0x400805f0
Serial Port 0 - Testing
0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ
0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ
0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ
0123456789:;<=>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ
```

but with v2.0.2 there is no output after the message  Port 0 - Testing```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

chegewara, 2022-01-13:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-uart.c#L157

SuGlider, 2022-01-13:
> https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-uart.c#L157

Yes, This was removed in #6133.
This change, along, with MODBUS issue, is solved also in #6133

chegewara, 2022-01-13:
Yes, i found the PR right after posting the link.

closed 2022-01-17 by me-no-dev

collin80, 2022-01-25:
I'd like to also complain about hitting this issue. Manually applying #6133 does fix it. My problem was a Tesla Model S module sketch. The modules speak serial at 612.5k and you send on TX to the bank of modules which send from one to the other then loop all the way back to RX. I would not call this loopback. Loopback tends to suggest that TX is directly connected to RX by a wire with nothing in between or the hardware is directly doing this. No, this is more like token ring where the signal flows from device to device and ends up back where it started. But, really, the issue is one of full duplex. No matter what anyone says, the offending code in V2.02 most certainly puts the hardware into half duplex and this doesn't work if you have both wires active at once whether that's because the signal from TX is looping back or whether it is because you otherwise have full duplex for some other reason. So, the issue is fixed but that doesn't do normal users any good since there has not been another release. So, hopefully my verbose complaint helps anyone else searching for answers.

SuGlider, 2022-01-25:
Thank you for the comment.

We will release 2.0.3 in a week or two and it will have the fix #6133 and others more new features.

Please stay tuned!
Thanks!

PR #6133 (2022-01-17): Fixes UART MODBUS and Loopback issue
## Summary
This PR fixes an issue created by #6026 not allowing UART loopback to work correctly.
It fixes both at the same time:
- makes flush() to wait up to the end of UART transmission before returning, fixing issue related to MODBUS #5877 
- solves the UART loopback issue in https://github.com/espressif/arduino-esp32/pull/5549#issuecomment-1009642889 and #6126

## Impact
None.

## Related links
Fixes #6126 
https://github.com/espressif/arduino-esp32/pull/5549#issuecomment-1009642889
#5877 
#4603