espressif/arduino-esp32 #6127: GPIO-17 outputs some 40 MHz signal since v2.0.x

TD-er, 2022-01-11:
### Board

ESP32-WROOM-UE 16MB

### Device Description

Some boards I made using original Espressif ESP32-WROOM-UE 16MB modules.
These boards can be equipped with LAN8720A, but this is optional.
Powered via stable 5V, but this also occurs when powered via USB programmer.

### Hardware Configuration

This happens regardless whether Ethernet is used (or even support for it included in the build).
Previous SDK was not showing this behavior, but builds using IDF4.4 show this behavior.

Both tested with PSRAM support enabled and disabled. (`BOARD_HAS_PSRAM` define)
N.B. the WROOM does not have PSRAM present.

### Version

other

### IDE Name

PlatformIO

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Using the latest code, using IDF4.4  espressif/arduino-esp32 2.0.2
Prepared by @Jason2866 `platform                    = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2/platform-tasmota-espressif32-2.0.2.zip`

Pre IDF-4.x builds do not show this behavior.

GPIO-17 does output some pattern of ranges of pulses.
It looks like some serial data, but then with much higher frequency. It is 40 MHz.

Zoomed out to get some idea of the pattern:
[image]

Zoomed in on a single block, to show the frequency: 
[image]

It looks as if it is a mirror of the SPI clock of the flash or something like that.
Is that possible?

I also tried to assign the same pin to use PWM as a status LED for my project (showing network activity)
[image]
This shows the 40 MHz frequency is set again right after it was set to the PWM frequency for controlling the LED.

This happens with or without even starting Ethernet.
Also tested with `BOARD_HAS_PSRAM` defined and not defined.

Am I missing something here?

### Sketch

```cpp
No quick test prepared yet.
```

### Debug Message

```plain
No debug output
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-01-11:
@TD-er  You use Tasmota fork build. This behaviour is normal since we use different settings for PSRAM than the official version
https://github.com/Jason2866/esp32-arduino-lib-builder/blob/master/sdkconfig.esp32#L440-L461
GPIO16 and 17 does check for PSRAM at boot and needs to be reinit for normal usage (when no PSRAM connected) from the application.

TD-er, 2022-01-11:
@Jason2866 
Yep that was it. So this can be closed before it was even triaged :)

closed 2022-01-11 by TD-er