espressif/arduino-esp32 #6128: xtensa-esp32-elf-gdb.exe Fails to Execute with Missing DLL Errors (v2.0.1 && v2.0.2)

VisualMicro, 2022-01-12:
### Board

ESP32 Dev Module

### Device Description

ESP32 Dev Module attached via JTAG to ESP PROG probe.
https://www.visualmicro.com/pics/Debug-Help-ESP32-PROG-JTAG-Connections.png

### Hardware Configuration

No external devices beyond debug probe.

### Version

v2.0.1

### IDE Name

Visual Micro + Visual Studio 2022

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

**If this is the incorrect project for the included build of the xtensa-esp32-elf-gcc tool, please advise where this should go instead.**

When executing the xtensa-esp32-elf-gdb.exe included in the board package at _esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2\bin\xtensa-esp32-elf-gdb.exe_, the below missing DLL errors are shown:
[image]

This also occurs with the esp32s2 binaries in: 
_esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2\bin\xtensa-esp32s2-elf-gdb.exe_

**Note:** There is another GDB Exe included at _esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r1\bin\xtensa-esp32-elf-gdb.exe_ which does execute correctly, however all other tools are in the r2 folder, so it is difficult to correctly derive the r1 isolated exe.

If there is an alternative way to derive the correct GDB exe path please advise.

### Sketch

```cpp
void setup() {
    Serial.begin(115200);
    pinMode(17, OUTPUT); 
}
void loop() {
  digitalWrite(17, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(10000);              // wait for a second
  digitalWrite(17, LOW);    // turn the LED off by making the voltage LOW
  delay(10000);              // wait for a second
}
```

### Debug Message

```plain
---------------------------
xtensa-esp32-elf-gdb.exe - System Error
---------------------------
The code execution cannot proceed because libstdc++-6.dll was not found. Reinstalling the program may fix this problem. 

---------------------------
xtensa-esp32-elf-gdb.exe - System Error
---------------------------
The code execution cannot proceed because libgcc_s_sjlj-1.dll was not found. Reinstalling the program may fix this problem.
```

### Other Steps to Reproduce

Double click the below EXE
esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2\bin\xtensa-esp32-elf-gdb.exe
or
esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2\bin\xtensa-esp32s2-elf-gdb.exe

It should result in the GDB Command Prompt being shown:
[image]

It only shows the below errors:
[image]

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

igrr, 2022-01-12:
@me-no-dev JFYI, this was fixed in esp2021r2-patch1 toolchain build for Windows. Just-released esp-2021r2-patch2 also includes this fix.

VisualMicro, 2022-01-12:
Thanks for confirming, is this planned to be referenced by the esp package index.json file in an upcoming release?
(https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json)

VisualMicro, 2022-04-07:
Confirmed this is now working correctly on v2.0.3-RC1, thanks for the update and fix.

closed 2022-04-07 by VojtechBartoska