espressif/arduino-esp32 #6130: RTC with UART, bus busy

peter1a, 2022-01-12:
### Board

ESP32 Wrower-E

### Device Description

Nothing special

### Hardware Configuration

SDA, SCL to tow I2C devices

### Version

latest master

### IDE Name

Arduino IDE

### Operating System

macOS

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

If I'm using RTC (PCF8523) which uses I2C also as UART(SC16IS752) I keep getting error (if using each component separately in sketch it works as it should).

### Sketch

```cpp
#include <SPI.h>
#include <Wire.h>
#include <SC16IS752.h>

#define GPIO 0
#define EXP_PIN 21
#include "RTClib.h"

RTC_PCF8523 rtc;

char daysOfTheWeek[7][12] = {"Sunday", "Monday", "Tuesday", "Wednesday", "Thursday", "Friday", "Saturday"};

SC16IS752 i2cuart = SC16IS752(SC16IS750_PROTOCOL_I2C, 0x48);

void setup()
{
Serial.begin(115200);
Serial.println("Start testing");

pinMode(EXP_PIN, OUTPUT);
digitalWrite(EXP_PIN, HIGH);

if (! rtc.begin()) {
Serial.println("Couldn't find RTC");
Serial.flush();
abort();
}
rtc.start();

i2cuart.begin(SC16IS752_DEFAULT_SPEED, SC16IS752_DEFAULT_SPEED); //baudrate setting
if (i2cuart.ping() != 1) {
Serial.println("Device not found");
while (1);
} else {
Serial.println("Device found");
}

i2cuart.pinMode(GPIO, OUTPUT);
i2cuart.digitalWrite(GPIO, LOW);
}

void loop()
{
i2cuart.digitalWrite(GPIO, HIGH);
delay(1000);
i2cuart.digitalWrite(GPIO, LOW);
delay(1000);

DateTime now = rtc.now();
Serial.print(now.year(), DEC);

delay(5000);
}
```

### Debug Message

```plain
10:14:26.202 -> [D][esp32-hal-i2c.c:1345] i2cProcQueue(): Gross Timeout Dead start=0x28cf0, end=0x28d22, =50, max=50 error=1
10:14:26.202 -> [E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbec1c
10:14:26.202 -> [I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60027000 date=0x16042000
10:14:26.202 -> [I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb8568
10:14:26.202 -> [I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=1
10:14:26.202 -> [I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
10:14:26.237 -> [I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
10:14:26.237 -> [I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=1
10:14:26.237 -> [I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb8600 bits=200
10:14:26.237 -> [I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb8630
10:14:26.237 -> [I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb864c
10:14:26.237 -> [I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=2
10:14:26.237 -> [I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=1
10:14:26.237 -> [I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=-1
10:14:26.274 -> [I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=1
10:14:26.274 -> [I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
10:14:26.274 -> [I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
10:14:26.274 -> [I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
10:14:26.274 -> [I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-01-12:
@me-no-dev can you please take look?

toml2, 2022-01-17:
Seem to have a similar problem with a similar chip (PCF2129A). It worked in Arduino Ide with version 2.0.2. Moved to platformIO. Another I2C device (oled) working perfectly the PCF2129A responds with a garbled message. Tested over multiple devices pull up around 3.7k effective

peter1a, 2022-01-17:
> I was using arduino-esp32 @ 2.0.0 changing to 2.0.2 seemed to have done the trick in my case :
>
> platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.2 <https://github.com/espressif/arduino-esp32.git#2.0.2>

Which changes were made?

Lp,
Peter

1A INTERNET
Peter Zupančič, mag. | Programiranje, mobilni razvoj 
1A INTERNET d.o.o., Naselje nuklearne elektrarne 2, 8270 Krško
T: +386 41 862 299  |  E: ***@***.*** ***@***.***>
Izdelava spletnih strani in razvoj mobilnih aplikacij 
www.1ainternet.net <http://www.1ainternet.net/> | www.mojeure.si <https://www.mojeure.si/> | www.malcajt.com <https://malcajt.com/>

VojtechBartoska, 2022-04-22:
Hello, due to no answer I'm closing this issue as expired.

If you still need help, please reopen it.

Thanks for understanding!

closed 2022-04-22 by VojtechBartoska