espressif/arduino-esp32 #6139: Allow authmode setting in WiFi.begin and WiFi.connect

Moddingear, 2022-01-16:
### Related area

Wifi

### Hardware specification

Esp32

### Is your feature request related to a problem?

Currently the ESP32 can only connect to WPA-secured WiFi using the default library, whereas it has support for all other security profiles.

### Describe the solution you'd like

Add a parameter in WiFi.begin, or add another WiFi.begin where we can give our own config.

me-no-dev, 2022-01-19:
it will actually use the security profile that the AP is advertising.

patricklaf, 2022-01-24:
With 2.0.2 release, ESP32 can not join AP with hidden SSID and WEP security. It works well with release 1.0.6.
If I revert this [commit](https://github.com/espressif/arduino-esp32/commit/40a5c1e46193eb19a146375d40b17ed75fc07275#diff-49411c03b96772a5013214751456f728c1cb3262b032d5121164892a8f3a4531) the ESP32 joins correctly.

me-no-dev, 2022-01-24:
@patricklaf this has since been reverted in master: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiSTA.cpp#L95

closed 2025-01-16 by Parsaabasi