espressif/arduino-esp32 #6140: AttachInterrupt does not work on output of PWM (2.0.2)

mstegen, 2022-01-16:
### Board

ESP32Dev module

### Device Description

I'm using a PWM output, and setup a pin interrupt so i can detect the start of the PWM signal.
This works fine in 2.0.1, but not on 2.0.2

### Hardware Configuration

no

### Version

latest master

### IDE Name

PlatfromIO

### Operating System

Win10/Linux

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

`attachInterrupt` should be able to attach to the output signal of the PWM signal generated by ledC.
this does work in 2.0.1, but not in 2.0.2.

I noticed that there were changes made to the LEDC code (https://github.com/espressif/arduino-esp32/pull/6045)
when using the old (2.0.1) code for `ledcAttachPin` in my code, it works again.
So something in the new code, does not allow a PIN interrupt to be attached to the PWM output.

Below is test code to reproduce the problem.

### Sketch

```cpp
#include <Arduino.h>

#define PIN_PWM_OUT 19
#define PWM_CHANNEL 0

volatile uint32_t isrCounter = 0;

// PWM pin low to high transition ISR
void IRAM_ATTR onPulse() {
    // increment counter
    isrCounter++;
}

void setup() {

    pinMode(PIN_PWM_OUT, OUTPUT);            

    // Setup PWM on channel 0, 1000Hz, 10 bits resolution
    ledcSetup(PWM_CHANNEL, 1000, 10);             // channel 0  => Group: 0, Channel: 0, Timer: 0
    ledcAttachPin(PIN_PWM_OUT, PWM_CHANNEL);      // attach the channels to the GPIO to be controlled
    ledcWrite(PWM_CHANNEL, 512);                  // channel 0, duty cycle 50%

    // Setup PIN interrupt on rising edge
    attachInterrupt(PIN_PWM_OUT, onPulse, RISING);
    
    Serial.begin(115200);
    while (!Serial);
}

void loop() {
  
    Serial.printf("ISR Counter: %u\n",isrCounter);
    delay(1000);
}
```

### Debug Message

```plain
The Timer in above test code will not increase.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-01-19:
@P-R-O-C-H-Y will you please give this a shot?

TD-er, 2022-01-20:
I'm looking into the `ledcWrite` and related functions too on 2.0.2
Are you sure it is a counting issue and not just that it doesn't output a PWM signal on the pin?

I can't get the ESP32 to output PWM signal on GPIO pins since 2.0.2

P-R-O-C-H-Y, 2022-01-20:
Hi @mstegen, move function `pinMode(PIN_PWM_OUT, OUTPUT);` after `ledcAttachPin(PIN_PWM_OUT, PWM_CHANNEL);` and it will work again :)
Making a PR, that will add the `pinMode(PIN_PWM_OUT, OUTPUT);` in ledcAttachPin so there will be no need to call that separately in your code 👍 

```
void setup() {

    // Setup PWM on channel 0, 1000Hz, 10 bits resolution
    ledcSetup(PWM_CHANNEL, 1000, 10);             // channel 0  => Group: 0, Channel: 0, Timer: 0
    ledcAttachPin(PIN_PWM_OUT, PWM_CHANNEL);      // attach the channels to the GPIO to be controlled
   
    pinMode(PIN_PWM_OUT, OUTPUT); //moved here

    ledcWrite(PWM_CHANNEL, 512);                  // channel 0, duty cycle 50%

    // Setup PIN interrupt on rising edge
    attachInterrupt(PIN_PWM_OUT, onPulse, RISING);
    
    Serial.begin(115200);
    while (!Serial);
}

```

@TD-er The LEDC PWM outputs are working as normal :) No problem in 2.0.2
If you use the sketch above, it has to work :)

TD-er, 2022-01-20:
OK will test.
Does the code (with the proposed changes) still work on older SDK versions?

P-R-O-C-H-Y, 2022-01-20:
@TD-er  Yes tried with 2.0.2and 2.0.0 and both works as expected :)

closed 2022-01-20 by me-no-dev

TD-er, 2022-01-20:
I probably found why it isn't working on my setup.
I have a function to try an find a free channel so I'm not interfering with other libraries.
On pre-2.0.2 it worked just fine, but now apparently `ledcRead()` is no longer returning 0 if it is free to use.
Will have to look into it a bit more and if I found something indicating it is a bug, I will open a new issue for it.

mstegen, 2022-01-20:
> Hi @mstegen, move function `pinMode(PIN_PWM_OUT, OUTPUT);` after `ledcAttachPin(PIN_PWM_OUT, PWM_CHANNEL);` and it will work again :) Making a PR, that will add the `pinMode(pin, OUTPUT);` in ledcAttachPin so there will be no need to call that separately in your code 👍
> 
Great @P-R-O-C-H-Y ! thanks a lot for fixing this. 
If i only knew that it this could be fixed this easily...

Miko-Mako, 2022-12-15:
delete "pinMode(PIN_PWM_OUT, OUTPUT);", PWM is a separate block without CPU. And you cannot connect an interrupt to this output.

G-EDM, 2024-02-03:
> delete "pinMode(PIN_PWM_OUT, OUTPUT);", PWM is a separate block without CPU. And you cannot connect an interrupt to this output.

Are you sure? It worked with the old version. I'm using MCPWM to generate PWM and also attached an interrupt to the Pin. It worked until today after upgrading from 3.x to the new 6.5. It was working pretty nice so it is possible.

PR #6160 (2022-01-20): Fixing interrupts in LEDC
## Summary
Added `pinMode(pin,OUTPUT);` to ledcAttach function to ensure that GPIO pin interrupts will work again.

## Impact
Fix for ledc output pin interrupt

## Related links
Closes #6140
commit a5932064 (2022-01-20): Fixing interrupts in LEDC (#6160)

Fixes: #6140