espressif/arduino-esp32 #6141: MbedTLS randomly crashes on HTTP POST (IDFGH-6583)

jgamble-simple, 2022-01-17:
## Environment

- Module or chip used:  ESP32-WROVER-E
- IDF version: v4.4-dev-2313-gc69f0ec329 (version 2.0.0 of arduino framework)
- Build System: platformio
- Compiler version:  8.4.0+2021r1
- Operating System: Windows
- (Windows only) environment type: Windows Subsystem for Linux
- Using an IDE?: No
- Power Supply: 3.3V

## Problem Description

I have the following function which runs once every 10 seconds:

```
void heartBeat(void)
{
  HTTPClient http;

  http.begin(/* Stuff */); //Specify destination for HTTP request

  String dataInput = /* Stuff */;
  int httpResponseCode = http.POST(dataInput);   //Send the actual POST request
}
```
Every once and a while it hangs in the mbedtls code, with the following stack trace:

```
0x4023b937: mpi_sub_hlp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/bignum.c:1382
 (inlined by) mbedtls_mpi_sub_abs at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/bignum.c:1429
0x4023bf77: mbedtls_mpi_mod_mpi at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/bignum.c:1959
 (inlined by) mbedtls_mpi_mod_mpi at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/bignum.c:1943
0x4024da69: calculate_rinv at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/port/esp_bignum.c:131 (discriminator 2)
0x4024db9d: __wrap_mbedtls_mpi_exp_mod at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/port/esp_bignum.c:321
0x40247295: mbedtls_rsa_public at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/rsa.c:790
0x402487c1: mbedtls_rsa_rsassa_pkcs1_v15_verify at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/rsa.c:2431
0x40248879: mbedtls_rsa_pkcs1_verify at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/rsa.c:2489
0x402457d1: rsa_verify_wrap at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/pk_wrap.c:113
0x40245476: mbedtls_pk_verify_restartable at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/pk.c:309
0x4017d61c: ssl_parse_server_key_exchange at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_cli.c:2904
 (inlined by) mbedtls_ssl_handshake_client_step at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_cli.c:3840
0x40179772: mbedtls_ssl_handshake_step at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:8189
0x40179790: mbedtls_ssl_handshake at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:8213
0x401797f2: mbedtls_ssl_handshake at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:8206 (discriminator 1)
0x400fabbd: start_ssl_client(sslclient_context*, char const*, unsigned int, int, char const*, char const*, char const*, char const*, char const*, bool) at C:\Simple\PIMP\esp32-firmware/local_lib/WiFiClientSecure/src/ssl_client.cpp:251
0x400f9fbc: WiFiClientSecure::connect(char const*, unsigned short, char const*, char const*, char const*) at C:\Simple\PIMP\esp32-firmware/local_lib/WiFiClientSecure/src/WiFiClientSecure.cpp:130
0x400fa166: WiFiClientSecure::connect(char const*, unsigned short) at C:\Simple\PIMP\esp32-firmware/local_lib/WiFiClientSecure/src/WiFiClientSecure.cpp:112
 (inlined by) WiFiClientSecure::connect(char const*, unsigned short) at C:\Simple\PIMP\esp32-firmware/local_lib/WiFiClientSecure/src/WiFiClientSecure.cpp:108
0x402923cb: WiFiClientSecure::connect(char const*, unsigned short, int) at C:\Simple\PIMP\esp32-firmware/local_lib/WiFiClientSecure/src/WiFiClientSecure.cpp:117
0x4010106f: HTTPClient::connect() at C:\Simple\PIMP\esp32-firmware/C:/Users/dkendall/.platformio/packages/framework-arduinoespressif32@src-2567d2800cf655927e3f141014aee93c/libraries/HTTPClient/src/HTTPClient.cpp:1117
 (inlined by) HTTPClient::connect() at C:\Simple\PIMP\esp32-firmware/C:/Users/dkendall/.platformio/packages/framework-arduinoespressif32@src-2567d2800cf655927e3f141014aee93c/libraries/HTTPClient/src/HTTPClient.cpp:1081
0x40101e59: HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at C:\Simple\PIMP\esp32-firmware/C:/Users/dkendall/.platformio/packages/framework-arduinoespressif32@src-2567d2800cf655927e3f141014aee93c/libraries/HTTPClient/src/HTTPClient.cpp:582
0x4010201e: HTTPClient::POST(String) at C:\Simple\PIMP\esp32-firmware/C:/Users/dkendall/.platformio/packages/framework-arduinoespressif32@src-2567d2800cf655927e3f141014aee93c/libraries/HTTPClient/src/HTTPClient.cpp:510
 (inlined by) HTTPClient::POST(String) at C:\Simple\PIMP\esp32-firmware/C:/Users/dkendall/.platformio/packages/framework-arduinoespressif32@src-2567d2800cf655927e3f141014aee93c/libraries/HTTPClient/src/HTTPClient.cpp:515
0x400dea06: heartBeat(bool) at C:\Simple\PIMP\esp32-firmware/src/application.cpp:1567 (discriminator 1)
0x400f340d: network_task(void*) at C:\Simple\PIMP\esp32-firmware/src/network.cpp:1505

```

This looks like a bug in the mbedtls stuff.....

It sounds like these issues:

https://github.com/espressif/esp-idf/issues/1681
https://github.com/espressif/esp-idf/issues/3603

But the fixes for those issues is in the version of idf that I am using.....

VojtechBartoska, 2022-04-11:
Hello, can you please retest this on v2.0.3-rc1?

jgamble-simple, 2022-04-11:
@VojtechBartoska In the latest version of our code we have solved this by removing HTTPS from our 10 second "heartbeat" message.....
I have removed that "solution" and have ported to v2.0.3-rc1 and we will se what happens......I am going to leave it run all day and inspect the logs at the end of the day.

VojtechBartoska, 2022-04-11:
@jgamble-simple Thanks for help!

jgamble-simple, 2022-04-11:
@VojtechBartoska Ok....it seems to have not failed yet.....I would have expected it to if it was going to.....I am going to swap out the firmware back to 2.0.0 and confirm that with my current stuff I still get failures......I will leave that run overnight.....If we see failures overnight, I would say that yes this issue is fixed in v2.0.3-rc1

VojtechBartoska, 2022-04-12:
Amazing, we are looking forward for your results

jgamble-simple, 2022-04-12:
@VojtechBartoska With it back on 2.0.0 it did fail in the expect way......So I would say this test concludes that v2.0.3-rc1 has a fix for this problem

VojtechBartoska, 2022-04-12:
Ok, closing. Thanks for testing this!

closed 2022-04-12 by VojtechBartoska