espressif/arduino-esp32 #6142: Problem with ethernet init after reset

Stanimir-Petev, 2022-01-17:
### Board

ESP32-EVB rev.I

### Device Description

The ESP32-EVB is connected only to the computer through USB-Micro cable (for power and serial monitor) and router via standard LAN cable.

### Hardware Configuration

No external hardware or boards attached to the ESP32-EVB

### Version

latest master

### IDE Name

Arduino 1.8.13

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Hello!
I am an employee of Olimex and we have found a problem (that is specific for one of our boards) with the default ethernet example provided with the package (ETH_LAN8720).

On some of our boards ESP32-EVB rev.I (and older) there is a problem with ethernet initialization (ETH.begin () call) immediately after reset. If the USB is unplugged and plugged again it works fine but whenever the reset button is pressed the PHY can't initialize.
After some testing we have figured out that the ESP starts while the PHY is still in reset. And adding a simple delay of 300-350 milliseconds at the start of the sketch solves the issue. But that leaves the question how to implement this change considering the code is part of your package.

I was about to do a pull request but wasn't sure where exactly should I add it. I know that using a preprocessor #if and the macro definition of the board I can isolate this change specifically for ESP32-EVB so it doesn't affect the example for any other.
It's simple for coding - just add
```
#if defined ARDUINO_ESP32_EVB
delay (350);
#endif
```
at the start of the setup function. But I also don't think this is the appropriate place considering this code is supposed to be an example that is a starting point for further sketch development and placing the code there is somewhat ugly.

Is there a more elegant way to implement this change into the package making it specific for ESP32-EVB only (for example adding it somehow to the board variant files or something like that)?

### Sketch

```cpp
The default Ethernet --> ETH_LAN8720 example
```

### Debug Message

```plain
This is the message I receive after reset from the button:

10:09:40.005 -> E (233) lan87xx: lan87xx_init(499): wrong chip ID
10:09:40.005 -> E (233) esp_eth: esp_eth_driver_install(223): init phy failed

If unplug the USB micro cable and plug it again:

10:12:08.003 -> ETH Started
10:12:08.003 -> ETH Connected
10:12:08.003 -> ETH MAC: 08:3A:F2:3B:01:3F, IPv4: 192.168.0.151, FULL_DUPLEX, 100Mbps
```

### Other Steps to Reproduce

With this initialization in the setup() way it works

```
void setup()
{
#if defined ARDUINO_ESP32_EVB
  delay (350);
#endif
  Serial.begin(115200);
  WiFi.onEvent(WiFiEvent);
  ETH.begin();
}
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-01-17:
IMO the delay should be in ETH.begin(); Specifically:
https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/src/ETH.cpp -> ETHClass::begin().

Possibly around here:
`switch(type){
        case ETH_PHY_LAN8720:
            eth_phy = esp_eth_phy_new_lan8720(&phy_config);
            break;`

tcpipInit(); is called both before and after ethernet options are configured.  It may take device testing to verify where the delay should be.

sauttefk, 2022-01-25:
This is due to the reset-/supervisor-chip `U8` on the ESP32-EVB that creates a 470ms delay (`OSC_DIS`)on cold-start before the crystal oscillator for the PHY-clock starts.
Caveat: This board has an issue with warm-starts, as then the crystal oscillator is already running and sending its 50MHz signal to `GPIO0` causing a 50:50 chance of getting in the serial bootloader instead of starting the application.

TD-er, 2022-01-25:
> Caveat: This board has an issue with warm-starts, as then the crystal oscillator is already running and sending its 50MHz signal to `GPIO0` causing a 50:50 chance of getting in the serial bootloader instead of starting the application.

About this, do you also see this happening when the unit crashes?

VojtechBartoska, 2022-04-22:
I'm closing this one as solved, if you still face the issue on latest v2.0.3-rc1 please reopen the issue.

closed 2022-04-22 by VojtechBartoska