espressif/arduino-esp32 #6143: SD Mount Fail

GeoffroyPudda, 2022-01-17:
### Board

ESP32 Devkit

### Device Description

The devices used are [Freematics One+ Model A Boards](https://freematics.com/pages/products/freematics-one-plus/), based on ESP32.

### Hardware Configuration

SD module is connected according
SPI: GPIO18(SCK),GPIO19(MISO),GPIO23(MOSI)
SD CS: GPIO18

### Version

v2.0.1

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

Arduino-ESP32 : V1.0.6
No mounting issues, no writing issues.

Arduino-ESP32 : V2.0.1
Mounting issue exists. Once it appears, it only goes away by flashing back to 1.0.6.

Arduino-ESP32 : V2.0.2
Mounting issue exists. Once it appears, it only goes away by flashing back to 1.0.6.
Additionnal issue is that the I/O operations of the SD Card are much slower (~10 times compared to 2.0.1 and 1.0.6).

Seems related to [issue 5998](https://github.com/espressif/arduino-esp32/issues/5998) and [issue 5701](https://github.com/espressif/arduino-esp32/issues/5701)
The solution provided in issue 5701, which is to upgrad to the latest Arduino Core doesn't work.

### Sketch

```cpp
Tests are done using the SD example sketch given with the corresponding Arduino-ESP32 version.

/*
* Connect the SD card to the following pins:
*
* SD Card | ESP32
* D2 -
* D3 SS
* CMD MOSI
* VSS GND
* VDD 3.3V
* CLK SCK
* VSS GND
* D0 MISO
* D1 -
*/
#include "FS.h"
#include "SD.h"
#include "SPI.h"

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
Serial.printf("Listing directory: %s\n", dirname);

File root = fs.open(dirname);
if(!root){
Serial.println("Failed to open directory");
return;
}
if(!root.isDirectory()){
Serial.println("Not a directory");
return;
}

File file = root.openNextFile();
while(file){
if(file.isDirectory()){
Serial.print(" DIR : ");
Serial.println(file.name());
if(levels){
listDir(fs, file.name(), levels -1);
}
} else {
Serial.print(" FILE: ");
Serial.print(file.name());
Serial.print(" SIZE: ");
Serial.println(file.size());
}
file = root.openNextFile();
}
}

void createDir(fs::FS &fs, const char * path){
Serial.printf("Creating Dir: %s\n", path);
if(fs.mkdir(path)){
Serial.println("Dir created");
} else {
Serial.println("mkdir failed");
}
}

void removeDir(fs::FS &fs, const char * path){
Serial.printf("Removing Dir: %s\n", path);
if(fs.rmdir(path)){
Serial.println("Dir removed");
} else {
Serial.println("rmdir failed");
}
}

void readFile(fs::FS &fs, const char * path){
Serial.printf("Reading file: %s\n", path);

File file = fs.open(path);
if(!file){
Serial.println("Failed to open file for reading");
return;
}

Serial.print("Read from file: ");
while(file.available()){
Serial.write(file.read());
}
file.close();
}

void writeFile(fs::FS &fs, const char * path, const char * message){
Serial.printf("Writing file: %s\n", path);

File file = fs.open(path, FILE_WRITE);
if(!file){
Serial.println("Failed to open file for writing");
return;
}
if(file.print(message)){
Serial.println("File written");
} else {
Serial.println("Write failed");
}
file.close();
}

void appendFile(fs::FS &fs, const char * path, const char * message){
Serial.printf("Appending to file: %s\n", path);

File file = fs.open(path, FILE_APPEND);
if(!file){
Serial.println("Failed to open file for appending");
return;
}
if(file.print(message)){
Serial.println("Message appended");
} else {
Serial.println("Append failed");
}
file.close();
}

void renameFile(fs::FS &fs, const char * path1, const char * path2){
Serial.printf("Renaming file %s to %s\n", path1, path2);
if (fs.rename(path1, path2)) {
Serial.println("File renamed");
} else {
Serial.println("Rename failed");
}
}

void deleteFile(fs::FS &fs, const char * path){
Serial.printf("Deleting file: %s\n", path);
if(fs.remove(path)){
Serial.println("File deleted");
} else {
Serial.println("Delete failed");
}
}

void testFileIO(fs::FS &fs, const char * path){
File file = fs.open(path);
static uint8_t buf[512];
size_t len = 0;
uint32_t start = millis();
uint32_t end = start;
if(file){
len = file.size();
size_t flen = len;
start = millis();
while(len){
size_t toRead = len;
if(toRead > 512){
toRead = 512;
}
file.read(buf, toRead);
len -= toRead;
}
end = millis() - start;
Serial.printf("%u bytes read for %u ms\n", flen, end);
file.close();
} else {
Serial.println("Failed to open file for reading");
}

file = fs.open(path, FILE_WRITE);
if(!file){
Serial.println("Failed to open file for writing");
return;
}

size_t i;
start = millis();
for(i=0; i<2048; i++){
file.write(buf, 512);
}
end = millis() - start;
Serial.printf("%u bytes written for %u ms\n", 2048 * 512, end);
file.close();
}

void setup(){
Serial.begin(115200);
if(!SD.begin()){
Serial.println("Card Mount Failed");
return;
}
uint8_t cardType = SD.cardType();

if(cardType == CARD_NONE){
Serial.println("No SD card attached");
return;
}

Serial.print("SD Card Type: ");
if(cardType == CARD_MMC){
Serial.println("MMC");
} else if(cardType == CARD_SD){
Serial.println("SDSC");
} else if(cardType == CARD_SDHC){
Serial.println("SDHC");
} else {
Serial.println("UNKNOWN");
}

uint64_t cardSize = SD.cardSize() / (1024 * 1024);
Serial.printf("SD Card Size: %lluMB\n", cardSize);

listDir(SD, "/", 0);
createDir(SD, "/mydir");
listDir(SD, "/", 0);
removeDir(SD, "/mydir");
listDir(SD, "/", 2);
writeFile(SD, "/hello.txt", "Hello ");
appendFile(SD, "/hello.txt", "World!\n");
readFile(SD, "/hello.txt");
deleteFile(SD, "/foo.txt");
renameFile(SD, "/hello.txt", "/foo.txt");
readFile(SD, "/foo.txt");
testFileIO(SD, "/test.txt");
Serial.printf("Total space: %lluMB\n", SD.totalBytes() / (1024 * 1024));
Serial.printf("Used space: %lluMB\n", SD.usedBytes() / (1024 * 1024));
}

void loop(){
}
```

### Debug Message

```plain
[ 11043][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed

[ 11043][E][sd_diskio.cpp:795] sdcard_mount(): f_mount failed: (3) The physical drive cannot work

[ 11346][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Rippatec, 2022-01-22:
Hi,
I am also having the SD mount issue. All works OK with 1.06 but after 2 and up to current github version (this post date) I am unable to mount an SD card. (its a custom PCB). I tried editing the delay (#6162) from 300 to 500 (in github version) but did not fix the issue. 
This has been tried with my own software and the basic supplied example test sketch. Same failure result.
PCB has an SPI display, some fast Ferroelectric SPI ram and the SD card on the same bus.
All of these work fine together with 1.06 but after v2, ONLY the SD card no longer functions, with the failure to mount as above.

P-R-O-C-H-Y, 2022-01-24:
Hi, duplicate to #6081. We are discussing and testing changes there :)

PR #6162 (2022-01-31): Fix - SD mount issue
## Summary
On many issues with SD card mount, there is problem with sdWait() only before sending the 1st init cmd 0 = GO_IDLE_STATE. After this command sdWait() works as expected.

Edited GO_IDLE_STATE command:
-> sdWait() is ignored there, only warning message is logged.

Only in sdSelectCard timeout was 300 so added longer timeout as in other calls for sdWait.
According to issue #6107, making a timeout of 500 as in other calls should solve the issue.

Cannot reproduce issues with mounting SD cards.
Will appreciate someone to try this fix.

## Impact
Fixing SD card mounti [... cut]
closed 2022-01-31 by me-no-dev