espressif/arduino-esp32 #6154: WebServer for mobile browsers like FF

geocheats2, 2022-01-19:
### Related area

Automatic address resolution

### Hardware specification

ESP32

### Is your feature request related to a problem?

I have a weird problem while on Chrome for android while typing hostname.local most of the times I get the requested webpage running on esp32 for some reason Firefox for android tends to stick https://www.hostname.local in front of my request. I did a quick search but couldn't find anything related. 

### Describe the solution you'd like

I was wondering if its possible to have https://hostname.local && https://www.hostname.local redirect to http://hostname.local in order the request to be handled correctly by the WebServer

### Describe alternatives you've considered

I thought of making a server.on(www.hostname.local , handlerequest) but i din't think that would work and it didn't, what i tried
```

void handlewww(){
  server.send(301,"text/html","/");
}

 String www = ("www." + String(network_config_struct_0.HOSTNAME) + ".local");
  server.on(www , handlewww);
```

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2022-04-11:
Hello., is this still valid?

geocheats2, 2022-04-11:
i believe so, but as far as i have understood because most of those changes needed are made before compilation of LiWP the change of this on the Arduino level is highly improbable, you may close this if you like.