espressif/arduino-esp32 #6163: esp32_wps:58:41: error: 'arduino_event_info_t' has not been declared

PantalaLabs, 2022-01-20:
### Board

ESP32 devkit V1

### Device Description

ESP32 devkit V1

### Hardware Configuration

no gpio connection

### Version

latest master

### IDE Name

visual studio code

### Operating System

w10

### Flash frequency

40mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

copy and paste the example
all include paths updated

### Sketch

```cpp
sketck exactly the same as provided by the library
```

### Debug Message

```plain
esp32_wps:58:41: error: 'arduino_event_info_t' has not been declared

 void WiFiEvent(WiFiEvent_t event, arduino_event_info_t info)
```

### Other Steps to Reproduce

just try to upload the example to the esp32

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

FernandoGarcia, 2022-03-22:
Actually this error doesn't exists when using ESP32 V2.0.2.

VojtechBartoska, 2022-04-11:
Seems this have been solved. I'm closing the issue, if you still face it, please reopen this ticket.

closed 2022-04-11 by VojtechBartoska