espressif/arduino-esp32 #6167: CRC error - Boards update

rtek1000, 2022-01-20:
### Board

ESP32-CAM Ai-Thinker

### Device Description

ESP32-CAM Ai-Thinker

### Hardware Configuration

ESP32-CAM Ai-Thinker

### Version

v2.0.2

### IDE Name

Arduino IDE 1.8.19

### Operating System

Xubuntu 20.04

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

There is an error when updating the ESP32 board.

### Sketch

```cpp
None
```

### Debug Message

```plain
CRC doens't match
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

rtek1000, 2022-01-20:
[image]

me-no-dev, 2022-01-21:
probably really a temporary issue with Github or something on your system causing CRC to fail. There is no general issue with the package otherwise

VojtechBartoska, 2022-04-11:
Hello @rtek1000, can I consider this as solved? Thanks

VojtechBartoska, 2022-05-03:
Hi,

I'm closing the issue due to no response. 

If needed you can reopen it. 

Thanks for your contribution & understanding!

closed 2022-05-03 by VojtechBartoska