espressif/arduino-esp32 #6168: Please add instructions for manual installation in the portable folder as well

rtek1000, 2022-01-21:
### Related area

Board

### Hardware specification

ESP32

### Is your feature request related to a problem?

I was unable to manually install the esp32 board in the portable folder

### Describe the solution you'd like

Please add instructions for manual installation in the portable folder as well:

https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

I was able to install in the Arduino fixed folder,

(/opt/arduino-1.8.19/hardware/espressif/)

But I didn't find instructions to install in the portable folder.

I also didn't find instructions on how to install manually in the Arduino15 folder

(/home/user_name/.arduino15/)

### Describe alternatives you've considered

I had to install manually because CRC error is occurring on the first 150MB file. So I had to download more than 1GB manually and it didn't give any errors.

It would be great if the download doesn't get lost when CRC error occurs.

Ref.: https://github.com/espressif/arduino-esp32/issues/6167

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2022-01-21:
what do you mean the portable folder? You should install it in your Sketch folder in subfolder called hardware. Like here: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#linux

VojtechBartoska, 2022-05-16:
is this still valid?

closed 2022-06-14 by VojtechBartoska