espressif/arduino-esp32 #6170: Can't recognize I2C device from specific commit.

Boronare, 2022-01-21:
### Board

Lilygo TTGO T-OI Plus

### Device Description

MPU6050(GY-87) attached I2C with SCL on GPIO18 and SDA on GPIO19.

### Hardware Configuration

GPIO18 & 19 are connected to I2C device.

### Version

latest master

### IDE Name

Platformio

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

since f87107dedb6aff148622672376ae6a4aefdfd80c, `TwoWire::endTransmission` returns 2.
tracing that, `i2c_master_write_to_device` returns `ESP_FAIL`.
I tried changing SDA,SCL to other GPIO(are not connected) and returned same.

just before that commit -on ce85cf03cc46cf680dbf5a1d9666952c5e23833d-  I could use I2C. Writing and reading worked well. (but process suddenly stopped then watchdog appeared on this version)

Still not available in recent master(0b4516eef50d447dee1342d9780efcbd67a13f24).

Well, it can be issue about wire initialization, anyway, not work since that version.

### Sketch

```cpp
//from https://github.com/SlimeVR/SlimeVR-Tracker-ESP/blob/main/src/main.cpp - setup()
    Wire.begin(PIN_IMU_SDA, PIN_IMU_SCL);
    uint8_t first = I2CSCAN::pickDevice(0x68, 0x69, true); // pickDevice calls isI2CExist.
==================================
// /lib/i2cscan.cpp
    bool isI2CExist(uint8_t addr) { // addr = 0x68, mpu6050.
        Wire.beginTransmission(addr);
        byte error = Wire.endTransmission(); //<- Here. returns error = 2.
        Serial.printf("endTransmisson : %d\n",error);
        if(error == 0)
            return true;
        return false;
    }
```

### Debug Message

```plain
endTransmisson : 2
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-01-21:
The change that you point to just made the requirements for I2C a bit more strict. I have the feeling that somewhere in your code, or in a library you are using, there is a wrong chain of commands. The example above works for me to scan if device is available.

Boronare, 2022-01-21:
Ah, sry. I told wrong.
[`i2c_master_cmd_begin`](https://github.com/espressif/arduino-esp32/blob/6a7bcabd6b7a33f074f93ed60e5cc4378d350b81/cores/esp32/esp32-hal-i2c.c#L175) returned `ESP_FAIL`, not `i2c_master_write_to_device`. Isn't "it ran and returned error" means it passed some strict checks before?

basic approach code to this library is like this.
```
Wire.begin(19,18);
Wire.setClock(400000);
delay(500);
Wire.beginTransmission(0x68);
byte error = Wire.endTransmission();
Serial.printf("isI2CExist:68=%d\n",error);
```

jakkra, 2022-03-16:
Hi, I seem to have a very similar problem, however for me I'm getting timeout error instead. I have narrowed it down to be reproduceable using this i2c detect code:
In arduino-esp32 2.0.0 it works but in 2.0.1 and 2.0.2 it does not find any addresses.

```
#include <Wire.h>

void setup()
{
  Serial.begin (115200);  
  Wire.begin (27, 22);   // sda= GPIO_27 /scl= GPIO_22
}

void Scanner ()
{
  Serial.println ();
  Serial.println ("I2C scanner. Scanning ...");
  byte count = 0;

  for (byte i = 8; i < 120; i++)
  {
    Wire.beginTransmission (i);          // Begin I2C transmission Address (i)
    int res = Wire.endTransmission ();
    Serial.println(res);
    if (res == 0)  // Receive 0 = success (ACK response) 
    {
      Serial.print ("Found address: ");
      Serial.print (i, DEC);
      Serial.print (" (0x");
      Serial.print (i, HEX);     // PCF8574 7 bit address
      Serial.println (")");
      count++;
    }
  }
  Serial.print ("Found ");      
  Serial.print (count, DEC);        // numbers of devices
  Serial.println (" device(s).");
}

void loop()
{
  Scanner ();
  delay (100);
}
```

2.0.0 log:
```
I2C scanner. Scanning ...
Found address: 10 (0xA)
Found address: 13 (0xD)
Found address: 16 (0x10)
Found address: 19 (0x13)
Found address: 22 (0x16)
...
```

2.0.1 and 2.0.1 no devices found.
Wire.endTransmission () returns -5
```
I2C scanner. Scanning ...
5
5
...
Found 0 device(s).
```

Is this scanner example missing something that is needed after the driver changes in 2.0.1? @me-no-dev 

If it matters I'm trying to get the Adafruit Motor Shield V2 working in the esp-idf, but first I want to get it working in Arduino a there is an Arduino library for it, however not working on 2.0.1+

me-no-dev, 2022-03-16:
@jakkra can you try this code so we can see better whet is returned?
```cpp

void scanI2C(){
  uint8_t devices, error, address;
  Serial.println("Scanning for I2C devices...");
  devices = 0;
  for(address = 0x01; address < 0x7f; address++){
    Wire.setClock(100000);
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (error == 0){
      devices++;
      Serial.printf("I2C device found at address 0x%02X\n", address);
    } else if(error == 4){
      Serial.printf("Error at address 0x%02X\n", address);
    } else if(error == 5){
      Serial.printf("Timeout at address 0x%02X\n", address);
    } else if(error != 2){
      Serial.printf("Unknown error %u at address 0x%02X\n", error, address);
    }
  }
  Serial.printf("%u I2C devices found\n", devices);
}

```

VojtechBartoska, 2022-05-04:
I'm closing the issue as expired due to no answer.

If needed, please reopen it.

Thanks for understanding.

closed 2022-05-04 by VojtechBartoska