espressif/arduino-esp32 #6173: HTTPS_OTA_Update.ino fails with: esp-tls: mbedtls_ssl_handshake returned -0x7280

claaudiam, 2022-01-21:
### Board

M5Stack core2

### Device Description

M5Stack core2 EduKit

### Hardware Configuration

No

### Version

latest master

### IDE Name

platformIO

### Operating System

Windows10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Hi,
Trying to run [HTTPS_OTA_Update.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/examples/HTTPS_OTA_Update/HTTPS_OTA_Update.ino
) from Update lib using my own server and fails with: 
```
E (22929) esp-tls: mbedtls_ssl_handshake returned -0x7280
```
Googling I found [this](https://github.com/espressif/esp-mqtt/issues/98#issuecomment-464493353) which suggests to enable mbedtls debugging.

Please, how can I do this, or can you suggest what could be causing the mbedtls_ssl_handshake returned -0x7280?

Thanks for any help

### Sketch

```cpp
// This sketch provide the functionality of OTA Firmware Upgrade
#include "WiFi.h"
#include "HttpsOTAUpdate.h"
// This sketch shows how to implement HTTPS firmware update Over The Air.
// Please provide your WiFi credentials, https URL to the firmware image and the server certificate.

static const char *ssid     = "GalaxyA124C2A"; // "FRITZ!Box_7530_HD";  // your network SSID (name of wifi network)
static const char *password = "iocq8229";//"52588545631720690851"; // your network password

static const char *url = "https://test1.quarklink.tech:6000/fota?deviceid=7cfd1f5138384e1f9086579ec7741c8a1b951fecf671b3689ecb18d6b778c36f"; //state url of your firmware image

static const char *server_certificate = "-----BEGIN CERTIFICATE-----\n"
"MIIBXjCCAQSgAwIBAgIIFprucTzoUAUwCgYIKoZIzj0EAwIwEjEQMA4GA1UEAxMH\n"
"T0VNUm9vdDAgFw0yMTA4MTMxNzM5NDVaGA8yMDUxMDgwNjE3Mzk0NVowEjEQMA4G\n"
"A1UEAxMHT0VNUm9vdDBZMBMGByqGSM49AgEGCCqGSM49AwEHA0IABA3E7nFFzgR7\n"
"4TyNAk/cy2mcm/MjlCuO8/LTF6SoyrlojjVlHk9UtGQ3S9dEK9oTYlni+RP/FQkz\n"
"b0fttY8OHFmjQjBAMA4GA1UdDwEB/wQEAwIChDAPBgNVHRMBAf8EBTADAQH/MB0G\n"
"A1UdDgQWBBRDzuhGETJk+ORn7ktqHgJ+FE2jYTAKBggqhkjOPQQDAgNIADBFAiEA\n"
"mk+m6WQnS2yG7Rfjc/nNnKUTMBGk6s3oEKjY7jltIoMCIAoZMHhCKe+M5au46fp3\n"
"L6J7+HWLyIpOu00Rb2t4DbiB\n"
"-----END CERTIFICATE-----\n";

static HttpsOTAStatus_t otastatus;

void HttpEvent(HttpEvent_t *event)
{
    switch(event->event_id) {
        case HTTP_EVENT_ERROR:
            Serial.println("Http Event Error");
            break;
        case HTTP_EVENT_ON_CONNECTED:
            Serial.println("Http Event On Connected");
            break;
        case HTTP_EVENT_HEADER_SENT:
            Serial.println("Http Event Header Sent");
            break;
        case HTTP_EVENT_ON_HEADER:
            Serial.printf("Http Event On Header, key=%s, value=%s\n", event->header_key, event->header_value);
            break;
        case HTTP_EVENT_ON_DATA:
            break;
        case HTTP_EVENT_ON_FINISH:
            Serial.println("Http Event On Finish");
            break;
        case HTTP_EVENT_DISCONNECTED:
            Serial.println("Http Event Disconnected");
            break;
    }
}

void setup(){

    Serial.begin(115200);
    Serial.print("Attempting to connect to SSID: ");
    WiFi.begin(ssid, password);
   
    // attempt to connect to Wifi network:
    while (WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(1000);
    }

    Serial.print("Connected to ");
    Serial.println(ssid);
    
    HttpsOTA.onHttpEvent(HttpEvent);
    Serial.println("Starting OTA");
    HttpsOTA.begin(url, server_certificate, true); 

    Serial.println("Please Wait it takes some time ...");
}

void loop(){
    otastatus = HttpsOTA.status();
    if(otastatus == HTTPS_OTA_SUCCESS) { 
        Serial.println("Firmware written successfully. To reboot device, call API ESP.restart() or PUSH restart button on device");
    } else if(otastatus == HTTPS_OTA_FAIL) { 
        Serial.println("Firmware Upgrade Fail");
    }
    delay(1000);
}
```

### Debug Message

```plain
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Attempting to connect to SSID: [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
.....[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
.[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.142.61, MASK: 255.255.255.0, GW: 192.168.142.116
Connected to GalaxyA124C2A
Starting OTA
Please Wait it takes some time ...
E (22929) esp-tls: mbedtls_ssl_handshake returned -0x7280
E (22931) esp-tls: Failed to open new connection
E (22931) TRANS_SSL: Failed to open a new connection
E (22932) HTTP_CLIENT: Connection failed, sock < 0
E (22937) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (22944) esp_https_ota: Failed to establish HTTP connection
Http Event Disconnected
Http Event Disconnected
Firmware Upgrade Fail
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

KARTIK-eng, 2022-01-29:
I'm facing exactly the same problem. @claaudiam did you find any solution yet?

claaudiam, 2022-01-31:
no, not yet

torntrousers, 2022-01-31:
Looking in Git history the sample was added back on [here](https://github.com/espressif/arduino-esp32/commit/1fd5cd79c52c7896985a5646211c4a169e612b10#diff-faa2ba8369b1bf18ba2664f435b50dc80c39fee04dd4f7430b5ca3ae0c9d4e84) 25th August 2020. Looking at the arduino-esp32 [release history](https://github.com/espressif/arduino-esp32/releases?page=2) that must have been using either [1.0.4](https://github.com/espressif/arduino-esp32/releases/tag/1.0.5-rc1) or master going to [1.0.5-RC1](https://github.com/espressif/arduino-esp32/releases/tag/1.0.5-rc1).  Maybe as a start try using 1.0.4 or 1.0.5-RC1 to see if the sample worked then?

KARTIK-eng, 2022-02-01:
Any leads? Maybe I can look into it too

KARTIK-eng, 2022-02-01:
> Looking in Git history the sample was added back [here](https://github.com/espressif/arduino-esp32/commit/1fd5cd79c52c7896985a5646211c4a169e612b10#diff-faa2ba8369b1bf18ba2664f435b50dc80c39fee04dd4f7430b5ca3ae0c9d4e84) 25th August 2020. Looking at the arduino-esp32 [release history](https://github.com/espressif/arduino-esp32/releases?page=2) that must have been using either [1.0.4](https://github.com/espressif/arduino-esp32/releases/tag/1.0.5-rc1) or master going to [1.0.5-RC1](https://github.com/espressif/arduino-esp32/releases/tag/1.0.5-rc1). Maybe as a start try using 1.0.4 or 1.0.5-RC1 to see if the sample worked then?

@torntrousers I've used so many old libraries and codes that work for the current arduino-esp32 release. So I don't think thats our best bet to try. Ive been searching around and found that the issue might lie in the way the certificate is presented in the program. However didn't find the way that works.

KARTIK-eng, 2022-02-01:
> Looking in Git history the sample was added back [...]

@torntrousers I've used so many old libraries and codes that work for the current arduino-esp32 release. So I don't think thats our best bet to try. Ive been searching around and found that the issue might lie in the way the certificate is presented in the program. However didn't find the way that works.

igor-d-n, 2022-02-14:
Check free heap size. Mbedtls don't work good. Trying to set options like #define MBEDTLS_SSL_IN_CONTENT_LEN 4096 and it has no any effect.

KARTIK-eng, 2022-02-23:
> Check free heap size. Mbedtls don't work good. Trying to set options like #define MBEDTLS_SSL_IN_CONTENT_LEN 4096 and it has no any effect.

I checked the free heap size. It's 370000. What should I do now?

VojtechBartoska, 2022-04-11:
Hello, can you please retest this on v2.0.3-rc1?

torntrousers, 2022-04-14:
I'd already tried that but no it doesn't fix this issue.

VojtechBartoska, 2022-04-14:
I'm adding this to our Roadmap and we will investigate this more.

Josh-TapNoa, 2024-01-02:
This is a server EOF response? U sure all is well on server side OP?

closed 2025-01-16 by Parsaabasi