espressif/arduino-esp32 #6175: .begin() on FFat builtin flash fails if using WiFi

dlyckelid, 2022-01-21:
### Board

ESP32 Wrover E 16MB

### Device Description

It is a completely custom board and I cannot provide the layout for it becuase it is a product under development.

### Hardware Configuration

EN: Reset button
GPIO36: Digital input 1 thru optocoupler
GPIO39: Digital input 2 thru optocoupler
GPIO34: Digital input 3 thru optocoupler
GPIO35: Digital input 4 thru optocoupler
GPIO32: Digital input 5 thru optocoupler
GPIO33: Digital input 6 thru optocoupler
GPIO25: Digital input 7 thru optocoupler
GPIO26: Digital input 8 thru optocoupler
GPIO27: VSPI CS Winbond 8MB memory
GPIO14: HSPI_CLK KX023-1025 Accelerometer
GPIO12: HSPI_MISO KX023-1025 Accelerometer
GPIO13: HSPI_MOSI KX023-1025 Accelerometer
GPIO23: VSPI_MOSI Winbond 8MB memory
GPIO22: SIM7070G_TX (Software serial)
GPIO21: Digital output LED
GPIO19: VSPI_MISO Winbond 8MB memory
GPIO19: VSPI_CLK Winbond 8MB memory
GPIO05: SIM7070G_RX (Software serial)
GPIO04: SIM7070G_PWR Digital output
GPIO01: BOOT Button
GPIO02: Digital output LED (Yes not ideal but short on IO...)
GPIO15: HSPI_CS KX023-1025 Accelerometer

### Version

v1.0.6

### IDE Name

Visual Micro

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Hi!

I am trying to use FFat for my project to save settings in different files.
I have a esp32-Wrover-E(16MB) on a completely custom PCB.

The issue I have is that I cant get FFat to work together with WiFi. I use WiFi now to display a userinterface with ESPAsyncWebServer and also use it to connect to internet in some cases (Other I use a 4G-connection).

The first I tried was to do FFat.begin() in the setup step and then not any FFat.end()
When trying this WiFi will not work and even if there is a AP available if I connect to it nothing is answering.

Next I changed this to have FFat.begin() and then FFat.end() after every operation and in this case the WiFi work normally but when I try to save the configuration I get an error when using FFat.begin()
"Mounting FFat partition failed! Error: 257"

I Guess that the internal Flash on Wrover uses the same SPI-bus as WiFi but I can not find any information on how to make them work together. I have seen examples of it working with ESPAsyncWebServer and SPIFFS but looking thru the code I cannot find anything special

### Sketch

```cpp
FFat.begin()
```

### Debug Message

```plain
"Mounting FFat partition failed! Error: 257"
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

dlyckelid, 2022-01-24:
I have now tested the same code with SPIFFS and then all seems to work fine.
So the issue seems to be with FFat library

me-no-dev, 2022-01-24:
seems you are still on 1.0.6 Could you please try with 2.0.2?

dlyckelid, 2022-02-02:
> seems you are still on 1.0.6 Could you please try with 2.0.2?

Hi!

I am testing that right now. But having issues updating because it is such a big project and it seems to still use some old libraries somehow.

dlyckelid, 2022-02-02:
The issue still remains in 2.0.2
SPIFFS works as expected but FAT does not

dlyckelid, 2022-02-04:
I have found the issue and it is not about the Fat at all. I used the ESP32 Wrover profile and that defaults  to 4MB flash and I cannot change that in visual micro. Using the ESP32 dev module profile set to 16MB flash makes everything work as it should.

closed 2022-02-04 by dlyckelid