espressif/arduino-esp32 #6177: Compiling Error

UcanLahana, 2022-01-21:
### Board

DOIT ESP32 DEVKIT V1

### Device Description

ESP-WROOM-32

### Hardware Configuration

An EByte E32 LoRa module at M0(D5),M1(D4),Tx2(Rx),Rx2(Tx) and an 32x128 OLED scrren at D21,D22 (SCL SDA), DHT11 at D23.

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I can't load code to the ESP32.

### Sketch

```cpp
#include "Arduino.h"
#include "LoRa_E32.h"
#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

LoRa_E32 e32ttl(&Serial2, 15, 5, 4);

#define OLED_RESET 4
Adafruit_SSD1306 display(OLED_RESET);
#define NUMFLAKES 10
#define XPOS 0
#define YPOS 1
#define DELTAY 2
double count=0;

#define LOGO16_GLCD_HEIGHT 32
#define LOGO16_GLCD_WIDTH  128 

#if (SSD1306_LCDHEIGHT != 32)
#error("Height incorrect, please fix Adafruit_SSD1306.h!");
#endif

struct Signal {
  char type[10] = "Orenda";
  byte temp[5];
  byte hum[5];
} data;
 
 
void setup() {
  Serial.begin(9600);
  e32ttl.begin();
  e32ttl.setMode(MODE_2_POWER_SAVING);
  delay(500);
  display.begin(SSD1306_SWITCHCAPVCC, 0x3C);
  display.display();
  delay(2000);
  esp_sleep_enable_ext0_wakeup(GPIO_NUM_15,LOW);
  esp_light_sleep_start();
  display.clearDisplay();
}
```

### Debug Message

```plain
Arduino:1.8.19 (Windows 10), Kart:"DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

& was unexpected at this time.

exit status 1

DOIT ESP32 DEVKIT V1 kartı için derleme hatası.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-01-25:
You need to narrow down the problem.  I suspect your issue is with LORA.  This line looks suspicious:
`LoRa_E32 e32ttl(&Serial2, 15, 5, 4);`

Temporarily comment out the code related to LORA and see if the rest compiles.

VojtechBartoska, 2022-04-11:
Still relevant?

VojtechBartoska, 2022-05-03:
Hi,

I'm closing the issue due to no response. 

If needed you can reopen it. 

Thanks for your contribution & understanding!

closed 2022-05-03 by VojtechBartoska