espressif/arduino-esp32 #6180: M5Core-Ink compile fails at ESP32 2.0.2

thorv, 2022-01-23:
### Board

M5Stack-CoreInk

### Device Description

M5Stack-CoreInk, no option.

### Hardware Configuration

default

### Version

v2.0.2

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 11

### Flash frequency

N/A

### PSRAM enabled

no

### Upload speed

921600

### Description

With only M5CoreInk.h, empty setup()/loop(),
compile failed.

### Sketch

```cpp
#include <M5CoreInk.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Message

```plain
Compiling library "M5-CoreInk"

"C:\\Users\\default\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\gcc8_4_0-esp-2021r2/bin/xtensa-esp32-elf-g++" 
---
(snip)
---
C:\Users\default\Documents\Arduino\libraries\M5-CoreInk\src\M5CoreInk.cpp: In member function 'int M5CoreInk::begin(bool, bool, bool)':

C:\Users\default\Documents\Arduino\libraries\M5-CoreInk\src\M5CoreInk.cpp:23:33: error: call of overloaded 'begin(int, int, int)' is ambiguous

         Wire.begin(32, 33, 10000);

                                 ^

In file included from C:\Users\default\Documents\Arduino\libraries\M5-CoreInk\src\utility/BM8563.h:4,

                 from C:\Users\default\Documents\Arduino\libraries\M5-CoreInk\src\M5CoreInk.h:8,

                 from C:\Users\default\Documents\Arduino\libraries\M5-CoreInk\src\M5CoreInk.cpp:1:

C:\Users\default\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\Wire\src/Wire.h:79:10: note: candidate: 'bool TwoWire::begin(int, int, uint32_t)'

     bool begin(int sda=-1, int scl=-1, uint32_t frequency=0); // returns true, if successful init of i2c bus

          ^~~~~

C:\Users\default\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\Wire\src/Wire.h:80:10: note: candidate: 'bool TwoWire::begin(uint8_t, int, int, uint32_t)'

     bool begin(uint8_t slaveAddr, int sda=-1, int scl=-1, uint32_t frequency=0);

          ^~~~~

(Japanese says using library M5-CoreInk ver 0.0.2)
次のフォルダのライブラリM5-CoreInkバージョン0.0.2を使用中：C:\Users\default\Documents\Arduino\libraries\M5-CoreInk 

(Japanese says using library Wire ver 2.0.0)
次のフォルダのライブラリWireバージョン2.0.0を使用中：C:\Users\default\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\Wire 

(Japanese says using library SPI ver 2.0.0)
次のフォルダのライブラリSPIバージョン2.0.0を使用中：C:\Users\default\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\SPI 

exit status 1
```

### Other Steps to Reproduce

w/ board library 1.0.6, no error.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-01-24:
Hello @thorv, probably the problem is on Library side now. 

@me-no-dev Can you please take a look on this issue? Wire.h was refactored recently and it seems it's not compatible now with mentioned library.

me-no-dev, 2022-01-24:
Yes, this needs to be fixed on the library side. `Wire.begin(32, 33, (uint32_t)10000);` should do the trick

closed 2022-01-24 by me-no-dev