espressif/arduino-esp32 #6185: Enable UART Flow Control

GioTB, 2022-01-24:
### Related area

UART

### Hardware specification

ESP32

### Is your feature request related to a problem?

No, i just need to use uart flow control.

### Describe the solution you'd like

Enable support for flow control (RTS and CTS, perhaps it would be also usefull to enable DTR and DSR)

### Describe alternatives you've considered

Add a parameter on the "begin" function that allows to select if the serial should use CTS/RTS and a pin assigment (might be just adding de pins, and if they are different than -1 the flow control would be enabled).
Also a "setRts" and "setCts" function could be implemented to enable one or both flow control pins

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2022-01-24:
@SuGlider Can you please take a look?

SuGlider, 2022-01-24:
@GioTB 

Thanks for the suggestion.

In the meanwhile, Arduino UART is based on IDF, thus all the IDF calls can be used in conjunction with Arduino ```HardwareSerial```\
You can activate it by using specific IDF calls within your Arduino Sketch.

You can use this IDF call with ```UART_PIN_NO_CHANGE``` for the pins that you don't want to change.
[```uart_set_pin(uart_port_t uart_num, int tx_io_num, int rx_io_num, int rts_io_num, int cts_io_num)```](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#_CPPv412uart_set_pin11uart_port_tiiii)

You can also set Hardware Flow Control mode to use CTS, RTS together or each one isolated.
[```uart_set_hw_flow_ctrl(uart_port_t uart_num, uart_hw_flowcontrol_t flow_ctrl, uint8_t rx_thresh)```](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#_CPPv421uart_set_hw_flow_ctrl11uart_port_t21uart_hw_flowcontrol_t7uint8_t)

For instance:
``` dart
#include "driver/uart.h"
void setup() {
  Serial.begin(115200);   // uses default RX/TX pins for UART 0

  // it sets pin 2 for RTS and pin 4 for CTS in UART 0 (Arduino Serial)
  uart_set_pin(UART_NUM_0, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE, 2, 4);
  
  // UART_HW_FLOWCTRL_CTS_RTS or only UART_HW_FLOWCTRL_CTS | UART_HW_FLOWCTRL_RTS
  // UART_HW_FLOWCTRL_DISABLE to disable Hardware Control on UART
  // 64 is the Threshold for RX Flow control when RTS is used. It goes from 0 to 127 (bytes).
  uart_set_hw_flow_ctrl(UART_NUM_0, UART_HW_FLOWCTRL_CTS_RTS, 64);
}
```

Please let me know if this would solve your current need.

GioTB, 2022-01-24:
Hello, thanks for your response. I didn´t know the esp-idf functions where available on the Arduino framework, i tested it, and it works!, but, it still has the problem of the queue of the HardwareSerial filling up if i don´t empty it, of course this wouldn´t be a "hardware flow control" implementation, and i could just set the buffer to a bigger size, but i need totally confident that the flow control will prevent the other device from sending data. Since the hardware serial doesn´t give direct access to that queue i can´t check it on my code to empty it or to disable CTS line if the queue it´s full. Never the less since i see i have access to the esp-idf functions i´ll implement the project using those. As an idea, for the implementation of this flow control on the arduino framework: on the uart_isr it could check if the queue it´s "almost full" (like the Rx thresshold), for de-asserting the CTS line, so any byte will be lost.

Thanks for your help!, if there is anything that i could help let me know.

SuGlider, 2022-01-24:
Arduino HardwareSerial is implemented on top of IDF. Thus any IDF call will affect directly HardwareSerial as well.

I didn't test it, but ```uart_set_hw_flow_ctrl(UART_NUM_0, UART_HW_FLOWCTRL_CTS_RTS, 64);``` shall rise RTS line when ESP32 has more than 64 bytes, in this example, in FIFO that have not been sent to IDF UART Driver Queue - which is the same used by Serial.read(), for instance.

Anyway, we have now a UART ISR Callback that may help your application in dealing with incoming UART data.
```Serial.onReceive(userCallBackFunc);``` - PR #6134

Please test it and let me know by droping a message in this issue.
Thanks!

GioTB, 2022-01-25:
I´m using platformio, and it seems that it´s using the 1.0.6 version, so i´m trying to upgrade it to the lastest version (if i´m not wrong it´s the 4.4). As far as i can see from the code, i think it should work exactly the way you say it, once i try it i will let you know, Thanks!!!!

SuGlider, 2022-01-25:
1.0.6 will use IDF 3.3. It doesn't mix HardwareSerial with IDF calls and the issue you commented with Queue and FIFO will actually happen.

The Arduino Core 2.0.2 uses IDF 4.4 and as said it's HardwareSerial is built on top of IDF, which will allow you to just use Serial.begin() followed by IDF RTS setup call that will work seamless all together.

SuGlider, 2022-02-12:
See PR #6272

GioTB, 2022-02-13:
Anaother comment about the hardware flow control, when i implemented it at first it didin´t work well, and a colleague found out that it was becuase on the uart_begin function it has de uart_set_mode to "UART_MODE_RS485_HALF_DUPLEX" and after changing it to UART_MODE_UART, it works great.
so the final code we have to implement to make it work was:

Serial.begin(115200);
uart_set_pin(UART_NUM_0, UART_PIN_NO_CHANGE, UART_PIN_NO_CHANGE,RTS_PIN , CTS_PIN);
uart_set_hw_flow_ctrl(UART_NUM_0, UART_HW_FLOWCTRL_CTS_RTS, 64);
uart_set_mode(UART_NUM_0, UART_MODE_UART);

SuGlider, 2022-02-14:
Yes, it was also fixed and removed in a prior PR #6133

PR #6272 (2022-02-17): Fixes UART pin setting + adds CTS/RTS HW Flow Control
## Summary
This PR has two components:
1- Fixes Serial.begin() in order to keep RX/TX pins the same, as previouly set, when those parameter are not used.
2 - Adds Hardware Flow Control and adds RTS/CTS pins setting 

## Impact

Serial will keep RX/TX pins unmodified in case of several ```begin()``` calls:

``` dart
void setup() {
    Serial1.begin(115200, SERIAL_8N1, 18, 19);
    Serial1.println("HELLO WORLD!");
  
    Serial1.begin(115200);
    Serial1.begin(115200);
    Serial1.println("HELLO WORLD!  2nd Time!");
}
```

Modified and New f [... cut]
closed 2022-02-17 by me-no-dev