espressif/arduino-esp32 #6186: 300625

nccwarp9, 2022-01-25:
### Board

DOIT ESP32 DEVKIT V1

### Device Description

plain devkit

### Hardware Configuration

no additional connections

### Version

v2.0.2

### IDE Name

Vs code

### Operating System

win 10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

115200

### Description

Im getting this error and I cant find any reference on google about it. 
I added the esp32 board to vs code via json provided in the release section.

and when I try to verify I get this error just after esp32 build script comes to portion of copying 
cmd /c if exist "d:\\Programing\\Arduino\\portal\\WebSystem - ESP32\\partitions.csv" COPY /y  .... 

'"C:\x64\ansicon"' is not recognized as an internal or external command,
operable program or batch file.
        1 file(s) copied.

exit status 1

### Sketch

```cpp
none
```

### Debug Message

```plain
Using board 'esp32' from platform in folder: C:\Users\vvv\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2
Using core 'esp32' from platform in folder: C:\Users\vvv\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2
cmd /c if exist "d:\\Programing\\Arduino\\portal\\WebSystem - ESP32\\partitions.csv" COPY /y "d:\\Programing\\Arduino\\portal\\WebSystem - ESP32\\partitions.csv" "C:\\Users\\vvv\\AppData\\Local\\Temp\\arduino_build_264535\\partitions.csv"
'"C:\x64\ansicon"' is not recognized as an internal or external command,
operable program or batch file.
        1 file(s) copied.

exit status 1
DEBUG StatusLogger Stopping LoggerContext[name=1e6f5c3, org.apache.logging.log4j.core.LoggerContext@11a9022]
DEBUG StatusLogger Stopping LoggerContext[name=1e6f5c3, org.apache.logging.log4j.core.LoggerContext@11a9022]...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-04-11:
Hello, is this still relevant?

VojtechBartoska, 2022-05-03:
Hi,

I'm closing the issue due to no response. 

If needed you can reopen it. 

Thanks for your contribution & understanding!