espressif/arduino-esp32 #6187: Enabling C++ Exceptions compiler option

srgbin, 2022-01-25:
### Board

ESP32-C3

### Device Description

ESP32-C3

### Hardware Configuration

no

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Ubuntu 20.04

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Compiling ESP32 BLE Arduino library by nkolban requires the C++ Exceptions compiler option to be enabled.

### Sketch

```cpp
/*
    Based on Neil Kolban example for IDF: https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLE%20Tests/SampleServer.cpp
    Ported to Arduino ESP32 by Evandro Copercini
    updates by chegewara
*/

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

// See the following for generating UUIDs:
// https://www.uuidgenerator.net/

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  BLEDevice::init("Long name works now");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

  pCharacteristic->setValue("Hello World says Neil");
  pService->start();
  // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined! Now you can read it in your phone!");
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);
}
```

### Debug Message

```plain
#error "C++ exception handling must be enabled within make menuconfig. See Compiler Options > Enable C++ Exceptions."
```

### Other Steps to Reproduce

Enabling option in the sdkconfig file does not solve the problem

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

atanisoft, 2022-01-25:
C++ Exceptions *ARE* enabled in 2.0.x:
https://github.com/espressif/arduino-esp32/blob/6a7bcabd6b7a33f074f93ed60e5cc4378d350b81/tools/sdk/esp32/sdkconfig#L233-L234

Is there a reason you are not using the built-in BLE library that is based on the no longer maintained `https://github.com/nkolban/esp32-snippets/` repository?

srgbin, 2022-01-26:
C++ Exceptions ARE disabled in 2.0.x: for esp32c3
https://github.com/espressif/arduino-esp32/blob/6a7bcabd6b7a33f074f93ed60e5cc4378d350b81/tools/sdk/esp32c3/sdkconfig#L205
I am using the library https://github.com/nkolban/ESP32_BLE_Arduino. By built-in BLE library do you mean the ArduinoBLE library https://www.arduino.cc/en/Reference/ArduinoBLE? 
Indicated that this library does not support esp32c3.

atanisoft, 2022-01-26:
Ahh, normally these are kept in sync. I'm not sure why the C3 file would not have it enabled whereas the others do.

If you can, please open a PR to modify [this](https://github.com/espressif/esp32-arduino-lib-builder/blob/master/sdkconfig.esp32c3) file and @me-no-dev will review for merge and it will show up over here in this repository soon thereafter via an IDF sync PR.

igrr, 2022-01-26:
@me-no-dev I think we could convert the existing target-specific sdkconfig files into:

sdkconfig.defaults (common options which are different from the default ones)
sdkconfig.defaults.esp32 (esp32-specific options)
sdkconfig.defaults.esp32s2 (esp32s2-specific options)

and so on.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#target-dependent-sdkconfig-defaults
(i can submit a PR to arduino-lib-builder for this, if you agree)

srgbin, 2022-01-26:
Great, it would be great if you submit a PR for this.

me-no-dev, 2022-01-27:
> @me-no-dev I think we could convert the existing target-specific sdkconfig files into:
>
> sdkconfig.defaults (common options which are different from the default ones) sdkconfig.defaults.esp32 [...]

I do agree :)

me-no-dev, 2022-02-24:
This will come when we merge the S3 support

closed 2022-03-28 by me-no-dev