espressif/arduino-esp32 #6205: Serial hangs up on ESP32 Arduino version 2.0.2

technoblogy, 2022-01-28:
### Board

Adafruit Feather ESP32

### Device Description

Plain board.

### Hardware Configuration

No.

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

macOS 10.13.6

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Echoing text in the Serial Monitor causes dropped characters. Run the following sketch, then cut and paste the following text into the Serial Monitor input field, and press Return:

$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ

The text is echoed as:

````text
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRS
````
(without the line break), so the last 8 characters have been dropped.

### Sketch

```cpp
void setup() {
  Serial.begin(9600);
}

void loop (void) {
  if (Serial.available()) Serial.print((char)Serial.read());
}
```

### Debug Message

```plain
No debug output.
```

### Other Steps to Reproduce

I have repeated the same test on ESP32 Arduino core version 1.0.6 and it works fine; in other words, it echoes:
````text
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
````
(without the line break).

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mstegen, 2022-01-30:
Have you tried to increase the receive buffer?
i.e.
`Serial.setRxBufferSize(1024);`

I've found the new uart implementation to be much more robust then the previous (1.0.6) version.
Especially when using all three uarts at the same time.

technoblogy, 2022-01-30:
@mstegen thanks for the suggestion.

No, that doesn't seem to have any effect, either on my simple test above, or on my real application (uploading programs to uLisp running on ESP boards).

SuGlider, 2022-01-30:
@technoblogy 
Indeed this is a bug with V2.0.2.
Please try using the latest version in Github (git checkout master)

This issue has been fixed in PR #6133
You can also apply this fix to your local Arduino folder to test it. 

Please let me know.
Thanks.

technoblogy, 2022-01-30:
@SuGlider Thank you for your reply. I confirm that applying that fix solves the problem on the following boards I've tested Thanks!:

Adafruit ESP32 Feather
DOIT ESP32 DEVKIT v1
ESP32-C3 DevKit

However I'm still unable to get Serial to work reliably on ESP32-S2 boards. Should this also fix that case? Thanks!

SuGlider, 2022-01-30:
@technoblogy 
Yes, the lastest master version of this repository shall make all the SoCs Serial work very reliably.
I tested this specific issue with the ESP32-S2-Saola-1-v1.2 board and it works correctly.

In case you have found some other specific issue with HardwareSerial, please open a new issue and I'll analyze it.
Thanks!

closed 2022-01-31 by VojtechBartoska