espressif/arduino-esp32 #6209: entry 0x400805f0 E (134) cam_hal: cam_dma_config(280): frame buffer malloc failed E (134) cam_hal: cam_config(364): cam_dma_config failed

robinmollah, 2022-01-30:
### Board

ESP32-Cam

### Device Description

Burning ESP32 Cam module with Arduino on a breadboard

### Hardware Configuration

Arduino 5V - ESP32 5V
Arduino GND - ESP32 GND

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

MacOS

### Flash frequency

80 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Can't take photo. I can't initiate the camera.

### Sketch

```cpp
/*********
  Rui Santos
  Complete instructions at https://RandomNerdTutorials.com/esp32-cam-projects-ebook/
  
  Permission is hereby granted, free of charge, to any person obtaining a copy of this software and associated documentation files.
  The above copyright notice and this permission notice shall be included in all copies or substantial portions of the Software.
*********/

#include "esp_camera.h"
#include <WiFi.h>
#include "esp_timer.h"
#include "img_converters.h"
#include "Arduino.h"
#include "fb_gfx.h"
#include "soc/soc.h"             // disable brownout problems
#include "soc/rtc_cntl_reg.h"    // disable brownout problems
#include "esp_http_server.h"

// Replace with your network credentials
const char* ssid = "***";
const char* password = "**";

#define PART_BOUNDARY "123456789000000000000987654321"

#define CAMERA_MODEL_AI_THINKER
//#define CAMERA_MODEL_M5STACK_PSRAM
[... 414 lines cut ...]
```

### Debug Message

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1324
ho 0 tail 12 room 4
load:0x40078000,len:13508
load:0x40080400,len:3604
entry 0x400805f0
E (134) cam_hal: cam_dma_config(280): frame buffer malloc failed
E (134) cam_hal: cam_config(364): cam_dma_config failed
E (135) camera: Camera config failed with error 0xffffffff
Camera init failed with error 0xffffffff...........
WiFi connected
Camera Stream Ready! Go to: http://192.168.104.27
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-01-31:
Are you sure that PSRAM is enabled? Could you please change the debug level to "Debug" and post the output of the boot console again?

DmytroKorniienko, 2022-02-23:
@me-no-dev 
Same error for 2.0.2 at Platformio (on previous SDK work as expected)

```
[env:esp32cam_dirty]
framework = arduino
board_build.filesystem = littlefs

platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
platform_packages =
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.2

board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
;board_build.flash_mode = qio ; broken >>> boot loop
board_build.flash_mode = dio
board = esp32cam
...
```

```
14:43:04.211 > setup: free heap  : 262068
14:43:04.211 > setup: free PSRAM  : 4192139
...
14:43:04.578 > E (1254) cam_hal: cam_dma_config(280): frame buffer malloc failed
14:43:04.579 > E (1255) cam_hal: cam_config(364): cam_dma_config failed
14:43:04.580 > E (1255) camera: Camera config failed with error 0xffffffff
14:43:04.582 > Error initializing the camera
```

VojtechBartoska, 2022-04-11:
Hello, can you please retest this on v2.0.3-rc1?

Tony763, 2022-05-08:
Hi, for me, with 2.0.4 still not work.
```
E (211) cam_hal: cam_dma_config(292): frame buffer malloc failed
E (212) cam_hal: cam_config(376): cam_dma_config failed
E (212) camera: Camera config failed with error 0xffffffff
Camera init failed with error 0xffffffffCamera capture failed
```

me-no-dev, 2022-05-09:
The error suggests that you either have PSRAM not enabled or it is failing and does not function normally.

DmytroKorniienko, 2022-05-11:
@VojtechBartoska @me-no-dev
> any updates?

Fixed for me in 2.0.3, thank you.

youngda, 2022-05-18:
The same problem 2.0.3 can't be used, 1.0.6 is normal

Tony763, 2022-05-18:
Hello @youngda, do you have a PSRAM enabled? I use platform.io and project I wanted to build did not have it.

```
build_flags =
    -DBOARD_HAS_PSRAM
    -mfix-esp32-psram-cache-issue
```

With this, cam works again.

VojtechBartoska, 2022-05-18:
folks, is this still valid?

rottifant, 2022-05-28:
I got the same error with V2.0.3. (with Arduino IDE)
I enabled PSRAM and it works fine now.

DiegoEscobedo17, 2022-06-28:
Someone can help me ? 
This is my code:
#include <WebServer.h>
#include <WiFi.h>
#include <esp32cam.h>
 
const char* WIFI_SSID = "ESCOBEDO";
const char* WIFI_PASS = "diecelescobedo";

WebServer server(80);

static auto loRes = esp32cam::Resolution::find(320, 240);
static auto hiRes = esp32cam::Resolution::find(800, 600);
void 
handleBmp()
{
  if (!esp32cam::Camera.changeResolution(loRes)) {
    Serial.println("SET-LO-RES FAIL");
  }

  auto frame = esp32cam::capture();
  if (frame == nullptr) {
    Serial.println("CAPTURE FAIL");
    server.send(503, "", "");
    return;
  }
  Serial.printf("CAPTURE OK %dx%d %db\n", frame->getWidth(), frame->getHeight(),
                static_cast<int>(frame->size()));

  if (!frame->toBmp()) {
    Serial.println("CONVERT FAIL");
    server.send(503, "", "");
    return;
  }
  Serial.printf("CONVERT OK %dx%d %db\n", frame->getWidth(), frame->getHeight(),
                static_cast<int>(frame->size()));
  
  server.setContentLength(frame->size());
  server.send(200, "image/jpeg");
  WiFiClient client = server.client();
  frame->writeTo(client);
}

void
serveJpg()
{
  auto frame = esp32cam::capture();
  if (frame == nullptr) {
    Serial.println("CAPTURE FAIL");
    server.send(503, "", "");
    return;
  }
  Serial.printf("CAPTURE OK %dx%d %db\n", frame->getWidth(), frame->getHeight(),
                static_cast<int>(frame->size()));
  
  server.setContentLength(frame->size());
  server.send(200, "image/jpeg");
  WiFiClient client = server.client();
  frame->writeTo(client);
}

void 
handleJpgLo()
{
  if (!esp32cam::Camera.changeResolution(loRes)) {
    Serial.println("SET-LO-RES FAIL");
  }
  serveJpg();
}
 
void 
handleJpgHi()
{
  if (!esp32cam::Camera.changeResolution(hiRes)) {
    Serial.println("SET-HI-RES FAIL");
  }
  serveJpg();
}
void
handleJpg()
{
  server.sendHeader("Location", "/cam-hi.jpg");
  server.send(302, "", "");
}
 
void 
handleMjpeg()
{
  if (!esp32cam::Camera.changeResolution(hiRes)) {
    Serial.println("SET-HI-RES FAIL");
  }
  
  Serial.println("STREAM BEGIN");
  WiFiClient client = server.client();
  auto startTime = millis();
  int res = esp32cam::Camera.streamMjpeg(client);
  if (res <= 0) {
    Serial.printf("STREAM ERROR %d\n", res);
    return;
  }
  auto duration = millis() - startTime;
  Serial.printf("STREAM END %dfrm %0.2ffps\n", res, 1000.0 * res / duration);
}
 
 
void  
setup(){
  Serial.begin(115200);
  Serial.println();
  {
    using namespace esp32cam;
    Config cfg;
    cfg.setPins(pins::AiThinker);
    cfg.setResolution(hiRes);
    cfg.setBufferCount(2);
    cfg.setJpeg(80);
 
    bool ok = Camera.begin(cfg);
    Serial.println(ok ? "CAMERA OK" : "CAMERA FAIL");
  }
  WiFi.persistent(false);
  WiFi.mode(WIFI_STA);
  WiFi.begin(WIFI_SSID, WIFI_PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
  Serial.print("http://");
  Serial.println(WiFi.localIP());
  Serial.println("  /cam.bmp");
  Serial.println("  /cam-lo.jpg");
  Serial.println("  /cam-hi.jpg");
  Serial.println("  /cam.mjpeg");
 
  server.on("/cam.bmp", handleBmp);
  server.on("/cam-lo.jpg", handleJpgLo);
  server.on("/cam-hi.jpg", handleJpgHi);
  server.on("/cam.jpg", handleJpg);
  server.on("/cam.mjpeg", handleMjpeg);
 
  server.begin();
}
 
void loop()
{
  server.handleClient();

AND THE ERROR IS THIS:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
entry 0x400805f8
E (197) psram: PSRAM ID read error: 0xffffffff

E (336) cam_hal: cam_dma_config(271): frame buffer malloc failed
E (336) cam_hal: cam_config(355): cam_dma_config failed
E (337) camera: Camera config failed with error 0xffffffff
CAMERA FAIL
http://192.168.1.15
  /cam.bmp
  /cam-lo.jpg
  /cam-hi.jpg
  /cam.mjpeg

please someone

youngda, 2022-06-28:
> Hello @youngda, do you have a PSRAM enabled? I use platform.io and project I wanted to build did not have it.
> 
> ```
> build_flags =
>     -DBOARD_HAS_PSRAM
>     -mfix-esp32-psram-cache-issue
> ```
> 
> With this, cam works again (2.0.4).

Did not find 2.0.4, tested version 2.0.3, in arduino ide, opened PSRAM, and a new error appeared, I went back to 1.0.6 and it works normally.

`
abort() was called at PC 0x40156e5e on core 1

Backtrace:0x400833c5:0x3ffb24800x4008d555:0x3ffb24a0 0x4009293d:0x3ffb24c0 0x40156e5e:0x3ffb2540 0x400d45af:0x3ffb25e0 0x400d464d:0x3ffb2800 0x400d62a9:0x3ffb2820 

ELF file SHA256: 0000000000000000`

Ryan38435435345353, 2022-06-28:
can someone help me with this error 

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
E (145) cam_hal: cam_dma_config(292): frame buffer malloc failed
E (146) cam_hal: cam_config(376): cam_dma_config failed
E (146) camera: Camera config failed with error 0xffffffff
.WiFi connected

192.168.86.67
Stream Link: rtsp://192.168.86.67:8554/mjpeg/1

My code

/****************************************************************************************************************************************************
 *  TITLE: HOW TO BUILD A $9 RSTP VIDEO STREAMER: Using The ESP-32 CAM Board || Arduino IDE - DIY #14
 *  DESCRIPTION: This sketch creates a video streamer than uses RTSP. You can configure it to either connect to an existing WiFi network or to create
 *  a new access point that you can connect to, in order to stream the video feed.
 *
 *  By Frenoy Osburn
 *  YouTube Video: https://youtu.be/1xZ-0UGiUsY
 *  BnBe Post: https://www.bitsnblobs.com/rtsp-video-streamer---esp32
 ****************************************************************************************************************************************************/

  /********************************************************************************************************************
 *  Board Settings:
 *  Board: "ESP32 Wrover Module"
 *  Upload Speed: "921600"
 *  Flash Frequency: "80MHz"
 *  Flash Mode: "QIO"
 *  Partition Scheme: "Hue APP (3MB No OTA/1MB SPIFFS)"
 *  Core Debug Level: "None"
 *  COM Port: Depends *On Your System*
 *********************************************************************************************************************/
 
#include "src/OV2640.h"
#include <WiFi.h>
#include <WebServer.h>
#include <WiFiClient.h>

#include "src/SimStreamer.h"
#include "src/OV2640Streamer.h"
#include "src/CRtspSession.h"

//#define ENABLE_OLED //if want use oled ,turn on thi macro
//#define SOFTAP_MODE // If you want to run our own softap turn this on
#define ENABLE_WEBSERVER
#define ENABLE_RTSPSERVER

#ifdef ENABLE_OLED
#include "SSD1306.h"
#define OLED_ADDRESS 0x3c
#define I2C_SDA 14
#define I2C_SCL 13
SSD1306Wire display(OLED_ADDRESS, I2C_SDA, I2C_SCL, GEOMETRY_128_32);
bool hasDisplay; // we probe for the device at runtime
#endif

// Select camera model
//#define CAMERA_MODEL_WROVER_KIT
//#define CAMERA_MODEL_ESP_EYE
//#define CAMERA_MODEL_M5STACK_PSRAM
//#define CAMERA_MODEL_M5STACK_WIDE
#define CAMERA_MODEL_AI_THINKER
[... 217 lines cut ...]
            // check if we are overrunning our max frame rate
            now = millis();
            if(now > lastimage + msecPerFrame)
                printf("warning exceeding max frame rate of %d ms\n", now - lastimage);
        }

        if(session->m_stopped) {
            delete session;
            delete streamer;
            session = NULL;
            streamer = NULL;
        }
    }
    else {
        client = rtspServer.accept();

        if(client) {
            //streamer = new SimStreamer(&client, true);             // our streamer for UDP/TCP based RTP transport
            streamer = new OV2640Streamer(&client, cam);             // our streamer for UDP/TCP based RTP transport

            session = new CRtspSession(&client, streamer); // our threads RTSP session and state
        }
    }
#endif
}

DiegoEscobedo17, 2022-06-28:
@wecane Hi, I already changed the board for the one you mentioned, but the error persists, I'm beginning to think that they are problems of frequency or upload speed.

[image]

Attached image of my settings, please please, can someone help me?

lbernstone, 2022-06-28:
Not a problem of frequency or upload speed.  The problem is exactly what the error message says.  Your psram is either not available, or is damaged.  Note that psram requires 2 pins, and that usually includes gpio16, which is exposed on many devices.

DiegoEscobedo17, 2022-06-28:
So, do you think it's a good idea to change my FTDI Module?

youngda, 2022-07-25:
Had the same problem with sdk 2.0.4, works fine in version 1.0.6,
`[10:33:29.340]收←◆Taking picture...

[10:33:30.240]收←◆ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
`

If add http upload：

`abort() was called at PC 0x4015afb2 on core 1

Backtrace:0x400838fd:0x3ffb24800x4008d5a1:0x3ffb24a0 0x40092a21:0x3ffb24c0 0x4015afb2:0x3ffb2540 0x400d4753:0x3ffb25e0 0x400d47f1:0x3ffb2800 0x400d6599:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0`

vlastahajek, 2022-08-11:
Those having older ESP32 camera devices without PSRAM: make sure you  set `fb_location`  to `CAMERA_FB_IN_DRAM`
 of `camera_config_t`, and `fb_count' to `1`.
 E.g. for M5Stack ESP32Cam:
 ```cpp
camera_config_t esp32cam_config{
    .pin_pwdn = -1, 
    .pin_reset = 15,
    .pin_xclk = 27,
    .pin_sscb_sda = 25,
    .pin_sscb_scl = 23,
    .pin_d7 = 19,
    .pin_d6 = 36,
    .pin_d5 = 18,
    .pin_d4 = 39,
    .pin_d3 = 5,
    .pin_d2 = 34,
    .pin_d1 = 35,
    .pin_d0 = 17,
    .pin_vsync = 22,
    .pin_href = 26,
    .pin_pclk = 21,
    .xclk_freq_hz = 20000000,
    .ledc_timer = LEDC_TIMER_0,
    .ledc_channel = LEDC_CHANNEL_0,
    .pixel_format = PIXFORMAT_JPEG,
    .frame_size = FRAMESIZE_SVGA,
    .jpeg_quality = 12, 
    .fb_count = 1,       
    .fb_location = CAMERA_FB_IN_DRAM
};
```
E.g. for [Micro-RTSP](https://github.com/geeksville/Micro-RTSP), edit `src/OV2640.cpp`

P-R-O-C-H-Y, 2022-11-15:
Hi folks, is this issue still valid or can be closed? Thanks

bigFin, 2023-08-22:
@P-R-O-C-H-Y 
I think so. I added the relevant build flags my branch of the ESP32-WROVER examples and will submit a pr on that side. 

https://github.com/bigFin/Freenove_ESP32_WROVER_Board

nhattran02, 2024-08-16:
> Are you sure that PSRAM is enabled? Could you please change the debug level to "Debug" and post the output of the boot console again?

How about if my custom board not support psram (ESP32S3 wroom 1 N16) ? Can I still use this camera library?

me-no-dev, 2024-08-16:
> Do you know if I can still use this camera library?

Only with JPEG and low resolutions