espressif/arduino-esp32 #6210: Bad alloc very rarely

collin80, 2022-01-31:
### Board

ESP32 WROOM32

### Device Description

I have a closed source project where I am controlling Tesla Model S battery modules. It uses an ESP32 as the processor. It talks TTL serial to the modules.

### Hardware Configuration

TTL Serial over Serial1 at 612500 baud to the modules which form a loop. TX goes to the first module and then loops until the last module returns data on RX.

### Version

v2.0.2

### IDE Name

Arduino IDE 1.8.19

### Operating System

Ubuntu 21.10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

After an indeterminate amount of time, but usually days (3.5 days the last time) my sketch will crash with a bad allocation assertion then die. The thing is, there is plenty of heap and stack and nothing has changed over those days. The whole thing works fine until one day it just gets a bad allocation when trying to set up memory to check for UDP frames (at least that's where it breaks in the attached back trace). I presume it could die at any random "new" call. But, heap diagnostics do not seem to show any good reason for this.

### Sketch

```cpp
Not at liberty to share the sketch but I will show the relevant output and the trace dump below
```

### Debug Message

```plain
Here is the backtrace:
Decoding 12 results
0x400839fd: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 402
0x4008d45d: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c line 121
0x40092841: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c line 46
0x401655e3: __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc line 47
0x4016562a: std::terminate() at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc line 57
0x40164d1b: __cxa_throw at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc line 95
0x4016502a: operator new(unsigned int) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/new_op.cc line 54
0x40164db1: operator new[](unsigned int) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/new_opv.cc line 32
0x400e09ab: WiFiUDP::parsePacket() at /home/collin/.arduino15/packages/esp32/hardware/esp32/2.0.2/libraries/WiFi/src/WiFiUdp.cpp line 210
0x400dd40b: checkforUDPCAN() at /home/collin/projects/EVTV_Proprietary/TeslaBMS_EVTV/wifi_handler.cpp line 399
0x400d71a0: loop() at /home/collin/projects/EVTV_Proprietary/TeslaBMS_EVTV/TeslaBMS_EVTV.ino line 552
0x400e7c01: loopTask(void*) at /home/collin/.arduino15/packages/esp32/hardware/esp32/2.0.2/cores/esp32/main.cpp line 50

You probably think I must have run out of memory. Doesn't look like it. Here's the output of debugging that was just a short time before this crash happened:

Heap summary for capabilities 0x00001000:
  At 0x3ffb8000 len 6688 free 112 allocated 4052 min_free 112
    largest_free_block 0 alloc_blocks 30 free_blocks 0 total_blocks 30
  At 0x3ffb0000 len 25480 free 104 allocated 22924 min_free 88
    largest_free_block 0 alloc_blocks 24 free_blocks 1 total_blocks 25
  At 0x3ffae6e0 len 6192 free 60 allocated 3764 min_free 60
    largest_free_block 0 alloc_blocks 17 free_blocks 0 total_blocks 17
  At 0x3ffb6388 len 7288 free 216 allocated 4284 min_free 104
    largest_free_block 0 alloc_blocks 52 free_blocks 1 total_blocks 53
  At 0x3ffb9a20 len 16648 free 96 allocated 14076 min_free 96
    largest_free_block 0 alloc_blocks 26 free_blocks 0 total_blocks 26
  At 0x3ffcc530 len 80592 free 1052 allocated 74400 min_free 876
    largest_free_block 0 alloc_blocks 248 free_blocks 4 total_blocks 252
  At 0x3ffe0440 len 15072 free 2868 allocated 9200 min_free 228
    largest_free_block 1044 alloc_blocks 70 free_blocks 13 total_blocks 83
  At 0x3ffe4350 len 113840 free 95724 allocated 15880 min_free 51692
    largest_free_block 62452 alloc_blocks 6 free_blocks 3 total_blocks 9
  Totals:
    free 100232 allocated 148580 min_free 53256 largest_free_block 62452

Loop times   Big: 17169398  Small: 256 NBD: 6741 LowHeap: 53256
Stack high watermark: 4100

This seems to show plenty of memory available for allocation. But, still the backtrace clearly indicates that there was a BAD_ALLOC throw. This doesn't make sense to me.
```

### Other Steps to Reproduce

Wait potentially several days of uptime.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-01-31:
In the example you provided, the crash does indeed happen in the UDP library:
`WiFiUDP::parsePacket() at /home/collin/.arduino15/packages/esp32/hardware/esp32/2.0.2/libraries/WiFi/src/WiFiUdp.cpp line 210`

Looking at that line in the source WiFiUdp.cpp:
`char * buf = new char[1460];`

Is it possible there isn't a contiguous 1460 bytes available on the heap?  I highly doubt the "new" operator in Xtensa is to blame here.

mrengineer7777, 2022-01-31:
In my project I use a Ticker to periodically (every 3 days) reboot the module. If that is an option, I'm happy to provide code.

collin80, 2022-02-02:
Well, in the heap dump it shows this output:
  At 0x3ffe4350 len 113840 free 95724 allocated 15880 min_free 51692
    largest_free_block 62452 alloc_blocks 6 free_blocks 3 total_blocks 9

Since the UDP code isn't asking for any special sort of memory I assume this big chunk could be used? In that case the largest free block is 62452 bytes which is plenty for a 1.4k allocation. It shows free blocks too. So, I still don't quite understand why it cannot allocate the memory when it appears that the heap is fine. Now, it is possible that the situation changed between when the heap diagnostics were sent and when the crash happened. I was outputting diagnostics every 10 seconds so there could have been as much as 9.999 seconds between the time the heap data was output and when the UDP call was executed. But, the function chain shown looks correct. So, I'm still at a loss here. I do sort of suspect heap corruption. I suppose it's possible that over time some piece of code writes past a buffer and slowly the heap is corrupted until it blows up. The heap diagnostics don't seem to support this and it is not faulting with heap corruption. 

I do agree, it seems *VERY* unlikely to be a problem deep in the standard library. I have to assume that code has been used and abused for a very long time by a very large number of people. So, I'd suspect it's fairly error free. But, it sure seems to be coming down to a bad allocation for *some* reason. One thing I have to wonder is why doesn't the new call just fail? All the code checks for a null reply! Couldn't it just return a nullptr and more gracefully handle this condition than to send an exception?! I guess I could trap the exception but it seems like someone used the "shotgun to fix a hangnail" approach here.

collin80, 2022-02-02:
I found this exact issue discussed here: https://github.com/luc-github/ESP32SSDP/issues/26

It seems to exactly what is happening to me on exactly the same line of code in WifiUdp. So, presumably for whatever reason, at least temporarily the needed RAM is being used up and the call can fail but, at least for the issue I found, it worked for them most of the time and retrying worked thereafter. That seems to match up with my experience of having random uptime with mystery faults that drop the program out.  I will investigate AsyncUDP or otherwise do what they suggest (and I mentioned above) and just trap the bad_alloc exception, ignoring it and retrying later.

closed 2022-02-02 by collin80

mrengineer7777, 2022-02-05:
@collin80 Nice find.  The try/catch is a good workaround for the failed alloc.