espressif/arduino-esp32 #6214: Arduino IDE 1.8.19 The system cannot find the path specified.

xenois, 2022-01-31:
### Board

any

### Device Description

none

### Hardware Configuration

none

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

none

### PSRAM enabled

no

### Upload speed

none

### Description

Compiling anything for esp32 gives me the error: 
```

The system cannot find the path specified.

exit status 1

Error compiling for board ESP32 Dev Module.
```
Everything is fresh installed.

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:
}
```

### Debug Message

```plain
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\(user)\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\(user)\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\(user)\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none -ide-version=10819 -build-path C:\Users\(user)\AppData\Local\Temp\arduino_build_547404 -warnings=none -build-cache C:\Users\(user)\AppData\Local\Temp\arduino_cache_534413 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-gcc8_4_0-esp-2021r2.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2 -prefs=runtime.tools.mkspiffs.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2 -prefs=runtime.tools.xtensa-esp32-elf-gcc-gcc8_4_0-esp-2021r2.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2 -prefs=runtime.tools.riscv32-esp-elf-gcc-gcc8_4_0-esp-2021r2.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2 -prefs=runtime.tools.esptool_py.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.1.0 -prefs=runtime.tools.esptool_py-3.1.0.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.1.0 -prefs=runtime.tools.mklittlefs.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -verbose C:\Users\(user)\AppData\Local\Temp\untitled1804568218.tmp\sketch_jan31a\sketch_jan31a.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\(user)\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\(user)\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\(user)\Documents\Arduino\libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none -ide-version=10819 -build-path C:\Users\(user)\AppData\Local\Temp\arduino_build_547404 -warnings=none -build-cache C:\Users\(user)\AppData\Local\Temp\arduino_cache_534413 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-gcc8_4_0-esp-2021r2.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32s2-elf-gcc\gcc8_4_0-esp-2021r2 -prefs=runtime.tools.mkspiffs.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2 -prefs=runtime.tools.xtensa-esp32-elf-gcc-gcc8_4_0-esp-2021r2.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2 -prefs=runtime.tools.riscv32-esp-elf-gcc-gcc8_4_0-esp-2021r2.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\riscv32-esp-elf-gcc\gcc8_4_0-esp-2021r2 -prefs=runtime.tools.esptool_py.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.1.0 -prefs=runtime.tools.esptool_py-3.1.0.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.1.0 -prefs=runtime.tools.mklittlefs.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\tools\mklittlefs\3.0.0-gnu12-dc7f933 -verbose C:\Users\(user)\AppData\Local\Temp\untitled1804568218.tmp\sketch_jan31a\sketch_jan31a.ino
Using board 'esp32' from platform in folder: C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1
Using core 'esp32' from platform in folder: C:\Users\(user)\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1
cmd /c if exist "C:\\Users\\(user)\\AppData\\Local\\Temp\\untitled1804568218.tmp\\sketch_jan31a\\partitions.csv" COPY /y "C:\\Users\\(user)\\AppData\\Local\\Temp\\untitled1804568218.tmp\\sketch_jan31a\\partitions.csv" "C:\\Users\\(user)\\AppData\\Local\\Temp\\arduino_build_547404\\partitions.csv"
The system cannot find the path specified.
exit status 1
Error compiling for board ESP32 Dev Module.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

atanisoft, 2022-01-31:
do you have a space in your username?

VojtechBartoska, 2022-01-31:
Hello @xenois, take a look on your path in your local machine to arduino esp-32 directory. (spaces and so on).

Seems like something simple, you can try to simply set up your environment again.

Follow the instructions here: https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/installing.html#installing-using-boards-manager

xenois, 2022-01-31:
> do you have a space in your username?

nope.

xenois, 2022-01-31:
> Hello @xenois, take a look on your path in your local machine to arduino esp-32 directory. (spaces and so on).
>
> Seems like something simple, you can try to simply set up your environment again. [...]

I tried with boards manager (stable and dev) also with manual git clone. The problem persists. The path doesnt have any empty spaces: `C:\Users\(username_without_space)\Documents\Arduino\hardware\espressif\esp32`

VojtechBartoska, 2022-01-31:
Path is wrong or some files are missing(were modified)....the path you shared in debug message is a bit weird. It must work if you go though Boards manager, just try e.g. only 'stable'  and see what's the result.

atanisoft, 2022-01-31:
> Can it be that Windows long path limit is interfering somehow?

That is quite possible, can you check if the path is longer than 255 characters (or bytes)?

xenois, 2022-01-31:
@atanisoft couldnt find any with more than 100 chars. I also changed at regedit the 'LongPathEnabled', still no difference.

xenois, 2022-01-31:
To any other unlucky fella that runs on this error too, the solution is: 
Open the `Registry Editor` and clear any values that are inside `HKEY_CURRENT_USER\Software\Microsoft\Command Processor\AutoRun`. 

Thanks anyway @atanisoft , @VojtechBartoska

closed 2022-01-31 by xenois

tolerious, 2024-06-17:
> To any other unlucky fella that runs on this error too, the solution is: Open the `Registry Editor` and clear any values that are inside `HKEY_CURRENT_USER\Software\Microsoft\Command Processor\AutoRun`.
> 
> Thanks anyway @atanisoft , @VojtechBartoska

@xenois This is very, very, very helpful answer and feedback, thanks a lot.
I've been struggling with this error for two weeks, I installed a software call [clink](https://mridgers.github.io/clink/), this software will write an autorun item in the `Registry Editor`.

spazbutt, 2024-10-01:
Hi, I have this problem but my registry doesnt show any autorun under commandprocessor.. do I need this Clink?
[image]