espressif/arduino-esp32 #6217: HTTPClient reports wrong filesize

SvenArke, 2022-01-31:
### Board

ESP32 

### Device Description

ESP32 WROOM32

### Hardware Configuration

Standard Devkit V1

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

HTTPClient reports a wrong filesize. File size on Webserver is = 39043411347 bytes

HTTP Client reports as filesize = 2147483647 bytes ($07FFFFFFF)

I need to open very big data files for streaming the content over a 8bit bus in 64kb chunks.

### Sketch

```cpp
unsigned int dsize = http.getSize();
```

### Debug Message

```plain
21:29:44.480 -> [459434][D][HTTPClient.cpp:1141] connect():  connected to www.sx-64.de:80
21:29:45.678 -> [460661][D][HTTPClient.cpp:1275] handleHeaderResponse(): code: 200
21:29:45.678 -> [460661][D][HTTPClient.cpp:1278] handleHeaderResponse(): size: 2147483647
21:29:45.726 -> [460662][D][HTTPClient.cpp:619] sendRequest(): sendRequest code=200
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-01-31:
Are you saying you are downloading a file that is 37,124.7 MB (37GB)?

The issue is here (HTTPClient.cpp:1248-1250):
`                if(headerName.equalsIgnoreCase("Content-Length")) {
                    _size = headerValue.toInt();
                }`
Thus Content-Length is limited to a signed int: 2^31-1=0x7FFFFFFF
https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPClient/src/HTTPClient.cpp

Instead, consider: Omit Content-Length header and use chunked Transfer-Encoding.

VojtechBartoska, 2022-04-11:
Still relevant?

VojtechBartoska, 2022-05-04:
I'm closing the issue as expired due to no answer.

If needed, please reopen it.

Thanks for understanding.

closed 2022-05-04 by VojtechBartoska