espressif/arduino-esp32 #6221: Serial hangs up on ESP32-S2 after about 320 characters

technoblogy, 2022-02-01:
### Board

Adafruit QT-Py ESP32-S2

### Device Description

Plain board.

### Hardware Configuration

No.

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

macOS 10.13.6

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Entering text into the Serial Monitor causes the Serial to hang up after about 320 characters. Run the following sketch, then cut and paste all the following text (total 512 characters) into the Serial Monitor input field, and press Return:

$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ
$0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ

The Serial Monitor prints:

64
128
192
256
320

ie it hangs up after 320 characters and doesn't reach 512.

NB I've implemented the patch suggested by @SuGlider from PR #6133 in Issue #6205.

### Sketch

```cpp
void setup() {  
  Serial.begin(9600);
}

int Count = 0;

void loop (void) {
  if (Serial.available()) {
    Serial.read(); Count++;
    if (Count % 64 == 0) Serial.println(Count);
  }
}
```

### Debug Message

```plain
No debug output.
```

### Other Steps to Reproduce

Also tested on UM FeatherS2 - hangs up in the same way.

Works fine on ESP32-C3 DevKit and Adafruit ESP32 Feather.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-02-01:
@technoblogy 
I have tested it with ESP32, ESP32-C3 and ESP32-S2 (Saola) - using the latest Github version (git checkout master).
I've tested it with 300, 4800, 9600, 38400, 57600, 115200 and 230400 baud.
All of them reached 512.

Could you test it with the complete latest Arduino Core Version?

Please let me know.
Thanks!

### Results with ESP-ROM version information - using 115200 baud for reading the boot messages

```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x510
load:0x4004c000,len:0xa50
load:0x40050000,len:0x28bc
entry 0x4004c18c
64
128
192
256
320
384
448
512
```

technoblogy, 2022-02-02:
> Could you test it with the complete latest Arduino Core Version?

@SuGlider I'm not quite sure how to do this. If I download **arduino-esp32-master** it's not in a form that I can put in `/library/Arduino15/packages`. Can you point me to some instructions?

SuGlider, 2022-02-02:
@technoblogy 
You can add a new set of ESP boards from github.
[Follow this instructions](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html)

This is how it will look like:
[image]

You shall see ```ESP32 Arduino (in sketchbook)``` in the  Menu-->Tools-->Board.

technoblogy, 2022-02-03:
I installed from the development link:

`https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json`

Is that correct? It still seems to be version 2.0.2.

SuGlider, 2022-02-05:
@technoblogy 
The idea is to create a folder under you regular sketch Arduino folder called ```hardware```

These are the commands for Linux:
```
cd <Your Arduino Sketch folder>
mkdir hardware
cd hardware
mkdir espressif
cd espressif
git clone --recursive https://github.com/espressif/arduino-esp32 esp32
```

After cloning the master branch into ```esp32```, it is necessary to run ```esp32/tools/get.py``` (or ```get.exe``` for Windows).
Then restart your Arduino IDE and you will see the new board ```ESP32 Arduino (in sketchbook)``` in the Board Menu.
Open it and select your ESP32 board. This will compile to the lastest ESP32 GitHub version.

To update the GitHub in you computer, just go to the ```esp32``` folder in ```<Your Arduino Sketch folder>/hardware/espressif/esp32``` and run ```git pull```. No need to run ```get.py``` again - run it once, in the first time.

technoblogy, 2022-02-05:
Thanks - I'll try it.

technoblogy, 2022-02-05:
@SuGlider I followed your instructions which worked without error, but when I try to compile I get:

fork/exec /Users/david/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++: no such file or directory

I can see that file, but it's in the path:

/Users/david/Documents/Arduino/hardware/espressif/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++

SuGlider, 2022-02-05:
Did you also run ```get.py```?

technoblogy, 2022-02-05:
Yes, and it ran without errors.

me-no-dev, 2022-02-05:
ha! could you pose a log from get.py?

technoblogy, 2022-02-05:
Log from get.py:
````text
Phobos-309:espressif david$ esp32/tools/get.py
System: Darwin, Bits: 64, Info: Darwin-17.7.0-x86_64-i386-64bit
Platform: x86_64-apple-darwin
Downloading riscv32-esp-elf-gcc8_4_0-esp-2021r2-macos.tar.gz ...
Done
Extracting riscv32-esp-elf-gcc8_4_0-esp-2021r2-macos.tar.gz ...
Downloading xtensa-esp32-elf-gcc8_4_0-esp-2021r2-macos.tar.gz ...
Done
Extracting xtensa-esp32-elf-gcc8_4_0-esp-2021r2-macos.tar.gz ...
Downloading xtensa-esp32s2-elf-gcc8_4_0-esp-2021r2-macos.tar.gz ...
Done
Extracting xtensa-esp32s2-elf-gcc8_4_0-esp-2021r2-macos.tar.gz ...
Downloading esptool-3.1.0-macos.tar.gz ...
Done
Extracting esptool-3.1.0-macos.tar.gz ...
Downloading x86_64-apple-darwin14.mklittlefs-c41e51a.200706.tar.gz ...
Done
Extracting x86_64-apple-darwin14.mklittlefs-c41e51a.200706.tar.gz ...
Downloading mkspiffs-0.2.3-arduino-esp32-osx.tar.gz ...
Done
Extracting mkspiffs-0.2.3-arduino-esp32-osx.tar.gz ...
Renaming mkspiffs-0.2.3-arduino-esp32-osx to mkspiffs ...
Platform Tools Installed
Phobos-309:espressif david$ 
````

SuGlider, 2022-02-06:
Ok. It seems fine.
So the error you get is from Arduino IDE?
Can you post the complete log?

technoblogy, 2022-02-06:
Complete log:
```text
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/david/Library/Arduino15/packages -hardware /Users/david/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/david/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/david/Documents/Arduino/libraries -fqbn=espressif:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none -vid-pid=239A_80AB -ide-version=10813 -build-path /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_266758 -warnings=all -prefs=build.warn_data_percentage=75 -verbose /Users/david/Documents/Arduino/SerialTestESP32-S2/SerialTestESP32-S2.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/david/Library/Arduino15/packages -hardware /Users/david/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/david/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/david/Documents/Arduino/libraries -fqbn=espressif:esp32:esp32:PSRAM=disabled,PartitionScheme=default,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,LoopCore=1,EventsCore=1,DebugLevel=none -vid-pid=239A_80AB -ide-version=10813 -build-path /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_266758 -warnings=all -prefs=build.warn_data_percentage=75 -verbose /Users/david/Documents/Arduino/SerialTestESP32-S2/SerialTestESP32-S2.ino
Using board 'esp32' from platform in folder: /Users/david/Documents/Arduino/hardware/espressif/esp32
Using core 'esp32' from platform in folder: /Users/david/Documents/Arduino/hardware/espressif/esp32
bash -c "[ ! -f \"/Users/david/Documents/Arduino/SerialTestESP32-S2\"/partitions.csv ] || cp -f \"/Users/david/Documents/Arduino/SerialTestESP32-S2\"/partitions.csv \"/var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_266758\"/partitions.csv"
bash -c "[ -f \"/var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_266758\"/partitions.csv ] || [ ! -f \"/Users/david/Documents/Arduino/hardware/espressif/esp32/variants/esp32\"/partitions.csv ] || cp \"/Users/david/Documents/Arduino/hardware/espressif/esp32/variants/esp32\"/partitions.csv \"/var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_266758\"/partitions.csv"
bash -c "[ -f \"/var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_266758\"/partitions.csv ] || cp \"/Users/david/Documents/Arduino/hardware/espressif/esp32\"/tools/partitions/default.csv \"/var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_266758\"/partitions.csv"
bash -c "[ -f \"/Users/david/Documents/Arduino/SerialTestESP32-S2\"/bootloader.bin ] && cp -f \"/Users/david/Documents/Arduino/SerialTestESP32-S2\"/bootloader.bin \"/var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_266758\"/SerialTestESP32-S2.ino.bootloader.bin || ( [ -f \"/Users/david/Documents/Arduino/hardware/espressif/esp32/variants/esp32\"/bootloader.bin ] && cp \"/Users/david/Documents/Arduino/hardware/espressif/esp32/variants/esp32\"/bootloader.bin \"/var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_266758\"/SerialTestESP32-S2.ino.bootloader.bin || cp -f \"/Users/david/Documents/Arduino/hardware/espressif/esp32\"/tools/sdk/esp32/bin/bootloader_qio_80m.bin \"/var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_266758\"/SerialTestESP32-S2.ino.bootloader.bin )"
bash -c "[ ! -f \"/Users/david/Documents/Arduino/SerialTestESP32-S2\"/build_opt.h ] || cp -f \"/Users/david/Documents/Arduino/SerialTestESP32-S2\"/build_opt.h \"/var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_266758\"/build_opt.h"
bash -c "[ -f \"/var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_266758\"/build_opt.h ] || touch \"/var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_266758\"/build_opt.h"
Detecting libraries used...
/Users/david/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4-beta1-308-gf3e0c8bc41\"" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -I/Users/david/Documents/Arduino/hardware/espressif/esp32/tools/sdk/esp32/include/config -I/Users/david/Documents/Arduino/hardware/espressif/esp32/tools/sdk/esp32/include/newlib/platform_include -I/Users/david/Documents/Arduino/hardware/espressif/esp32/tools/sdk/esp32/include/freertos/include -I/Users/david/Documents/Arduino/hardware/espressif/esp32/tools/sdk/esp32/include/freertos/include/esp_additions/freertos -I/Users/david/Documents/Arduino/hardware/espressif/esp32/tools/sdk/esp32/include/freertos/port/xtensa/include -I/Users/david/Documents/Arduino/hardware/espressif/esp32/tools/sdk/esp32/include/freertos/include/esp_additions -I/Users/david/Documents/Arduino/hardware/espressif/esp32/tools/sdk/esp32/include/esp_hw_support/include -I/Users/david/Documents/Arduino/hardware/espressif/esp32/tools/s [... cut ...] ld_266758/build_opt.h -I/Users/david/Documents/Arduino/hardware/espressif/esp32/cores/esp32 -I/Users/david/Documents/Arduino/hardware/espressif/esp32/variants/esp32 /var/folders/fd/7rbn3jxx5j942sk5kn0vhb740000gn/T/arduino_build_266758/sketch/SerialTestESP32-S2.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
fork/exec /Users/david/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++: no such file or directory
Error compiling for board ESP32 Dev Module.
```

SuGlider, 2022-02-06:
Try reinstall Arduino Core 2.0.2.
I think that doing it may bring the environment to a better state.
Keep the GitHub installation while reinstalling the 2.0.2.
Then try to run get.py again.

I hope it may make it work.

technoblogy, 2022-02-06:
I haven't got Arduino Core 2.0.2, I'm on 1.8.13.

SuGlider, 2022-02-06:
I think that 1.8.13 is the version of the Arduino IDE.

ESP Arduino Core 2.0.2 is the version of the Espressif Arduino Framework. You will see it when installing ESP board in the IDE.

technoblogy, 2022-02-06:
OK, I had deleted my ESP32 2.0.2 installation. I've reinstalled it and then got an error about esptool.

I added an **esptool** folder containing **esptool** and it now compiles from **ESP32 Arduino (in sketchbook)**.

However I still get the same behaviour with the UM featherS2 board. In other words the Serial Monitor prints:

64
128
192
256
320

ie it hangs up after 320 characters and doesn't reach 512. Sorry :(

SuGlider, 2022-02-07:
Did you select the ESO board from ```ESP32 Arduino (in sketchbook)``` option in the Arduino IDE Menu-->Tools-->Board?

technoblogy, 2022-02-07:
No, I selected **UM FeatherS2**. I don't have an entry for **ESO**. What's ESO?

SuGlider, 2022-02-07:
Sorry, I meant ESP.

You must select your board or at least the ```ESP32 Devkit board``` in the new ```ESP32 Arduino (in sketchbook)``` selection in the IDE menu.

technoblogy, 2022-02-07:
I'm sorry I don't yet have an ESP32-S2 Devkit board (I've got one on order).

I tried with an UM Feather S2 board:
Selected **ESP32 Arduino (in sketchbook)** -> **UM FeatherS2**
With the test program above, Serial hangs in the same way.

I tried with an Adafruit QT Py ESP32-S2 board:
Selected **ESP32 Arduino (in sketchbook)** -> **Adafruit QT Py ESP32-S2**
Again, with the test program above, Serial hangs in the same way.

I tried with an Adafruit ESP32 Feather board:
Selected **ESP32 Arduino (in sketchbook)** -> **Adafruit ESP32 Feather**
The test program above completes successfully.

So the complete latest Arduino Core Version behaves just like the 2.0.2 release version: it works on ESP32 but has Serial problems for me on ESP32-S2.

SuGlider, 2022-02-07:
Ok.

My suggestion is to hold up to we launch Arduino 2.0.3 which will include all those fixes.

Then you can try the example again and test it.

In case the issue still occurs, please reopen this issue and mention my login, so I can look at it.

I'll close it in this meantime.

technoblogy, 2022-02-07:
@SuGlider - I was going to suggest the same. Thanks - I'll report back!

technoblogy, 2022-02-08:
OK, good news! This morning I just received the ESP32-S2 Dev Kit I ordered (ESP32-S2 Saola 1R Dev Kit featuring ESP32-S2 WROVER), and tried this as follows:

Selected **ESP32 Arduino (in sketchbook)** -> **ESP32S2 Dev Module**
and the test program above completes successfully.
I've also tried uploading substantial programs over Serial to my uLisp interpreter, and it all works fine.

I also went back and checked the 2.0.2 Release version with the ESP32-S2 Dev Kit board and same settings, and this fails.

Have you made changes to the Serial in 2.0.3 that might only be affecting the ESP32-S2 Dev Kit?

The other thing that occurs to me is that all the boards I've tried that failed have a USB-C connector, whereas the ESP32-S2 Dev Kit has a USB Micro connector. Could that be significant?

SuGlider, 2022-02-08:
> Have you made changes to the Serial in 2.0.3 that might only be affecting the ESP32-S2 Dev Kit?

No. All changes are in the PR that I mentioned earlier. This PR applies to all the Chips.

> The other thing that occurs to me is that all the ESP32-S2 boards that failed the test have a USB-C connector, whereas the ESP32-S2 Dev Kit has a USB Micro connector. Could that be significant?

I don't see a relationship with the type of USB connector. Maybe it has to do with the Serial-USB conversor chipset of the board.

Usually the Dev Kit board comes with the CP2102 chip. I'm not sure about the board with USB-C. You can check it out in the USB log (lsusb or dmesg on Linux).

technoblogy, 2022-02-08:
> Usually the Dev Kit board comes with the CP2102 chip.

Yes, I see a CP2102 chip on there, but not on any of the other boards, so presumably they are using the ESP32-S2 built-in USB support. So that must be where the problem is.

SuGlider, 2022-02-08:
> > Usually the Dev Kit board comes with the CP2102 chip.
> 
> Yes, I see a CP2102 chip on there, but not on any of the other boards, so presumably they are using the ESP32-S2 built-in USB support. So that must be where the problem is.

Yes, it is possible!
I'll verify it. Thanks!

SuGlider, 2022-02-08:
@me-no-dev 

In fact this issue occurs when using USB CDC port from ESP32-S2.
When using UART it works fine.

The problem is with CDC.

VojtechBartoska, 2022-02-09:
Assigning this one to @SuGlider and @me-no-dev will help with support and provide info how to test this.

Jasoroony, 2022-02-12:
https://github.com/espressif/arduino-esp32/issues/5727

https://github.com/espressif/arduino-esp32/issues/5727#issuecomment-950588395
> If you have CDC on boot enabled, `Serial.setRxBufferSize (2048);` in `setup()` will not work.

Basically, the fixed setRxBufferSize() limit is 256 bytes at the moment when using that build option, and Espressif didn't want to fix it at the time, so the issue was closed without being resolved.

Be warned though @technoblogy - there are other issues I've discovered with the ESP32-S2 USB peripheral when you have busy traffic going in both directions.  I had to abandon that project due to that (no big deal, just disappointing).

technoblogy, 2022-03-05:
Any progress on this? It's currently preventing me from adding support for the ESP32-S2 boards to my project.

me-no-dev, 2022-03-07:
I will look into this as soon as we merge S3 support into master.

technoblogy, 2022-03-07:
Great - thanks!

SuGlider, 2022-03-10:
@technoblogy @Jasoroony 

Please verify/review/test PR #6413 and let me know.
Thanks.

technoblogy, 2022-03-11:
It seems to work perfectly on the boards I've tested; details below:

Testing CDC USB:

Adafruit Feather ESP32-S2
USB CDC on Boot - Enabled
-> Serial works perfectly

Adafruit QT-PY ESP32-S2
USB CDC on Boot - Enabled
-> Serial works perfectly

Unexpected Maker UM FeatherS2
USB CDC on Boot - Enabled
-> Serial works perfectly

Also retested the CP2102-based boards (that worked OK before):

ESP32S2 Dev Module (ESP32-S2-Soala-1_V1.2 board with WROOM)
USB CDC on Boot - Disabled
-> Serial works perfectly

ESP32S2 Dev Module (ESP32-S2-Soala-1_V1.2 board with WROVER)
USB CDC on Boot - Disabled
-> Serial works perfectly

Thanks! - David

SuGlider, 2022-03-11:
Thanks David for testing all those boards!

technoblogy, 2022-03-28:
When do you plan to release this version?

Jasoroony, 2022-03-28:
This certainly fixes all the buffering issues I found before.  I do however still have the device lock up with this sketch:

```c

#define SZ 16

static char buf [SZ];

void setup () {
    delay (1500);

    Serial.setRxBufferSize (2048);
    Serial.begin (115200);
}

void loop () {
    char ch;
    int j;
    int k;
    int l;

    if (Serial.available () >= 1) {
        ch = Serial.read ();
        memset (buf, ch, SZ - 1);
        buf [SZ - 1] = '\n';
        k = 0;
        l = SZ;
        while (l) {
            j = Serial.write (&(buf [k]), l);
            if (j != l)
                yield ();
            if (j > 0) {
                l -= j;
                k += j;
            }
        }
    }
}
```

As before, on Windows 7 x64, if I only send in 64 bytes at a time I.E. one input packet, and I wait until I get back 16 * 64 bytes before sending more, then it runs well forever.  If I send in 128 bytes at a time, it locks up eventually, sometimes having transferred multiple megabytes of data.

Tracking this down, I got as far as TinyUSB setting the FiFos up, when a transmit has data, and TinyUSB next sets up a concurrent receive FiFo which the PC responds to first - at that point the transmit FiFo registers magically say it's empty - but no interrupt fires, and I don't believe the PC gets that packet.  Attempting to work around that, resetting flags and registers etc. it still doesn't work on the next send, it seems that FiFo is corrupt and needs a full reset.

It's meant to be able to handle far more than 128 bytes, but that's the trick, with TinyUSB telling the device to be ready for either request from the PC (and the PC wanting to send that extra 64 byte packet) the chip's hardware seems to get confused.

I'm sorry if you don't believe me (not you David).

SuGlider, 2022-03-28:
> When do you plan to release this version?

One of the pending issue for releasing v2.0.3-RC was the support to the S3.
It seems solved now, with the latest commit to master.

I think that  v2.0.3-RC may go out in about a week or so.

SuGlider, 2022-03-28:
> As before, on Windows 7 x64, if I only send in 64 bytes at a time I.E. one input packet, and I wait until I get back 16 * 64 bytes before sending more, then it runs well forever. If I send in 128 bytes at a time, it locks up eventually, sometimes having transferred multiple megabytes of data.

@Jasoroony 
I tested the suggested sketch with Win10 and I can't reproduce any issues with it.
This issue olny happens with Win7?

Jasoroony, 2022-03-29:
I have confirmed it does the same thing on Windows 10 x86 (I don't have a machine with Windows 10 x64 to test though)
[testusbd.zip](https://github.com/espressif/arduino-esp32/files/8368879/testusbd.zip)
This is my test .exe which when run, opens COM1: and shows it fail after transferring a few MB of data.  Try it again on a real serial port with the same sketch and it will run forever (unless you press a key to stop it).  COM1 is usually available to set for your device in device-manager (in the device-port_settings-advanced dialog).

SuGlider, 2022-03-29:
@Jasoroony 
UART and CDC are way different. Both deal with serial communication but they work in a very different way in the layers 1 and 2 (phy and link from OSI/ISO model).

ESP32-S2 CDC  applies [Bulk Transfers](https://www.beyondlogic.org/usbnutshell/usb4.shtml). It is based on stream pipes that sends blocks of 64 bytes in sequence. The time between transferring two consecutive blocks is about 1 ms, based on periodic USB transactions.  Sending one character within 1 ms, the USB will transfer one byte in one block. 

USB Host (ESP32-S2) will receive these packet blocks and it will try to add them to the RX Queue (buffer) for whenever the application needs to read and process the data.

Your example uses a buffer of 2048 bytes. If it is not consumed fast enough, your sketch will end up in a Buffer overfow.
If it happens, the sketch may go to a bad state where buffer data is not consumed and CDC will flush all incomming data (maybe, it isgiving you the impression that it is stuck?!).

Uers can monitor Overflow states by setting "Core Debug Level: Verbose" in the IDE Menu. 
The core sends messages when it flushes incomming data that can't be placed in the RX Buffer.

USB CDC has not Flow Control - big difference with UART!
Thus, another solution would be to implement some sort of ACK/NACK for both side (Win7 and ESP32-S2) in your own application. It may implement a way to halt the transmission while no ACK comes from ESP32...

I hope it can help you to get your application working the way you need it.

Jasoroony, 2022-03-29:
I don't use 2048 bytes for CDC input in this example, I send 128 bytes and I wait for 16 * 128 (2048) bytes to come back, where I've increased the PC buffer to 4096 bytes.  This test case has worked on every other device (CDC or UART) I've tried it on and some of them are transferring data much faster than the ESP32-S2 without any issue (Pi4 in gadget mode).

Besides, If you ran my example and saw the error, then try opening an Arduino terminal window to COM1 after it happens and you terminate my testusbd.exe - if what you say is correct, you will be able to continue sending in characters and seeing 16 times the output without resetting the ESP32-S2...  But no, the device has locked up and you no longer get any output from it.

Jasoroony, 2022-03-29:
I should mention, the 4096 byte PC buffer is a device driver IO control setting I make, not my actual test code.

SuGlider, 2022-03-30:
Ok, I'll investigate it further.

SuGlider, 2022-04-01:
@Jasoroony 

I confirm that this issue you described with the sketch found at https://github.com/espressif/arduino-esp32/issues/6221#issuecomment-1080489314 happens with Arduino Core 2.0.2.

We have released a new Arduino Core 2.0.3-RC1 that has TinyUSB bug fixes.
I tested the sketch with this new version and it seems solved.

Please try it with the version 2.0.3-RC1 and let me know.
Thanks!

Jasoroony, 2022-04-02:
@SuGlider 

Maybe I'm doing something wrong, but the sketch still hangs for me with 2.0.3-RC1 (after a few MB being transferred).

Jasoroony, 2022-04-04:
@VojtechBartoska 

What response is being waiting for?  There's apparently some discrepancy betweem SuGlider's testing and mine, but mine certainly didn't work with 2.0.3-RC1.  I also checked various repositories (TinyUSB [main] and Espressif's/TinyUSB) and I don't see any recent bug fixes in either.  There was something about build issues there though, back on the 4th of March:

https://github.com/hathach/tinyusb/pull/1370
(fix build with esp32s2, also use dwc2 for esp example)

If you are just looking to close this issue, that's fine, I only came here to warn David (the blogger) that I don't use these devices because I found them unreliable (you can't be sure in many use-cases if it will lock-up or not).  If he can be sure that the device will only transmit when commanded (like the mass storage sketch) then it should be fine, and he can blog away.  Anything else like unsolicited GPS data being transmitted each second, and you risk the scenario in my example sketch here where it will randomly lock up due to a concurrent/bidirectional data clash.  Up to him I guess.

technoblogy, 2022-04-04:
I'm only testing my use case, which is uploading a long Lisp program to the ESP32-S2 over Serial, and as far as I can see that now works OK, so I'm happy!

Jasoroony, 2022-04-04:
Then I'll formally bow out.

SuGlider, 2022-04-04:
@Jasoroony 

I'm still testing your sketch in a more deeply way, separating TX and RX using CDC and UART for each direction, in order to check in which side is the issue.

I won't use the EXE file you sent, but instead I have created a Python Script to be used instead, and then to test it isolating CDC Tx and Rx.

I will publish the script here, so we can debate around it for building a common test case.

SuGlider, 2022-04-06:
@Jasoroony 

I have created a few test cases and some of the partial conclusions are:
1)  CDC ```write()``` can get stuck after a while, mainly when it is executed "repetitively too fast"
2) When CDC is only receiving data, it works fine, even in the "highest frequency transmission mode" - in which it will send "RX CDC FIFO Overflow" messages and then it will flush the data.
3) When CDC is only sending data, it works intermittentely and it can get stuck.

The CDC TX issue seems related to "TX FIFO full" that can't recover back and it won't send any data again.
Further testing and debugging is necessary, but it doesn't seem related to Arduino CDC layer, but actually to TinyUSB software layer.

SuGlider, 2022-04-06:
@Jasoroony 

I have an example, based on your sketch, that works fine and can send/receive more than 10 million bytes at a good transmission rate. This can be a work around example for your project:

I used a ESP32-S2 DevkitC-1 v1.0 board, that has both CDC and UART (CP2102) exposed USB ports.
In the menu I have enabled "**USB CDC On Boot**", therefore ```Serial``` will work with CDC and ```Serial0``` will work with UART. 

 ### ESP32-S2 Sketch with IDE Menu Option ```USB CDC On Boot: "Enabled"```
 
 ``` cpp
// 128 chars + '\0' at the end
static char buf [] = 
"1234567890abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ!#1234567890abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ!@";

void setup () {
  // CDC of ESP32-S2 with IDE option "USB CDC On Boot: Enabled"
  Serial.begin(115200);

  // UART for logging results
  Serial0.begin(115200);
  
  // Set, Ready, Go!
  delay(1000);
  Serial0.println("\n\nStarting... execute the Python script in 5 seconds...\n");
  delay(5000);
}

void loop() {
  static uint16_t indx = 0;
  static uint32_t round = 0;
  
  char outBuf[128];
  memset (outBuf, buf[indx], sizeof(outBuf));

  // writes bytes to CDC
  Serial.write(outBuf, sizeof(outBuf));

  // move to next char
  indx++;
  // do not send string terminator '\0'
  if (indx == sizeof(buf) - 1) {
    indx = 0;
  }

  memset(outBuf,'\0', sizeof(outBuf));
  
  uint16_t n_read = Serial.read(outBuf, sizeof(outBuf));
  // just to limit number of bytes printed to the SerialMonitor...
  
  outBuf[16] = '\0';

  //try to make it work... with no delay it gets stuck
  delay(10);

  // print loop number to UART
  Serial0.printf("#%d - %d bytes out - %d bytes read [%s]\n", ++round, round*sizeof(outBuf), n_read, outBuf);
}

```

### Windows 10 Python script that communicates back to the ESP32-S2
COM1 is used as the CDC Port - change it to your Win10 reported COM number port:

``` python
"""
	Python Script for testing ESP32 S2 Arduino issue #6221 
"""

import serial

[... 76 lines cut ...]
#1 - 128 bytes out - 0 bytes read []
#2 - 256 bytes out - 128 bytes read [1111111111111111]
#3 - 384 bytes out - 128 bytes read [2222222222222222]
#4 - 512 bytes out - 128 bytes read [3333333333333333]
#5 - 640 bytes out - 128 bytes read [4444444444444444]
#6 - 768 bytes out - 128 bytes read [5555555555555555]
#7 - 896 bytes out - 128 bytes read [6666666666666666]
#8 - 1024 bytes out - 128 bytes read [7777777777777777]
#9 - 1152 bytes out - 128 bytes read [8888888888888888]
#10 - 1280 bytes out - 128 bytes read [9999999999999999]
#11 - 1408 bytes out - 128 bytes read [0000000000000000]
#12 - 1536 bytes out - 128 bytes read [aaaaaaaaaaaaaaaa]
#13 - 1664 bytes out - 128 bytes read [bbbbbbbbbbbbbbbb]
#14 - 1792 bytes out - 128 bytes read [cccccccccccccccc]
#15 - 1920 bytes out - 128 bytes read [dddddddddddddddd]
#16 - 2048 bytes out - 128 bytes read [eeeeeeeeeeeeeeee]
#17 - 2176 bytes out - 128 bytes read [ffffffffffffffff]
#18 - 2304 bytes out - 128 bytes read [gggggggggggggggg]
#19 - 2432 bytes out - 128 bytes read [hhhhhhhhhhhhhhhh]
#20 - 2560 bytes out - 128 bytes read [iiiiiiiiiiiiiiii]
#21 - 2688 bytes out - 128 bytes read [jjjjjjjjjjjjjjjj]
#22 - 2816 bytes out - 128 bytes read [kkkkkkkkkkkkkkkk]

...
```

SuGlider, 2022-04-06:
@Jasoroony 

Sketch/Script used for testing CDC transmit to PC only - **about 4.5Mbps**
Both check if data sent is correct and in synch. Python script just displays number of 128 bytes chucks sent.
When there is a CDC TX Overflow the Python script will identify it and stop with a message.

#### Arduino Script - ESP32-S2
``` cpp
// 128 chars + '\0' at the end
static char buf [] = 
"1234567890abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ!#1234567890abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ!@";

void setup () {
  // CDC of ESP32-S2 with IDE option "USB CDC On Boot: Ebnabled"
  Serial.begin(2000000);  // for CDC it doesn't matter actually...

  // UART for logging results
  Serial0.begin(115200);
  
  // Set, Ready, Go!
  delay(1000);
  Serial0.println("\n\nStarting... execute the Python script in 5 seconds...\n");
  delay(5000);
}

// No output to make it faster!
void loop() {
  static uint16_t indx = 0;
  static uint32_t round = 0;
  
  char outBuf[128];
  memset (outBuf, buf[indx], sizeof(outBuf));

  // writes bytes to CDC
  Serial.write(outBuf, sizeof(outBuf));

  // move to next char
  indx++;
  // do not send string terminator '\0'
  if (indx == sizeof(buf) - 1) {
    indx = 0;
  }

```

#### Python Script - PC Win10
``` python
"""
	Python Script for testing ESP32 S2 Arduino issue #6221 
"""

import serial

print ("CDC App test for issue 6221")

try:
	# CDC baudrate: 500,000 - config as SERIAL_8N1 Arduino equivalent
	cdc = serial.Serial(port='com1', baudrate=500000, parity=serial.PARITY_NONE,
	stopbits=serial.STOPBITS_ONE, bytesize=serial.EIGHTBITS) #, timeout=0)
except:
	print("COM1: port is busy or unavailable")
	exit()

# Configure Serial In CDC with respective config in the sketch
S_IN = cdc

chars_128 = "1234567890abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ!#1234567890abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ!@"

round = 0
while 1:
	Bytes_read = S_IN.read(128)
	idx = round % 128
	if chr(Bytes_read[0]) != chars_128[idx] :
		print("Sequence error "+str(chr(Bytes_read[0]))+"<->"+str(chars_128[idx])+" while receiving data from ESP32-S2. Stopping. ["+str(Bytes_read)+"]")
		exit()
	round = round + 1
	print("#"+str(round))

```

Jasoroony, 2022-04-07:
> @Jasoroony
> 
> I have created a few test cases and some of the partial conclusions are:
> 
> 1. CDC `write()` can get stuck after a while, mainly when it is executed "repetitively too fast"
> 2. When CDC is only receiving data, it works fine, even in the "highest frequency transmission mode" - in which it will send "RX CDC FIFO Overflow" messages and then it will flush the data.
> 3. When CDC is only sending data, ~it works intermittentely and it can get stuck~. **It works fine.**
>    #### There was a mistake in my code. When only sending to the PC it reaches about 4.5Mbps - very fast!
> 
> The issue with CDC TX and RX in the same sketch seems related to "TX FIFO full" that can't recover back and it won't send any data again. Further testing and debugging is necessary, but it doesn't seem related to Arduino CDC layer, but actually to TinyUSB software layer.

That's also my conclusion.  It is in the TinyUSB layer too, I tested that with a sketch that only used "tud_cdc_n_available", "tud_cdc_n_read" & "tud_cdc_n_write".  However, I didn't see any issues in his software layer.  I got as far as the code in "dcd_esp32sx.c" (which is the hardware abstraction code specific for the ESP32-S2) which sets up the USB peripheral in the ESP32-S2 chip, I added some logging, and saw the full hardware Tx FiFo emptied without raising an interrupt (this happens when it's full with 64 bytes, and the TinyUSB library then sets up a concurrent Rx FiFo for 64 bytes which gets serviced first).  Since the interrupt didn't fire, TinyUSB doesn't clear the internal flag that says more can be transmitted.

Of course, I tried to code up some double-checking code in the Rx interrupt that tests if the Tx one has emptied - so I could reset the flags myself, which worked well, but on the next transmit attempt it still doesn't work properly - which is why I think that FiFo had become corrupt and needs a full reset.  I even tried a partial reset using the peripheral's registers, but that didn't work (obviously it could be made to work, but the documentation on the peripheral is very thin).

I spoke to hathach about it all, but he said there wan't much he could do either, because I'm using the library in this Arduino repo...  Along with me pointing to the hardware abstraction file, which he didn't write.  So I gave up.

SuGlider, 2022-04-07:
Thanks @Jasoroony for the information and your time debuging it.

There is an issue for sure. I'll try to go deeper into it and also to look for an USB expert to help on it

avillacis, 2022-04-13:
I am also affected by this bug when compiling sketches with v2.0.3RC1. Especially since I redirect the debugging messages to the USB CDC using `Serial.setDebugOutput(true);` at the start of my sketch.

However, I have tried the [MicroPython firmware for ESP32-S2](https://micropython.org/download/ESP32_S2_WROVER/) at version v1.18 (not the nightly build), and since the REPL runs through the USB CDC, I ran a loop to print a string one million times. It is still running at iteration 3xxxxx as I write these lines.

VojtechBartoska, 2022-04-14:
status update: We are now trying to figure out root of this issue with ESP-IDF team.

avillacis, 2022-05-06:
Problem still present when compiling sketch with recently released stable v2.0.3.

SuGlider, 2022-05-06:
Thanks @avillacis 

You are referring to the `Serial.setDebugOutput(true);` issue, right?

[64 more replies]

PR #6413 (2022-03-28): Fixes USB CDC setRxBufferSize(), begin(), _onRX()
## Summary
this PR fixes these issues:
1- ```USBCDC::setRxBufferSize()``` doesn't change RX Queue Size 
2- ```USBCDC::begin()``` always force 256 byte as RX Queue Size
3- ```USBCDC::_onRX()``` drops data when RX Queue Size is full
4- ```USBCDC::_onRX()``` implies in Arduino Task starvation when receiving more bytes than RX Queue can handle
5- ```USBCDC::end()``` resets the board when executed twice in sequence

## Impact
This PR allows ```USBCDC::setRxBufferSize()``` to be called either, before or after, ```USBCDC::begin()```.
It will work in a [... cut]