espressif/arduino-esp32 #6225: Add default 10MHz SD card frequency for SD SPI

grantwilk, 2022-02-03:
### Related area

SD SPI

### Hardware specification

ESP32

### Is your feature request related to a problem?

By default, ESP-IDF does not provide a default 10MHz SDMMC speed, even though there is a note about it in `sdmmc_common.c`. This means that using SD SPI with SD cards that do not support the 20MHz default speed requires additional steps and code modification to initialize.

```c
/* Find highest frequency in the following list,
     * which is below card->max_freq_khz.
     */
    const uint32_t freq_values[] = {
            SDMMC_FREQ_52M,
            SDMMC_FREQ_HIGHSPEED,
            SDMMC_FREQ_26M,
            SDMMC_FREQ_DEFAULT,
            //NOTE: in sdspi mode, 20MHz may not work. in that case, add 10MHz here.
    };
```

### Describe the solution you'd like

Please add a default `SDMMC_FREQ_10M` definition to `sdmmc_types.h` and add it to `freq_values` in `sdmmc_common.c`.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2022-02-04:
Thanks for contributing @grantwilk. Can you please help with triage, @P-R-O-C-H-Y ?

P-R-O-C-H-Y, 2022-03-23:
Hi @grantwilk,
This change needs to be made in ESP-IDF. I am in touch with them so once its available, I will let you know :)

VojtechBartoska, 2022-07-26:
**status update:** This feature is now in Review in ESP-IDF.

P-R-O-C-H-Y, 2023-08-16:
@grantwilk I have tested 10 Mhz SDMMC on `esp-idf-v5.1-libs` branch which uses IDF 5.1 and everything works :)
So I am closing this as it's already part of the `esp-idf-v5.1-libs` branch and will be in 3.0 version of Arduino-esp32.

closed 2023-08-16 by P-R-O-C-H-Y