espressif/arduino-esp32 #6227: fs file open for read|write

xinkiknix, 2022-02-03:
### Board

ESP32

### Device Description

ESP32 NodeMCU

### Hardware Configuration

na

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40

### PSRAM enabled

yes

### Upload speed

115200

### Description

I can not position in a file (_Seek_) and overwrite a portion. There is **no option** in FS for RW  (O_RDWR  or FREAD|FWRITE)
I can only open for FILE_READ, FILE_WRITE, FILE_APPEND.

### Sketch

```cpp
File file = fs.open(path, O_RDWR);
```

### Debug Message

```plain
\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2\xtensa-esp32-elf\sys-include\sys\_default_fcntl.h:50: note: it was later defined here
 #define O_RDWR  2  /* +1 == FREAD|FWRITE */
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-04-11:
Thanks for testing @xinkiknix, are you able to propose PR?

xinkiknix, 2022-04-11:
PR?

igrr, 2022-04-11:
> There is no option in FS for RW (O_RDWR or FREAD|FWRITE)

@xinkiknix When calling FS.open, you can pass any of the file open modes which you can normally pass to C `fopen` function. Please see its docs here: https://en.cppreference.com/w/c/io/fopen.
Depending on the desired behavior you can use "r+" or "a+" mode.

lbernstone, 2022-04-13:
Those defines are legacies from Arduino's SD.h.  ESP32 has POSIX compatibility under the covers, so provides emulated Arduino behaviors.

johnboiles, 2022-11-30:
I don't think it's safe to use `rw` in the Arduino APIs. For example in [this code](https://github.com/espressif/arduino-esp32/blob/d8f98450fc1f1e1646077260a22287df6332580a/libraries/FS/src/vfs_api.cpp#L62) `if((mode && mode[0] != 'r') && (!create)){` it will only create the file if the first character of the `const char *mode` is not `'r'`. So it will fail to create a file if the mode is specified as `'rw'`.

demosglok, 2023-01-15:
> I don't think it's safe to use `rw` in the Arduino APIs. For example in [this [...]

try "a+" and "w+". i don't know about "rw", only "a+", "r+", "w+" and I suppose "r+" shouldn't create file, while "w+" and "a+" should

johnboiles, 2023-01-15:
Ah yeah `rw` isn't a real mode I guess. I wanted `w+`. My bad!

Parsaabasi, 2025-01-08:
Hi everyone, 

Seems like this issue is resolved. I close it for now. 

In case the same issue occurs, or you would need to discuss it further, please reopen it and we will revisit it. 

Thanks

closed 2025-01-08 by Parsaabasi