espressif/arduino-esp32 #6236: ESP_ARDUINO_VERSION_PATCH has wrong value

scontini76, 2022-02-05:
### Board

all

### Device Description

N.A.

### Hardware Configuration

N.A.

### Version

v2.0.2

### IDE Name

all

### Operating System

all

### Flash frequency

N.A.

### PSRAM enabled

no

### Upload speed

N.A.

### Description

ESP_ARDUINO_VERSION_PATCH in [esp_arduino_version.h](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp_arduino_version.h) is not updated according to the patch release.

I think the ESP_ARDUINO_VERSION_* macros should always be updated before each release tag.

### Sketch

```cpp
N.A.
```

### Debug Message

```plain
N.A.
```

### Other Steps to Reproduce

N.A.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-02-05:
Will be fixed: https://github.com/espressif/arduino-esp32/pull/6072 :)

closed 2022-02-05 by me-no-dev

scontini76, 2022-02-05:
ops, sorry, I missed the already opened bug!