espressif/arduino-esp32 #6237: ESP32-C3 SD SPI mode, arduino example fails, IDF example works (compiled in an arduino project)

0x0fe, 2022-02-05:
### Board

ESP32-C3 dev module

### Device Description

ESP32-C3 dev module

### Hardware Configuration

Micro SDHC card 4GB, wired to  SCK:3, MOSI: 4, MISO:5, SD_CS:20

### Version

v2.0.2

### IDE Name

arduino IDE

### Operating System

windows 10

### Flash frequency

80M

### PSRAM enabled

no

### Upload speed

CDC

### Description

The Micro SD car socket is wired to GPIOs : SCK:3, MOSI: 4, MISO:5, SD_CS:20, 

I copied the IDF SPI SD card example into an arduino project, adjusted the GPIOs and it works perfectly, right away (after setting DMA to auto) at 20MHz and below.

But when i open the SD card SPI example provided in the Arduino IDE for ESP32-C3, set to the exact same pinout, on the exact same board / SD it fails miserably, with CRC errors and timeouts all over the place, regardless of the SPI bus speed (tested from 400K to 20M).

When i compare the traces with logic analyzer i cannot even compare, they are too dissimilar and the traces of the Arduino example are filled with 100ms timeouts. Basically all it can do is to read the card information, sector number, size, without error, but any write access leads to immediate failure.

### Sketch

```cpp
//**********IDE example********************
#include "FS.h"
#include "SD.h"
#include "SPI.h"

#define SCK         3
#define MOSI        4
#define MISO        5
#define SD_CS       20

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
  
    Serial.printf("Listing directory: %s\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("Failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println("Not a directory");
[... 390 lines cut ...]
E (10090) task_wdt: Tasks currently running:
E (10090) task_wdt: CPU 0: loopTask
System init 
E (11898) FGW: Initializing SD card
E (11898) FGW: Using SPI peripheral
Name: 00000
Type: SDHC/SDXC
Speed: 20 MHz
Size: 3748MB
E (11971) FGW: Opening file
E (12020) FGW: File written
E (12049) FGW: Renaming file
E (12061) FGW: Reading file
E (12064) FGW: Read from file: 'Hello 00000!'
E (12064) FGW: Card unmounted
```

### Other Steps to Reproduce

Well, to reproduce just wire a standard Micro SD to SCK:3, MOSI: 4, MISO:5, SD_CS:20 on a ESP32-C3 dev module or board and run each code, the IDE example fails, the IDF example works, which shows something is wrong in the implementation of the SD SPI on the C3.
It seems unlikely to be the SD driver itself since it has been proven. it works with the IDF example, which uses the same SD card driver, so maybe it has to do with the Arduino SPI driver / DMA? I compared quickly the SD.cpp from arduino with the SD driver of IDF and it seems identical for the SDCommand fonction.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-02-07:
can you try with GPIO under 18 for CS?

0x0fe, 2022-02-07:
no because the hardware is already done, so its in the copper. 
Also, how could it be related to the GPIO use when it works fine in the IDF on the same GPIO?
 its only the Arduino SPI/SD driver which has an issue, the IDF drivers are working as they should.

0x0fe, 2022-02-07:
good catch, i did not realized but that must be the problem as well (cf #6245)

P-R-O-C-H-Y, 2022-02-16:
@0x0fe I think we can close this as we did #6245 right?
Or do you still have some issues?