espressif/arduino-esp32 #6238: Serial monitor returning garbage

stupid-kid-af, 2022-02-06:
### Board

ESP32 Dev Module

### Device Description

ESP32 dev module 
Pin mode => 2

### Hardware Configuration

ESP32 dev module 
Pin mode => 2

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Arch Linux

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

9600

### Description

When I ran the program serial monitor thows garbage and even I set the baud rate to 115200 the problem remains same

### Sketch

```cpp
void setup() {

  Serial.begin(9600);
}

void loop() {
  Serial.print("Hello");   
  delay(1000);                       
}
```

### Debug Message

```plain
ۃ''⸮ ⸮⸮''⸮ ⸮⸮''⸮ ۃ''⸮ ⸮⸮'.⸮ ۃ''⸮ ⸮⸮'/⸮ ⸮⸮''⸮ ⸮⸮''⸮ ⸮⸮'g⸮ ⸮⸮''⸮ ⸮⸮'.⸮ ⸮⸮''⸮ ⸮⸮'/⸮ ⸮⸮''⸮ ۃ''⸮ ⸮⸮'.⸮ ۃ''⸮ ⸮⸮''⸮ ⸮⸮/'⸮`⸮⸮/'⸮ ⸮⸮'.⸮
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-02-07:
Did you connect to the ESP with the proper baudrate? There might be some garbage in the very beginning, because the bootloader outputs at 115200

stupid-kid-af, 2022-02-07:
> Did you connect to the ESP with the proper baudrate? There might be some garbage in the very beginning, because the bootloader outputs at 115200

It didn't work I set the baud rate to 115200 but isn't working

stupid-kid-af, 2022-02-07:
> 
Thanks but it's solved
I was not pressing the boot button after uploading.