espressif/arduino-esp32 #6242: IPv6 readiness and support "by default"

nuclearcat, 2022-02-07:
### Related area

tcp/ip stack

### Hardware specification

any

### Is your feature request related to a problem?

IPv6 support for IoT

### Describe the solution you'd like

I want to create issue where i can track esp32 ipv6 readiness, ask questions regarding code and propose patches.
First question i have:
libraries/WiFi/src/WiFiGeneric.cpp
function
esp_err_t WiFiGenericClass::_eventCallback(arduino_event_t *event)
event ARDUINO_EVENT_WIFI_STA_CONNECTED
any particular reason why this line commented?
`
        esp_netif_create_ip6_linklocal(esp_netifs[ESP_IF_WIFI_STA]);
`
Having it uncommented brings at least local IPv6 support.

I think if it might bring compatibility issues, i can send PR that will add some flag in wifi struct, which user can set to enable ipv6.
For user it will look something like:
<pre>
  Serial.begin(115200);
+ WiFi.IPV6(enable);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
</pre>

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

VojtechBartoska, 2022-04-11:
@nuclearcat Any updates?

nuclearcat, 2022-05-05:
Yes might work on it at free time, probably this week

VojtechBartoska, 2022-05-05:
thanks @nuclearcat!

nuclearcat, 2022-05-09:
I added series of patches in PR, that provide necessary support for IPv6, but it needs serious testing, especially as i dont have yet devboards on s2, s3 and c3.
Also i will test in more scenarios in upcoming days and probably will provide additional test/demo sketches.

nuclearcat, 2022-05-23:
Any feedback? :)

leonelsr, 2023-05-05:
So... Another year has passed. Any progress on IPv6 support for ESP32 on Arduino Environment?

nuclearcat, 2023-05-06:
Latest PR is https://github.com/espressif/arduino-esp32/pull/7722 , it's not mine anymore, but i believe can call myself as co-author and i'm totally happy with it (it is used also in Tasmota).
I suggest to support author and ping Espressif maintainers, as it's important matter to merge it.

anuragrao04, 2023-10-05:
What's the status of this PR? I have been trying to get an ipv6 address for my esp32 and it doesn't work. Here's the code:

```c
#include <WiFi.h>
#include <HTTPClient.h>

const char* ssid = "ssid";
const char* password = "intentionally_removed";

void setup() {
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.enableIpV6();

  WiFi.begin(ssid, password);

  Serial.println("\nConnecting to WiFi Network...");

  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(100);
  }
  
  Serial.println("\nConnected to the wifi network!");
}

void loop() {
  HTTPClient http;
  http.begin("http://api64.ipify.org");
  int httpCode = http.GET();

  if (httpCode > 0) {
    String payload = http.getString();
    Serial.println(payload);
  }
  http.end();
  delay(1000);
}
```

Am I missing something? [http://api64.ipify.org](http://api64.ipify.org) gives you an ipv6 address if you have one, else gives you an ipv4 address reply. 

When I run this code, I only get the ipv4 address and not the ipv6 one.

VojtechBartoska, 2024-02-20:
Hello,

this should be solved by this PR: https://github.com/espressif/arduino-esp32/pull/9016

Please give it a try, if needed you can reopen this ticket.

closed 2024-02-20 by VojtechBartoska