espressif/arduino-esp32 #6245: ESP32-C3, LOG outputs to both CDC and UART0 when CDC is selected

0x0fe, 2022-02-07:
### Board

ESP32-C3 dev module

### Device Description

esp32-c3 dev module

### Hardware Configuration

CDC

### Version

v2.0.2

### IDE Name

arduino IDE

### Operating System

windows 10

### Flash frequency

80M

### PSRAM enabled

no

### Upload speed

CDC

### Description

when CDC on boot is enabled all logging is expected to move on CDC, hwoever what ppaends is that the LOG ends up both on CDC and UART0, even tho the GPIOs of UART0 have been assigned to other functions in the setup, so of course this is a problem, in my case UART0 TX should be an interrupt input, but it continuously throws log. I tried various things such as commenting 
`CONFIG_ESP_CONSOLE_UART=y`
in sdkconfig, and also setting 
```
#define CONFIG_ESP_CONSOLE_UART_DEFAULT 0 
#define CONFIG_ESP_CONSOLE_UART 0 
```
in sdkconfig.h, and in the sketch before includes, 
```
#ifdef CORE_DEBUG_LEVEL
#undef CORE_DEBUG_LEVEL
#endif
#define CONFIG_CONSOLE_UART_NONE 1
#define CONFIG_ESP_CONSOLE_UART 0
```
to no avail.

### Sketch

```cpp
#ifdef CORE_DEBUG_LEVEL
#undef CORE_DEBUG_LEVEL
#endif
#define CONFIG_CONSOLE_UART_NONE 1
#define CONFIG_ESP_CONSOLE_UART 0

#define CORE_DEBUG_LEVEL 3
#define LOG_LOCAL_LEVEL ESP_LOG_DEBUG
#define SDMMC_FREQ_DEFAULT      2000

#include <stdio.h>
#include <string.h>
#include <sys/unistd.h>
#include <sys/stat.h>
#include "esp_err.h"
#include "esp_log.h"
#include "esp_vfs_fat.h"
#include "driver/sdspi_host.h"
#include "driver/spi_common.h"
#include "sdmmc_cmd.h"
#include "sdkconfig.h"

static const char *TAG = "FGW";

#define MOUNT_POINT "/sdcard"
#define USE_SPI_MODE

#define SPI_DMA_CHAN    SPI_DMA_CH_AUTO
#define PIN_NUM_MISO    (gpio_num_t)5
#define PIN_NUM_MOSI    (gpio_num_t)4
#define PIN_NUM_CLK     (gpio_num_t)3
#define PIN_NUM_CS      (gpio_num_t)20

#define ESP_LOGI       ESP_LOGE

void sd_test(void){
  
    esp_err_t ret;
    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
        .format_if_mount_failed = true,
        .max_files = 5,
        .allocation_unit_size = 16 * 1024
    };
    sdmmc_card_t* card;
    const char mount_point[] = MOUNT_POINT;
    ESP_LOGI(TAG, "Initializing SD card");

    ESP_LOGI(TAG, "Using SPI peripheral");
    sdmmc_host_t host = SDSPI_HOST_DEFAULT();
    spi_bus_config_t bus_cfg = {
        .mosi_io_num = PIN_NUM_MOSI,
        .miso_io_num = PIN_NUM_MISO,
        .sclk_io_num = PIN_NUM_CLK,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .max_transfer_sz = 4000,
    };
    ret = spi_bus_initialize((spi_host_device_t)host.slot, &bus_cfg, SPI_DMA_CHAN);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "Failed to initialize bus.");
        return;
    }

    // This initializes the slot without card detect (CD) and write protect (WP) signals.
    // Modify slot_config.gpio_cd and slot_config.gpio_wp if your board has these signals.
    sdspi_device_config_t slot_config = SDSPI_DEVICE_CONFIG_DEFAULT();
    slot_config.gpio_cs = PIN_NUM_CS;
    slot_config.host_id = (spi_host_device_t)host.slot;

    ret = esp_vfs_fat_sdspi_mount(mount_point, &host, &slot_config, &mount_config, &card);
    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {ESP_LOGE(TAG, "Failed to mount filesystem. ");} 
        else {ESP_LOGE(TAG, "Failed to initialize the card (%s). ", esp_err_to_name(ret));}
        return;
    }

    sdmmc_card_print_info(stdout, card);

    ESP_LOGI(TAG, "Opening file");
    FILE* f = fopen(MOUNT_POINT"/log.csv", "w");
    if (f == NULL) {
        ESP_LOGE(TAG, "Failed to open file for writing");
        return;
    }
    fprintf(f, "timestamp,%s\n", card->cid.name);
    fclose(f);
    ESP_LOGI(TAG, "File written");

    struct stat st;
    if (stat(MOUNT_POINT"/log2.csv", &st) == 0) {
        unlink(MOUNT_POINT"/log2.csv");
    }

    ESP_LOGI(TAG, "Renaming file");
    if (rename(MOUNT_POINT"/log2.csv", MOUNT_POINT"/log.csv") != 0) {
        ESP_LOGE(TAG, "Rename failed");
        return;
    }

    ESP_LOGI(TAG, "Reading file");
    f = fopen(MOUNT_POINT"/log.csv", "r");
    if (f == NULL) {
        ESP_LOGE(TAG, "Failed to open file for reading");
        return;
    }
    char line[64];
    fgets(line, sizeof(line), f);
    fclose(f);

    char* pos = strchr(line, '\n');
    if (pos) {
        *pos = '\0';
    }
    ESP_LOGI(TAG, "Read from file: '%s'", line);

    esp_vfs_fat_sdcard_unmount(mount_point, card);
    ESP_LOGI(TAG, "Card unmounted");
    spi_bus_free((spi_host_device_t)host.slot);
}
void setup() {
  
  pinMode(2, OUTPUT);
  digitalWrite(2, HIGH);
  Serial.begin(115200);
  while(!Serial); 
  delay(1800);
  Serial.println("System init ");
  sd_test();
}
void loop() {
  
}
```

### Debug Message

```plain
ESP-ROM:esp32c3-api1-20210207
System init 
E (3124) FGW: Initializing SD card
E (3124) FGW: Using SPI peripheral
Name: 00000
Type: SDHC/SDXC
Speed: 20 MHz
Size: 3748MB
E (3198) FGW: Opening file
E (3262) FGW: File written
E (3264) FGW: Renaming file
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-02-07:
nothing you change in sdkconfig would make a difference. The file is read-only and is there to show how ESP-IDF was configured in order to compile the libs for Arduino. I will suggest that you look into using Arduino as IDF component if you want to turn off UART0. Another thing you might want to do is to not use ESP-IDFs logging facility, but use the Arduino one instead: `ESP_LOGE(TAG,` becomes `log_e(`

me-no-dev, 2022-02-07:
oh... and... when you start serial, you should tell that you want debug to go there:
```cpp
Serial.begin(x);
Serial.setDebugOutput(true);
```

0x0fe, 2022-02-07:
> . I will suggest that you look into using Arduino as IDF component if you want to turn off UART0. Another thing you might want to do is to not use ESP-IDFs logging facility, but use the Arduino one instead:

Well, the only reason i end up with IDF code here is because the SD driver of the ESP32 core doesnt work in SPI (#6237) but yes, i agree that it is problematic to have idf logging clashing with ARduino serial HAL.

> Serial.begin(x);
> Serial.setDebugOutput(true);

i test that and see

0x0fe, 2022-02-07:
It did not help
[image]

me-no-dev, 2022-02-07:
so to recap:
- CDC On Boot is Enabled
- Serial (actually the CDC) is started with debug on
- You use Arduino's `log_x`
- You still get output from UART0 when you log

me-no-dev, 2022-02-07:
what does logging have to do with the SD driver you are using? Plus if you get this right, SD will start working :)

0x0fe, 2022-02-07:
I assuemd that using the IDF driver directly somehow implies the logging system will take over these two GPIOs with UART0.
I tested directly the Arduino SD example after adding the debug port redirection, i can confirm it does NOT work better

[image]

0x0fe, 2022-02-07:
[image]
[image]

me-no-dev, 2022-02-07:
are you showing IO20 (your SD SS) or 21? Signal looks good no?

0x0fe, 2022-02-07:
yes sorry, on this capture i connected GPIO20 (SD_CS) and did not update the label on the UI (so it shows 21 still).
The CS looks good, but it fails to read SD card and the overall SD transaction does not look good. BUt i think it is not related to UART0 / CDC.

0x0fe, 2022-02-07:
So, i did some test to confirm wether Arduino IDE behaves correctly regarding UART0 / CDC and here are the results. 
It turns out that println and log_e behave correctly, and printf and LOG_E redirect both to UART and CDC:

```
#define CORE_DEBUG_LEVEL 3
#define LOG_LOCAL_LEVEL ESP_LOG_DEBUG

#include "esp_log.h"

void setup() {

  Serial.begin(115200);
  Serial.setDebugOutput(true);
  while(!Serial); 
  delay(1500);
  Serial.println("println");
  printf("printf\n");
  log_e("log_e");
  ESP_LOGE("TEST", "LOGE");
}
void loop() {
}
```

[image]

P-R-O-C-H-Y, 2022-02-09:
> I assume that using the IDF driver for tje SD card somehow implies the IDF logging system will take over these two GPIOs with UART0. I tested the Arduino SPI SD examplewith:
> 
> `Serial.setDebugOutput(true);`
> 
> i can confirm it does NOT work better
> 
> [image]

This has been already fixed in master branch. Can you try using latest version (master branch pulled from git) of Arduino-esp32 instead of version 2.0.2?

P-R-O-C-H-Y, 2022-02-15:
@0x0fe 
So the issue you had is fixed now and this can be closed or you have still some issue?

0x0fe, 2022-02-15:
yes, further testing has confirmed the SD driver to be working on C3 in the main branch, i close this issue

closed 2022-02-15 by 0x0fe