espressif/arduino-esp32 #6246: idf_component.yml not conform with latest esp-idf

VOMediola, 2022-02-07:
### Board

doesn't matter

### Device Description

esp32

### Hardware Configuration

n/a

### Version

latest master

### IDE Name

esp-idf v4.4

### Operating System

linux

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

When calling menuconfig in project with arduino as component, the new file idf_component.yml throws an error

2 problems were found in the manifest file /home/test/blink/components/arduino/idf_component.yml:
Invalid manifest format
Unknown keys: tags
CMake Error at /home/test/tools/esp-idf-v4.4/tools/cmake/build.cmake:455 (message):
Call Stack (most recent call first):
  /home/test/tools/esp-idf-v4.4/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:8 (project)

When removing the file the menuconfig works

### Sketch

```cpp
none
```

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VOMediola, 2022-02-07:
@magicarm22 @kumekay I tag you as you are contributors of this file.

magicarm22, 2022-02-07:
@VOMediola What version of the idf-component-manager do you use? Did you update all python dependencies from ESP-IDF requirements.txt file?

VOMediola, 2022-02-07:
@magicarm22 I have
IDF component manager v1.0.1

I imagine requirment.txt is checked by python when I run install.sh?

kumekay, 2022-02-11:
@VOMediola @bbsan2k  Just a quick question, are you sure that you have installed in the IDF's virtual environment?

I.e. try to run (for linux/macos, for win export.ps1)

```
cd your/path/to/esp-idf
. ./export.sh
pip install idf-component-manager -U
python -m idf_component_manager -h
```
Unfortunately, ESP-IDF 4.4.0 was released with a slightly weaker requirement for component-manager, so install.sh doesn't enforce it. However, on clean installs, it should always download the latest version. 

Please let me know if you have the latest version of the component manager and problem is still present.

bbsan2k, 2022-02-11:
> @VOMediola @bbsan2k Just a quick question, are you sure that you have installed in the IDF's virtual environment?
>
> I.e. try to run (for linux/macos, for win export.ps1)
>
> ```
> cd your/path/to/esp-idf [...]

This actually fixed it. My component manager didn't have the right version as it seems, as it returned no version with the ```help```command.
Thanks!

closed 2022-04-11 by VojtechBartoska