espressif/arduino-esp32 #6247: Error: Expected ')' before numeric constant

VOMediola, 2022-02-07:
### Board

esp32

### Device Description

n/a

### Hardware Configuration

n/a

### Version

latest master

### IDE Name

esp-idf v4.4

### Operating System

linux

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

I have the same issue as described here https://github.com/Microsoft/azure-iot-developer-kit/issues/169 when trying to use Arduino core in https://github.com/project-chip/connectedhomeip/.

Solution is the same - replace 
`const IPAddress INADDR_NONE(0,0,0,0);`
with 
`const IPAddress IP_ADDR_NONE(0,0,0,0);`

in cores\arduino\IPAddress.h

### Sketch

```cpp
..
```

### Debug Message

```plain
..
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-03-04:
@VOMediola 
Could you please point to the application you have tried to build that caused such issue?
Are there instructions on how to build it?
I'll investigate this issue.
Thanks.

Sys64736, 2022-04-18:
I've also had this compile error before.   Changing the const IPAddress as stated above does get rid of the error and let's the compile finish. 

```
|-- <FS> 2.0.0
|-- <ESP8266 and ESP32 OLED driver for SSD1306 displays> 4.2.1
|   |-- <Wire> 2.0.0
|   |-- <SPI> 2.0.0
|-- <ArduinoJson> 6.19.3+sha.7abf875
|-- <DNSServer> 2.0.0
|   |-- <WiFi> 2.0.0
|-- <HTTPClient> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <WiFiClientSecure> 2.0.0
|   |   |-- <WiFi> 2.0.0
|-- <WebServer> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <FS> 2.0.0
|-- <WebSockets> 2.3.6+sha.28ed615
|   |-- <Ethernet> 2.0.0
|   |   |-- <WiFi> 2.0.0
|   |-- <SPI> 2.0.0
|   |-- <WiFi> 2.0.0
|   |-- <WiFiClientSecure> 2.0.0
|   |   |-- <WiFi> 2.0.0
|-- <WiFiClientSecure> 2.0.0
|   |-- <WiFi> 2.0.0
|-- <Wire> 2.0.0
Building in release mode
Compiling C:\.pio\build\esp32c3\src\OTA.cpp.o
Compiling C:\.pio\build\esp32c3\src\main.cpp.o
Generating partitions C:\.pio\build\esp32c3\partitions.bin
Compiling C:\.pio\build\esp32c3\libc0e\WiFi\WiFi.cpp.o
Compiling C:\.pio\build\esp32c3\libc0e\WiFi\WiFiAP.cpp.o
In file included from C:/Users/Mark/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/lwip/lwip/src/include/lwip/ip_addr.h:43,
                 from C:/Users/Mark/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/tcpip_adapter/include/tcpip_adapter_types.h:18,
                 from C:/Users/Mark/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/tcpip_adapter/include/tcpip_adapter.h:24,
                 from C:/Users/Mark/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/esp_netif/include/esp_netif.h:35,
                 from C:/Users/Mark/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/esp_event/include/esp_event_legacy.h:22,
                 from C:/Users/Mark/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/esp_event/include/esp_event.h:28,
                 from src/OTA.cpp:5:
C:/Users/Mark/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/lwip/lwip/src/include/lwip/ip4_addr.h:63:37: error: expected ')' before numeric constant        
 #define IPADDR_NONE         ((u32_t)0xffffffffUL)
                             ~       ^~~~~~~~~~~~
C:/Users/Mark/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/lwip/lwip/src/include/lwip/inet.h:71:29: note: in expansion of macro 'IPADDR_NONE'
 #define INADDR_NONE         IPADDR_NONE
                             ^~~~~~~~~~~
C:/Users/Mark/.platformio/packages/framework-arduinoespressif32/cores/esp32/IPAddress.h:94:17: note: in expansion of macro 'INADDR_NONE'
 const IPAddress INADDR_NONE(0, 0, 0, 0);
                 ^~~~~~~~~~~
Compiling C:\.pio\build\esp32c3\libc0e\WiFi\WiFiClient.cpp.o
Compiling C:\.pio\build\esp32c3\libc0e\WiFi\WiFiGeneric.cpp.o
Compiling C:\.pio\build\esp32c3\libc0e\WiFi\WiFiMulti.cpp.o
Compiling C:\.pio\build\esp32c3\libc0e\WiFi\WiFiSTA.cpp.o
*** [C:\.pio\build\esp32c3\src\OTA.cpp.o] Error 1
```
That's with: 2.02 and ½ ;)
HEAD is now at 3f79097d Add Preferences library API and tutorial documents (#6442)
Tool Manager: framework-arduinoespressif32 @ 0.0.0+sha.3f79097d has been installed!

SuGlider, 2022-04-21:
This issue is linked to the issue described in #6610

SuGlider, 2022-04-21:
We are currently working on a definitive solution.

SuGlider, 2022-04-22:
@Sys64736 @VOMediola 

The main issue is that <lwip/inet.h> declares #define INADDR_NONE IPADDR_NONE and it includes <lwip/ip4_addr.h> that declares #define IPADDR_NONE ((u32_t)0xffffffffUL) (IP 255.255.255.255)

When <lwip/inet.h> is included, it makes the Arduino declaration of "IPAddress INADDR_NONE(0,0,0,0)" invalid, causing a potential compilation error deending on the ```#include``` order, amonf other issues.

SuGlider, 2022-04-22:
We are currently working on a solution for it.

closed 2022-04-29 by me-no-dev

havrak, 2023-11-24:
Issue persist with idf 5.1.2 and 3.0.0 alpha2 build of Arduino esp. Declarations of INADDR_NONE between cores/esp32/IPAddress.h and components/lwip/lwip/src/include/lwip/inet.h are in conflict.

crossplatformdev, 2024-09-05:
This sketch produces the error.

```
#include <stdio.h>
#include <string.h>
#include <pthread.h>
#include <unistd.h>
#include <sys/socket.h>
#include <arpa/inet.h>
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "nvs_flash.h"
#include "driver/gpio.h"
#include "driver/adc.h"
#include "esp_adc/adc_continuous.h"
#include "lwip/sockets.h"
#include "esp_log.h"
#include "Arduino.h"
#include "soc/soc_caps.h"

#define CAPTURE_DEPTH 1472
#define CAPTURE_CHANNEL GPIO_NUM_35  // GPIO35, ADC1 channel 7
#define WIFI_SSID "sdr"
#define WIFI_PASS "sdr12345"
#define MAX_STORE_BUF_SIZE (CAPTURE_DEPTH * 2)

uint8_t data_dma;
bool data_val;
bool streaming;
adc_continuous_handle_t adc_handle;
static struct sockaddr_in dest_addr;
static int udp_sock;
static pthread_t adc_thread;
static pthread_t led_thread;
static timer_t led_timer;

static void udp_send_data(uint8_t *data, size_t len) {
    sendto(udp_sock, data, len, 0, (struct sockaddr *)&dest_addr, sizeof(dest_addr));
}

static bool adc_continuous_callback(adc_continuous_handle_t handle, const adc_continuous_evt_data_t *edata, void *user_data) {
    if (streaming && edata->size > 0) {
        udp_send_data(edata->conv_frame_buffer, edata->size);
        data_val = false;
    }
    return false;  // No need to wake any high priority task
}

static void* adc_continuous_task(void *arg) {
    uint8_t result[CAPTURE_DEPTH];
    uint32_t read_len = 0;
    while (1) {
        esp_err_t ret = adc_continuous_read(adc_handle, result, CAPTURE_DEPTH, &read_len, ADC_MAX_DELAY);
        if (ret == ESP_OK && streaming && read_len > 0) {
            udp_send_data(result, read_len);
            data_val = false;
        }
        usleep(10000);  // Sleep to prevent busy-waiting
    }
    return NULL;
}

static void start_stream() {
    if (streaming) return;

    data_dma = 0;
    data_val = false;
    streaming = true;

[... 162 lines cut ...]
    if ('s' == c) {
        start_stream();
    } else if ('x' == c) {
        stop_stream();
    }

    if('q' == c) {
        Serial.println("Quitting...");
        stop_stream();
        sleep(1);
        esp_wifi_disconnect();
        esp_wifi_stop();
        sleep(1);
        esp_wifi_deinit();
        sleep(1);
        if (adc_thread) {
            pthread_cancel(adc_thread);
            pthread_join(adc_thread, NULL);
        }
    }
}
```

The solution has been addressed here:
https://github.com/espressif/arduino-esp32/issues/4405

PR #6659 (2022-04-29): Fixes INADDR_NONE 
## Description of Change

Fixes ```IPAddress INADDR_NONE``` declaration when using Arduino WiFi or ETH.
This symbol was defined as ```0xffffffff``` by ```lwip /inet.h```, making it impossible to use INADDR_NONE correctly.

This PR only works when ```<wifi-provisioning/wifi_config.h>``` has a modification to include ```<lwip/ip4_addr.h>``` instead of ```<lwip/inet.h>```. This will be done directly to the sdk folder in the github structure and it has been fixed in IDF by a separated Merge Request. This will be reflected in the future, for good.

## Tests scenarios
This PR was  [... cut]
commit ce2cd111 (2022-04-29): Fixes INADDR_NONE  (#6659)

Description of Change

Fixes IPAddress INADDR_NONE declaration when using Arduino WiFi or ETH.
This symbol was defined as 0xffffffff by lwip /inet.h, making it impossible to use INADDR_NONE correctly.

This PR only works when <wifi-provisioning/wifi_config.h> has a modification to include <lwip/ip4_addr.h> instead of <lwip/inet.h>. This will be done directly to the sdk folder in the github structure and it has been fixed in IDF by a separated Merge Request. This will be reflected in the future, for good.

Tests scenarios

This PR was tested with all Arduino WiFi exa [... cut]