espressif/arduino-esp32 #6249: ESP32-S2 - can't toggle GPIO9 or GPIO11 after flashing / soft reset

calcut, 2022-02-08:
### Board

Adafruit Metro ESP32-S2

### Device Description

Adafruit Metro ESP32-S2

### Hardware Configuration

Just USB-C cable for programming and USB/UART cable on Serial0

### Version

v2.0.2

### IDE Name

Arduino 1.8.19

### Operating System

Macos 12.1

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

I try to set GPIOs 1 through 12 high during setup()

I find that GPIO 9 and 11 stay low after flashing (which does a rst:0xc (RTC_SW_CPU_RST))
I need to press the hardware reset button to get them to go high.

My guess is that this has something to do with the FSPI alternate functions of these pins.

Is there a workaround to avoid this behaviour, or is it somehow intentional?

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:
  pinMode(1, OUTPUT);
  pinMode(2, OUTPUT);
  pinMode(3, OUTPUT);
  pinMode(4, OUTPUT);
  pinMode(5, OUTPUT);
  pinMode(6, OUTPUT);
  pinMode(7, OUTPUT);
  pinMode(8, OUTPUT);
  pinMode(9, OUTPUT);
  pinMode(10, OUTPUT);
  pinMode(11, OUTPUT);
  pinMode(12, OUTPUT);
  digitalWrite(1, HIGH);
  digitalWrite(2, HIGH);
  digitalWrite(3, HIGH);
  digitalWrite(4, HIGH);
  digitalWrite(5, HIGH);
  digitalWrite(6, HIGH);
  digitalWrite(7, HIGH);
  digitalWrite(8, HIGH);
  digitalWrite(9, HIGH);
  digitalWrite(10, HIGH);
  digitalWrite(11, HIGH);
  digitalWrite(12, HIGH);
  
  
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

### Debug Message

```plain
waiting for download
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40028886
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x17c0
load:0x4004c000,len:0xfd4
load:0x40050000,len:0x2fa0
entry 0x4004c340
I (24) boot: ESP-IDF v4.3.1-480-g233dc30fb1 2nd stage bootloader
I (24) boot: compile time 10:09:11
I (24) boot: chip revision: 0
I (27) qio_mode: Enabling default flash chip QIO
I (33) boot.esp32s2: SPI Speed      : 80MHz
I (38) boot.esp32s2: SPI Mode       : QIO
I (42) boot.esp32s2: SPI Flash Size : 4MB
I (47) boot: Enabling RNG early entropy source...
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (71) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (78) boot:  2 ota_0            OTA app          00 10 00010000 00160000
I (86) boot:  3 ota_1            OTA app          00 11 00170000 00160000
I (93) boot:  4 uf2              factory app      00 00 002d0000 00040000
I (101) boot:  5 ffat             Unknown data     01 81 00310000 000f0000
I (108) boot: End of partition table
I (613) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=0a22ch ( 41516) map
I (620) esp_image: segment 1: paddr=0001a254 vaddr=3ffbeb80 size=01f94h (  8084) load
I (622) esp_image: segment 2: paddr=0001c1f0 vaddr=40024000 size=03e28h ( 15912) load
I (631) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=1ee68h (126568) map
I (657) esp_image: segment 4: paddr=0003ee90 vaddr=40027e28 size=06d54h ( 27988) load
I (664) esp_image: segment 5: paddr=00045bec vaddr=50000000 size=00010h (    16) load
I (669) boot: Loaded app from partition at offset 0x10000
I (669) boot: Disabling RNG early entropy source...

GPIO 9 and 11 are still low at this point, I then press the hardware reset...

ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x17c0
load:0x4004c000,len:0xfd4
load:0x40050000,len:0x2fa0
entry 0x4004c340
I (21) boot: ESP-IDF v4.3.1-480-g233dc30fb1 2nd stage bootloader
I (21) boot: compile time 10:09:11
I (21) boot: chip revision: 0
I (25) qio_mode: Enabling default flash chip QIO
I (30) boot.esp32s2: SPI Speed      : 80MHz
I (35) boot.esp32s2: SPI Mode       : QIO
I (40) boot.esp32s2: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (68) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (76) boot:  2 ota_0            OTA app          00 10 00010000 00160000
I (83) boot:  3 ota_1            OTA app          00 11 00170000 00160000
I (91) boot:  4 uf2              factory app      00 00 002d0000 00040000
I (98) boot:  5 ffat             Unknown data     01 81 00310000 000f0000
I (106) boot: End of partition table
I (610) esp_image: segment 0: paddr=00010020 vaddr=3f000020 size=0a22ch ( 41516) map
I (617) esp_image: segment 1: paddr=0001a254 vaddr=3ffbeb80 size=01f94h (  8084) load
I (619) esp_image: segment 2: paddr=0001c1f0 vaddr=40024000 size=03e28h ( 15912) load
I (628) esp_image: segment 3: paddr=00020020 vaddr=40080020 size=1ee68h (126568) map
I (654) esp_image: segment 4: paddr=0003ee90 vaddr=40027e28 size=06d54h ( 27988) load
I (661) esp_image: segment 5: paddr=00045bec vaddr=50000000 size=00010h (    16) load
I (666) boot: Loaded app from partition at offset 0x10000
I (666) boot: Disabling RNG early entropy source...

GPIO 9 and 11 are now high
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-02-08:
@P-R-O-C-H-Y Can you please help with this issue?

calcut, 2022-02-08:
I don't seem to see the same issue on a UM featherS2 board. Although that board uses the S2 chip directly instead of the WROVER module

P-R-O-C-H-Y, 2022-02-11:
Hi, linked your issue in PR that will close this. I have no issue with the refactored GPIO code.
Tested on ESP32S2 DevkitC with no issues. All pins are set High after flashing and after reset the same.

PR #6259 (2022-02-16): GPIO refactoring
## Summary
GPIO now using ESP-IDF API on all chips.
LEDC interrupt fix removed - no longer needed.
Edited pins_arduino.h in variants according to changes in gpio.

Known issue with GPIO interrupt on chip Esp32s3. Will be fixed in ESP-IDF.
For now GPIO interrupts work only on CORE 0.

## Impact
Close #6249
Close #6120

## Related links
closed 2022-02-16 by me-no-dev