espressif/arduino-esp32 #6254: OTA update error stream timeout

zark-zeugan, 2022-02-09:
### Board

ESP32 Dev Model

### Device Description

ESP32 with a attached SIM800L module.

### Hardware Configuration

CPU Freq: 240MHZ (WiFi/BT)
Flash Freq: 80MHz
Flash size: 4MB (32Mb)
Partition scheme: Default 4MB with spiffs
PSRAM: Disabled

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I am trying to use the on board GSM module to request a binary file from the firebase cloud storage to perfom an OTA update to my device. The size of the binary file is 599904bytes. I am able to receive this data. However, when I try to update using the command `size_t written = Update.writeStream(client);` I get the error `Error 6: UPDATE_ERROR_STREAM` which is a stream read timeout error. 
I am not sure what exactly to make out of this or how this could be solved. 

### Sketch

```cpp
#include <Wire.h>
#include <SPIFFS.h>
#include <ArduinoJson.h>
#include <Firebase_ESP_Client.h>
#include <ArduinoOTA.h>
#include <HTTPUpdate.h>
#include <Update.h>   

// ESP32 LilyGo-T-Call-SIM800 SIM800L_IP5306_VERSION_20190610 (v1.3) pins definition

#define MODEM_UART_BAUD 115200
#define MODEM_RST 5
#define MODEM_PWRKEY 4
#define MODEM_POWER_ON 23
#define MODEM_TX 27
#define MODEM_RX 26
#define I2C_SDA 21
#define I2C_SCL 22
#define LED_PIN 13
#define IP5306_ADDR 0x75
#define IP5306_REG_SYS_CTL0 0x00
#define TOUCHPIN 2
[... 379 lines cut ...]
Signal quality: 22
*************************

 Checking for new Updates ...

 Data Received - 
{"firmware":"/v0/b/engeletron-cloud.appspot.com/o/Teste%2Ffirmware.bin?alt=media&token=4a85d162-dea8-445a-82fe-81e99d9ee15f","version":"1.1"}
 
 Current version - 1.1
--- Firmware version not matching ---
Downloading binaries ...
Got application/octet-stream payload.
Got 599904bytes from server.
Content Length: 599904, isValidContentType: 1

############################################
Update may take 2-3 mins to complete...
############################################
>Written only :12694/599904. Retry?
Error Occured. Error #: 6
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-04-11:
Hello, can you please retest this on v2.0.3-rc1?

zark-zeugan, 2022-04-11:
> Hello, can you please retest this on v2.0.3-rc1?

It worked with an http server instead of an https one. 
What do you mean by v2.0.3-rc1?

VojtechBartoska, 2022-04-11:
ok, so you solved your issue?

2.0.3-rc1 is the latest release version of Arduino core ESP32: https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1

podaen, 2022-05-01:
I think it is related to my issue #6664, streams just breaks with the current version of WiFi

zark-zeugan, 2022-05-04:
> ok, so you solved your issue?
> 
> 2.0.3-rc1 is the latest release version of Arduino core ESP32: https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1

I did try with that but faced the same issues. However, I have resolved the issue now, had to use HTTP instead of HTTPS server. The system is still broken (the stream breaks) for downloading files from  HTTPS servers.

podaen, 2022-05-04:
In the mean time I have updated my sdk... And I'm sure now I doesn't run in to timeout, but it still breaks for me too. So no solution right now.

mudgalp, 2022-05-08:
> ### Board
> ESP32 Dev Model
> 
> ### Device Description
> ESP32 with a attached SIM800L module.
> 
> ### Hardware Configuration
> CPU Freq: 240MHZ (WiFi/BT) Flash Freq: 80MHz Flash size: 4MB (32Mb) Partition scheme: Default 4MB with spiffs PSRAM: Disabled
> 
> ### Version
> v2.0.2
> 
> ### IDE Name
> Arduino IDE
> 
> ### Operating System
> Windows 10
> 
> ### Flash frequency
> 80MHz
> 
> ### PSRAM enabled
> no
> 
> ### Upload speed
> 921600
> 
> ### Description
> I am trying to use the on board GSM module to request a binary file from the firebase cloud storage to perfom an OTA update to my device. The size of the binary file is 599904bytes. I am able to receive this data. However, when I try to update using the command `size_t written = Update.writeStream(client);` I get the error `Error 6: UPDATE_ERROR_STREAM` which is a stream read timeout error. I am not sure what exactly to make out of this or how this could be solved.
> 
> ### Sketch
> ```c++
> #include <Wire.h>
> #include <SPIFFS.h>
> #include <ArduinoJson.h>
> #include <Firebase_ESP_Client.h>
> #include <ArduinoOTA.h>
> #include <HTTPUpdate.h>
> #include <Update.h>   
> 
> // ESP32 LilyGo-T-Call-SIM800 SIM800L_IP5306_VERSION_20190610 (v1.3) pins definition
> 
> #define MODEM_UART_BAUD 115200
> #define MODEM_RST 5
> #define MODEM_PWRKEY 4
> #define MODEM_POWER_ON 23
> #define MODEM_TX 27
> #define MODEM_RX 26
> #define I2C_SDA 21
> #define I2C_SCL 22
> #define LED_PIN 13
> #define IP5306_ADDR 0x75
> #define IP5306_REG_SYS_CTL0 0x00
> #define TOUCHPIN 2
> 
> 
> // Set serial for debug console (to the Serial Monitor)
> #define SerialMon Serial
> // Set serial for AT commands (to the SIM800 module)
> #define SerialAT Serial1
> 
> // See all AT commands, if wanted
> #define DUMP_AT_COMMANDS
> 
> // Define the serial console for debug prints, if needed
> #define TINY_GSM_DEBUG SerialMon
> 
> 
> // See all AT commands, if wanted
> #define DUMP_AT_COMMANDS
[... 362 lines cut ...]
>  Data Received - 
> {"firmware":"/v0/b/engeletron-cloud.appspot.com/o/Teste%2Ffirmware.bin?alt=media&token=4a85d162-dea8-445a-82fe-81e99d9ee15f","version":"1.1"}
>  
>  Current version - 1.1
> --- Firmware version not matching ---
> Downloading binaries ...
> Got application/octet-stream payload.
> Got 599904bytes from server.
> Content Length: 599904, isValidContentType: 1
> 
> 
> ############################################
> Update may take 2-3 mins to complete...
> ############################################
> >Written only :12694/599904. Retry?
> Error Occured. Error #: 6
> ```
> 
> ### Other Steps to Reproduce
> _No response_
> 
> ### I have checked existing issues, online documentation and the Troubleshooting Guide
> * [x]  I confirm I have checked existing issues, online documentation and Troubleshooting guide.

i'm also facing the same issue with quectel's LTE module EC200S.

me-no-dev, 2022-05-09:
maybe there is a way to set the timeout of the client to something higher? GSM modems would be much slower than WiFi

mudgalp, 2022-05-11:
@zark-zeugan can you please help me in the same problem, i'm already using http server made by my team and LTE module EC200S from Quectel.

Thank you in advance.

VojtechBartoska, 2022-05-26:
Hello folks, do you still need help on this?

glen020, 2022-06-04:
I would like some help. Having the same issue. 
Downloading 768672 Bytes (~745KB)
It takes 959 seconds (~16 mins) to timeout.

podaen, 2022-06-04:
It seems that you don't empty the buffer engough. I remember doing it in 250ms for 1 mega. Never the less, it is still a hugh package size for a microcontroller to receive. Maybe you have to try with much less, and slitly increase it and see what it does.

glen020, 2022-06-08:
So I think it might be an Azure issue. Fails in an azure flask webpage BUT works in a "pythonanywhere.com" flask website. Will have to do more digging or hopefully someone here comes across something.

VojtechBartoska, 2022-10-10:
are there any updates on this guys? Thank you for your answers :)

zark-zeugan, 2022-10-23:
So, I have tried a lot of things on this and adding buffers or even breaking the data does not help much. The checksum in the end always does come errors. The only resolution I was able to find was to use a HTTP server like the AWS one to host my bin files on and was able to the system working for bigger file sizes of OTA Updates. 
I am not sure as to how much has the firmware of the system been developed to now. But as of now I have concluded that the system can only work with HTTP server and not with HTTPS.

closed 2022-10-23 by zark-zeugan

podaen, 2022-10-23:
Are you willing to try something?

hitecSmartHome, 2024-04-18:
I still got this error.