espressif/arduino-esp32 #6256: Absence Lilygo Esp32-s2 t8 1.1 (with and without sdCard)

limaomerces, 2022-02-09:
### Board

Lilygo Esp32-s2 t8 1.1

### Device Description

Lilygo Esp32-s2 t8 1.1 (with and without sdCard)

### Hardware Configuration

void

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

I do not know.

### PSRAM enabled

yes

### Upload speed

I do not know.

### Description

I would like to request that the model Lilygo T8 1.1 esp32-s2, could be incremented in the package.

Currently supported:

Esp32S2 Dev Module
ESP32S2 Native USB
TTGO Lora32 OLED
TTGO T1
TTGO T7 1.3 Mini 32
TTGO T7 1.4 Mini 32
TTGO T-Ol PLUS RISC-V Esp32-C3
SparkFun Esp32-s2 Thing Plus
Adafruit Metro Esp32-S2
Adafruit Feather Esp32-s2
Adafruit Feather Esp32-s2 TFT
Adafruit QT-py Esp32-s2
ATmegaZero Esp32-s2

But, it does not have the TTGO T8.
I bought from these links:
- https://pt.aliexpress.com/item/4001080714038.html? (with sdcard)
- https://pt.aliexpress.com/item/4001080875553.html? (without sdcard)

The developer has published a code, however, it is only for Esp-IDF (https://github.com/Xinyuan-LilyGO/LilyGo-esp32s2-base)

Would it be possible to support these models as well?

### Sketch

```cpp
void
```

### Debug Message

```plain
void
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-04-11:
Hello, are you able to send PR?

VojtechBartoska, 2022-04-13:
As we are not able to test and provide support for all third party boards, I'm adding label "Community help needed" so it means this issue can be only solved with Community help. In this case it's someone who has this board, can test it and provide support.

closed 2022-08-23 by VojtechBartoska