espressif/arduino-esp32 #6257: WiFiClientSecure connect memory leak

serpetti, 2022-02-10:
### Board

ESP32 Dev Module

### Device Description

Every time I create a connection with WiFiClientSecure library I looseabout 100B of heap memory. Everything works, I connect(), I get data and I stop(). But the during next connect() I loose other memory.
Thanks

### Hardware Configuration

esp32 dev kit 1 without PINs used

### Version

latest master

### IDE Name

Platform IO

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Memory leak

### Sketch

```cpp
wifiClientSecure = new WiFiClientSecure();

        wifiClientSecure->setCACert(root_ca);

        if (!wifiClientSecure->connect("www.hamqsl.com", 443))
            Serial.println("Connection failed!");
        else
        {
            wifiClientSecure->println("GET https://www.hamqsl.com/solarxml.php HTTP/1.0");
            wifiClientSecure->println("Host: www.hamqsl.com");
            wifiClientSecure->println("Connection: close");
            wifiClientSecure->println();
            
            while (wifiClientSecure->connected())
            {
                String line = wifiClientSecure->readStringUntil('\n');
                if (line == "\r")
                {
                    break;
                }
            }            
            
            int len=0;
            while (wifiClientSecure->available())
            {
                char c = wifiClientSecure->read();  
                xmlRaw += c;     
                len++;
            }
            
            wifiClientSecure->flush();
            wifiClientSecure->stop();                 
    
            Serial.Println(xmlRaw);
        }   
        if(wifiClientSecure) delete wifiClientSecure;
```

### Debug Message

```plain
No messages, only leak
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mrengineer7777, 2022-02-10:
Possibly related to PR #6062?

everslick, 2022-02-16:
are you sure, your socket is not in FIN_WAIT state? does the memory return after some time? like 3 minutes?

serpetti, 2022-02-16:
Hi Everslick and thank you for your answer. I've to check for socket status, but I'm sure that memory remains occupied even after dozens of minutes

VojtechBartoska, 2022-04-11:
Hello, can you please retest this on v2.0.3-rc1? Is it still relevant?

closed 2022-05-04 by VojtechBartoska