espressif/arduino-esp32 #6264: ESP32-C3 native USB CDC stops workng when Wifi is used

ladyada, 2022-02-12:
### Board

ESP32-C3 dev board

### Device Description

espressif ESP32-C3 devKitC-02-v1.1

### Hardware Configuration

raw USB cable on pins 18, 19, 5V and gnd

### Version

latest master

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

native

### Description

A plain blink/serial print sketch will print to the usb native CDC port just fine. 
Running anything with WiFi such as the Wifiscan example distributed with esp core, causes the USB CDC to disconnect and no longer enumerate until a hard reset
both blink/serial and wifiscan work ok if using the usb to serial CP2012n converter port.

### Sketch

```cpp
Working just fine either native or usb-serial port:

int led = 4;

void setup() {
  Serial.begin(115200);
  pinMode(led, OUTPUT);
}

int i=0;
void loop() {
  Serial.printf("hello %d!\n\r", i++);
  
  digitalWrite(led, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(100);               // wait for a second
  digitalWrite(led, LOW);    // turn the LED off by making the voltage LOW
  delay(100);               // wait for a second
}
```

Works only when using the cp2102n-usb-serial port. when enabling built in CDC on boot, USB port disappears once scan start:

```
/*
 *  This sketch demonstrates how to scan WiFi networks.
 *  The API is almost the same as with the WiFi Shield library,
 *  the most obvious difference being the different file you need to include:
 */
#include "WiFi.h"

void setup()
{
    Serial.begin(115200);

    // Set WiFi to station mode and disconnect from an AP if it was previously connected
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);

    Serial.println("Setup done");
}

void loop()
{
    Serial.println("scan start");

    // WiFi.scanNetworks will return the number of networks found
    int n = WiFi.scanNetworks();
    Serial.println("scan done");
    if (n == 0) {
        Serial.println("no networks found");
    } else {
        Serial.print(n);
        Serial.println(" networks found");
        for (int i = 0; i < n; ++i) {
            // Print SSID and RSSI for each network found
            Serial.print(i + 1);
            Serial.print(": ");
            Serial.print(WiFi.SSID(i));
            Serial.print(" (");
            Serial.print(WiFi.RSSI(i));
            Serial.print(")");
            Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN)?" ":"*");
            delay(10);
        }
    }
    Serial.println("");

    // Wait a bit before scanning again
    delay(5000);
}
```
```

### Debug Message

```plain
monitoring the uart output when USB CDC on Boot is enabled, shows the core is still running, just USB task no longer active:

23:00:27.558 -> ESP-ROM:esp32c3-api1-20210207
23:00:27.558 -> Build:Feb  7 2021
23:00:27.558 -> rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
23:00:27.558 -> SPIWP:0xee
23:00:27.558 -> mode:DIO, clock div:1
23:00:27.558 -> load:0x3fcd6100,len:0x38c
23:00:27.558 -> load:0x403ce000,len:0x6a8
23:00:27.558 -> load:0x403d0000,len:0x2370
23:00:27.558 -> SHA-256 comparison failed:
23:00:27.605 -> Calculated: 912c8ece44dc10c27ac7d0fbc21c881181838b5f87f42e05d6797a997e755775
23:00:27.605 -> Expected: e48257ed72b4ed90fbd1c7429f3866f76aab94952c8871b8e364f6e503cebde8
23:00:27.605 -> Attempting to boot anyway...
23:00:27.605 -> entry 0x403ce000
23:00:27.749 -> [   170][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 0 - WIFI_READY
23:00:27.749 -> [   209][V][WiFiGeneric.cpp:272] _arduino_event_cb(): STA Started
23:00:27.796 -> [   210][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 2 - STA_START
23:00:30.681 -> [  3135][V][WiFiGeneric.cpp:306] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 10
23:00:30.681 -> [  3136][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 1 - SCAN_DONE
...
```

last message on usb CDC before disappearing is
`23:04:22.557 -> [  5170][D][WiFiGeneric.cpp:831] _eventCallback(): Arduino Event: 0 - WIFI_READY`
```

### Other Steps to Reproduce

running micropython on an esp32-c3 with no cp2102n (only native usb cdc), wifi scan works just fine so it does not appear to be a hardware issue.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-02-12:
@Spritetm PTAL :)

dronecz, 2022-02-14:
I have same issue with support >2.0.0 . It is working in 2.0.0 . I tried WiFiScan & WiFiClientBasic sketch in Arduino 1.8.13 and 1.8.19  

Here is USB info when I´m not using WiFi: 
[image]

and here is when I´m using WiFi:
[image].

ladyada, 2022-02-14:
oh good idea, yes i can verify it works in 2.0.0

```
17:57:44.411 -> I (239) esp_image: segment 4: paddr=000b5844 vaddr=40389b98 size[  4282][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 0 - WIFI_READY
17:57:44.550 -> [  4396][V][WiFiGeneric.cpp:272] _arduino_event_cb(): STA Started
17:57:44.550 -> [  4397][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 2 - STA_START
17:57:44.644 -> Setup done
17:57:44.644 -> scan start
17:57:49.538 -> [  9410][V][WiFiGeneric.cpp:306] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 6
17:57:49.538 -> [  9410][D][WiFiGeneric.cpp:808] _eventCallback(): Arduino Event: 1 - SCAN_DONE
17:57:49.538 -> scan done
17:57:49.538 -> 6 networks found
...
```

me-no-dev, 2022-02-15:
We are looking into the issue. I hope to have results by tomorrow

Spritetm, 2022-02-15:
Looks like it's an ESP-IDF (4.4) issue. Long story short: ESP-IDF disables the USB PHY when WiFi is started if it thinks it's not being used, as it has a very slight impact (-1dB or so) on three of the WiFi channels. It seems it's a bit too pro-active on that, and it also disables it when the USB-serial-JTAG device is selected as a secondary console, which it is in the Arduino default setup. This, as the meme goes, kills the USB.

We'll get this fixed in ESP-IDF. Meanwhile, here's a workaround you can use:
```
#include "WiFi.h"                                                                                         

//there's an include for this but it doesn't define the function if it doesn't think it needs it, so manually declare the function
extern "C" void phy_bbpll_en_usb(bool en);

void setup()
{
    [...]
    WiFi.mode(WIFI_STA); //or any other statement that initializes the WiFi subsystem
    phy_bbpll_en_usb(true); //this brings the USB serial-jtag back to life. Suggest doing this immediately after wifi startup.
    [...]
}
```

me-no-dev, 2022-02-15:
Could you please try https://github.com/espressif/arduino-esp32/pull/6287 and see if that fixes the issue? cc @Jason2866

dronecz, 2022-02-15:
I did test this fix with 2.0.1 and 2.0.2 and I can confirm that issue with USB enumeration is gone.

me-no-dev, 2022-02-16:
Thanks for confirming @dronecz . I am merging the fix :)

closed 2022-02-16 by me-no-dev

ladyada, 2022-02-16:
also verified, wifi scan works great now! still having some SSID connection issues (oddly enough, works great and fast in micropython but not arduino?) but will open another issue if i can figure out whats up

me-no-dev, 2022-02-16:
@ladyada probably something in sdkconfig? Could you compare with micropython?

kostuch, 2022-04-05:
Hi,
I faced similar situation, when using BLE on ESP32-C3.
Connections and cabling are fine (if I flash chip with simple blink code, everything is ok).
But when I use code which connect to remote BLE device, the built-in USB become unavailable.
Device manager on windows then shows it as unknown device with message "A request for the USB device descriptor failed."
Normally (when ESP is flashed with simple blink) it shows COM port without problems.
Tried to add phy_bbpll_en_usb(true); after ble init, but it didn't help.
Any clue?

emeb, 2022-04-08:
Seeing a similar problem in ESP-IDF V5.0 with WiFi. Adding "phy_bbpll_en_usb(true);" after WiFi is started up doesn't result in working USB on the ESP32C3. Any suggestions on how to bypass this?

emeb, 2022-04-10:
I think I've got it figured out - the key is to add the phy_bbpll_en_usb(true); call prior to any wifi/network setup. Seems it's a flag to the OS that the USB shouldn't be disabled, rather than a command to enable it after it's been turned off.

kostuch, 2022-04-11:
> thanks @emeb I'll give this a go.

Could you also check described BLE-USB conflict, please?

DreamPWJ, 2022-07-22:
phy_bbpll_en_usb(true); doesn't work, is there any other solution?