espressif/arduino-esp32 #6269: Arduino IDE doesn't Compile on MacBook Pro M1 Pro

kilian0304, 2022-02-12:
### Board

ESP32 Dev Module; all from Espressif

### Device Description

Nothing, just the Dev Module

### Hardware Configuration

Nothing, just the Dev Module

### Version

v1.0.6

### IDE Name

Arduino IDE

### Operating System

macOS 12.3 Beta

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

11520

### Description

It doesn't Compile for my ESP32 Dev Module. For the official integrated boards from Arduino it compiles normal

### Sketch

```cpp
Just a test script
```

### Debug Message

```plain
/Users/kilianguggenberger/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/kilianguggenberger/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/liblto_plugin.so: error loading plugin: dlopen(/Users/kilianguggenberger/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/liblto_plugin.so, 0x0002): Library not loaded: /opt/osxcross/target/bin/../x86_64-apple-darwin12/lib/libstdc++.6.dylib
  Referenced from: /Users/kilianguggenberger/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/libexec/gcc/xtensa-esp32-elf/5.2.0/liblto_plugin.0.so
  Reason: tried: '/opt/osxcross/target/bin/../x86_64-apple-darwin12/lib/libstdc++.6.dylib' (no such file)
collect2: error: ld returned 1 exit status
exit status 1
Fehler beim Kompilieren für das Board ESP32 Dev Module.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-02-12:
Does the current release (v2.0.2) compile?

kilian0304, 2022-02-12:
Same error message with 2.0.0-rc3-nightly-20220212

philat7, 2022-02-13:
I have the identical problem. Same macOS build. V 2 ide doesn’t work for me either. Tested with Blink sketch to troubleshoot and same error

me-no-dev, 2022-02-13:
I am also on M1 Mac and have no issues with either IDE. Please do not report issues for 1.0.6 anymore. We have moved away from that and support 2.0.2 or higher. Als o the error above mentions `osxcross` which is not a part of our installation. Maybe your environment is broken?

philat7, 2022-02-13:
Getting more complex than me! Does this mean reinstalling my macOS? Already uninstalled and reinstalled both versions of the Arduino IDE to no avail

philat7, 2022-02-13:
Sorry - you're right, thank you! Just installed ver 2 IDE on my MacBook (old silicon but same os), worked perfectly, if you have no problems on M1 then i need to reinstall my OS that's all that's left I guess. Appreciate the push in the right direction.

me-no-dev, 2022-02-13:
maybe just remove `osxcross` or make sure that it's not exported to your env and used to compile. Reinstalling the whole OS is a bit overkill and will not tell you how to fix this issue if it happens again :)

philat7, 2022-02-13:
I'm in just the same place, same nonexistent file and directories. I'm told they're ghost files (but not by anyone reliable) I've gone into the hidden directories and folders, and code right out again none the wiser.  But - it ran right away on my MacBook (exactly the same build of OS but old silicon, me-no-dev is saying his runs on silicon, nothing else left than a messed up environment I think. - So, rather than understand the incomprehensible I''m going to do a reinstall, just waiting for it to back up. I can do primitive linux for a raspeberry pi, but this looks way more difficult to me. If you have the luxury of a reinstall maybe try it?

me-no-dev, 2022-02-13:
generally I do not expect something on the host to be able mess up the compilation. @igrr any clues? Could `/opt/osxcross` somehow come from the CI that built the toolchain?

philat7, 2022-02-15:
Overconfident, a simple reinstall made no difference. The delete and install option is going to take too much out that I want to keep. Maybe I'll just use my MacBook, frustrating though,, better to understand.

dpharris, 2022-02-15:
Jumping in, is https://github.com/tpoechtrager/osxcross related to this
problem?

David

philat7, 2022-02-16:
That looks like the problem exactly. The fix looks a bit complex for those like me not very used to Terminal emulator but from a quick read instructions appear very precise and full. Will look properly in next day or two and try it. 
really useful, thank you++

llamaonaskateboard, 2022-02-16:
I have this same issue on a x86 MacOS 12.3 Beta (21E5206e).

Running otool over liblto_plugin.0.so shows that it explicitly links to osxcross for whatever reason:
```
~/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/libexec/gcc/xtensa-esp32-elf/8.4.0 $ otool -L liblto_plugin.0.so
liblto_plugin.0.so:
	/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 744.19.0)
	/opt/osxcross/target/bin/../x86_64-apple-darwin12/lib/libstdc++.6.dylib (compatibility version 7.0.0, current version 7.25.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 169.3.0)
```
Theoretically the missing library should have always made it fail so it seems like something is triggering the lto plugin to be used when it wasn't previously.

A workaround that seems to work fine for me is installing gcc from Homebrew, creating the osxcross path and copying libstdc++.6.dylib:
```
brew install gcc
sudo mkdir -p /opt/osxcross/target/bin/../x86_64-apple-darwin12/lib/
sudo cp /usr/local/Cellar/gcc/11.2.0_3/lib/gcc/11/libstdc++.6.dylib /opt/osxcross/target/x86_64-apple-darwin12/lib/libstdc++.6.dylib
```

Jason2866, 2022-02-16:
If lto is generating the problem you have to use a ar and ranlib which has lto plugin supplied. This is achieved easiest by using the gcc-* equivalents (included in the provided toolchains)

kilian0304, 2022-02-16:
> 

After I did that, the error message is now

```
/Users/kilianguggenberger/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/kilianguggenberger/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/liblto_plugin.so: error loading plugin: dlopen(/Users/kilianguggenberger/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/liblto_plugin.so, 0x0002): Library not loaded: /opt/osxcross/target/bin/../x86_64-apple-darwin12/lib/libstdc++.6.dylib
  Referenced from: /Users/kilianguggenberger/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/libexec/gcc/xtensa-esp32-elf/5.2.0/liblto_plugin.0.so
  Reason: tried: '/opt/osxcross/target/bin/../x86_64-apple-darwin12/lib/libstdc++.6.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/opt/osxcross/target/x86_64-apple-darwin12/lib/libstdc++.6.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64'))
collect2: error: ld returned 1 exit status
```

llamaonaskateboard, 2022-02-17:
@Jason2866 In my case, I'm using PlatformIO to build so have limited control over the build process. This previously worked fine until something changed, trying to work out if it's MacOS, PIO or arduino-esp32 related - guessing MacOS.
I just found https://github.com/platformio/platform-espressif32/issues/727 so will discuss further there.

@kilian0304 Ah, Homebrew gcc must be x86 only then, you'll need a arm64 gcc package.

llamaonaskateboard, 2022-02-17:
I can reproduce this with the Arduino IDE as well.

Another workaround that works for both x86 and M1/arm64 is to edit platform.txt (`~/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/platform.txt`) and add `-fno-use-linker-plugin` to `compiler.libraries.ldflags`.

Note that I have no idea what the impact of not using the LTO plugin has on runtime/performance/stability/etc. but it seems to work fine for me.

You'll also need to follow https://github.com/espressif/arduino-esp32/issues/4717 if you haven't already done so to move from python2 to python3.

Jason2866, 2022-02-17:
@llamaonaskateboard  For Platformio it is even easier. Just do as described here https://github.com/platformio/platform-espressif32/pull/702
Or you can use my Platformio platform build which uses this.
```
platform                    = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2.2/platform-tasmota-espressif32-2.0.2.zip
```

llamaonaskateboard, 2022-02-17:
Ah ha, `platformio-build-esp32.py` in this repo at https://github.com/espressif/arduino-esp32/blob/master/tools/platformio-build-esp32.py#L83 has a lot of LINKFLAG options set including `-fno-lto`, however the latest version of framework-arduinoespressif32 (3.10006.210326) from the PlatformIO registry has significantly less options and does not have `-fno-lto`.

The same applies to the Arduino IDE as I just noticed my Boards Manager URL was set to https://dl.espressif.com/dl/package_esp32_index.json which only has up to 1.0.6. 
Updating the URL to https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json as per the current docs then gave me 2.0.2 which has `-fno-lto` in `platform.txt` already and works fine (after the python3 changes).
However, 2.0.2 fetches the gcc8_4_0-esp-2021r2 toolchain which still has a liblto_plugin that is linked to the /opt/osxcross absolute path so I don't see how LTO could ever work on MacOS with any of these toolchains?

@Jason2866 I just tried your platform package and although I got a tonne of ranlib warnings with the same liblto_plugin/libstdc++ message, it built successfully.

`platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream` as mentioned in https://github.com/espressif/arduino-esp32/issues/5436#issuecomment-885469600 works fine.

me-no-dev, 2022-02-17:
@llamaonaskateboard the thing is that I am using M1 Max machine (and M1 before that) and do not have any issues compiling for any of our chips. All flags that you see in PIO's scripts or platform.txt are automatically extracted from ESP-IDF's build.

Jason2866, 2022-02-17:
@llamaonaskateboard  i will not upgrade to MacOS 12.3 as long it is beta.
Project Tasmota is compiled using lto (Link time Optimization). Saves about 60k flash space for this project. No issues for ESP32/ESP32-S2. Does not work reliable with the C3. I use a Mac M1 Air for this. Not a single warning. Lto is verified working.
Let see what happens when 12.3 is released :-)

llamaonaskateboard, 2022-02-18:
Just running the xtensa-esp32-elf-g++ linker produces the error:

```
~/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2/bin $ ./xtensa-esp32-elf-g++ -Xlinker -v
collect2 version 8.4.0
~/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld -plugin ~/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so -plugin-opt=~/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/lto-wrapper -plugin-opt=-fresolution=/var/folders/j3/kf4rzjd54g9g5kg48n6fgdhw0000gn/T//ccQ2wW2v.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lnosys -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc ~/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/lib/crt0.o ~/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/crti.o ~/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/crtbegin.o -L~/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0 -L~/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2/bin/../lib/gcc -L~/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/lib -v -lstdc++ -lm -lgcc -lc -lnosys -lc -lgcc ~/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/crtend.o ~/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/crtn.o
~/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: ~/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so: error loading plugin: dlopen(~/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2/bin/../libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.so, 0x0002): Library not loaded: /opt/osxcross/target/bin/../x86_64-apple-darwin12/lib/libstdc++.6.dylib
  Referenced from: ~/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2/libexec/gcc/xtensa-esp32-elf/8.4.0/liblto_plugin.0.so
  Reason: tried: '/opt/osxcross/target/bin/../x86_64-apple-darwin12/lib/libstdc++.6.dylib' (no such file), '/opt/osxcross/target/x86_64-apple-darwin12/lib/libstdc++.6.dylib' (no such file)
collect2: error: ld returned 1 exit status
```

As a test, I tried changing the dylib path inside the LTO plugin from absolute to relative using `install_name_tool -change /opt/osxcross/target/bin/../x86_64-apple-darwin12/lib/libstdc++.6.dylib libstdc++.6.dylib liblto_plugin.0.so` and interestingly it stops the error and builds are successful and run.

I found this - https://openradar.appspot.com/FB9725981 - which might be related and sounds like maybe 12.3 beta "fixed" the behaviour or did something else entirely that's not mentioned in release notes relating to dlopen/dylibs and paths.
It will be interesting to see what happens on the public 12.3 release when it comes out.

Side note, libstdc++ is actually deprecated and replaced with libc++ on MacOS as per https://developer.apple.com/forums/thread/113746 so it might be worth looking at updating the toolchain build process regardless.

kilian0304, 2022-02-19:
> @kilian0304 Ah, Homebrew gcc must be x86 only then, you'll need a arm64 gcc package.

After I uninstalled the "normal" gcc and installed the gcc-arm-embedded my error message is:

`/Users/kilianguggenberger/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: /Users/kilianguggenberger/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/liblto_plugin.so: error loading plugin: dlopen(/Users/kilianguggenberger/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/liblto_plugin.so, 0x0002): Library not loaded: /opt/osxcross/target/bin/../x86_64-apple-darwin12/lib/libstdc++.6.dylib
  Referenced from: /Users/kilianguggenberger/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/libexec/gcc/xtensa-esp32-elf/5.2.0/liblto_plugin.0.so
  Reason: tried: '/opt/osxcross/target/bin/../x86_64-apple-darwin12/lib/libstdc++.6.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/opt/osxcross/target/x86_64-apple-darwin12/lib/libstdc++.6.dylib' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64'))
collect2: error: ld returned 1 exit status

Compilation error: exit status 1
`

I have no clue

gretel, 2022-02-19:
> After I uninstalled the "normal" gcc and installed the gcc-arm-embedded my error message is:

same here. 

@llamaonaskateboard
> Side note, libstdc++ is actually deprecated and replaced with libc++ on MacOS as per https://developer.apple.com/forums/thread/113746 so it might be worth looking at updating the toolchain build process regardless.

true, but `libc++` is not accepted well outside of apple's scope. `libstdc++` should work on all `posix`ish systems.

llamaonaskateboard, 2022-02-20:
@kilian0304 @gretel I think I misread your error message and got it around the wrong way, you want an x86 libstdc++ not arm64. I don't have a M1 so can't test. If homebrew won't provide it then search around or copy off a x86 Mac.

See my other comments above, there's other options:
- Update the Arduino board package to 2.0.2 which has LTO disabled by default
- Manually update your current platform.txt and add `-fno-lto` to `compiler.libraries.ldflags`
- Change liblto_plugin.0.so to not reference an absolute path of libstdc++.6.dylib
- Wait until non-beta 12.3 gets released in case the dylib behaviour changes again

@gretel Fair but it probably makes sense for packages for multiple platforms to utilise the available libraries for the platform. 
The esp8266 board toolchain doesn't dynamically link liblto_plugin against libstdc++ (or libc++ for that matter) at all so I wonder why the esp32 one is like this.

gretel, 2022-02-20:
@llamaonaskateboard thanks! but..

 * looks like `no-lto` does not keep the plugin from getting loaded
 * renaming the plugin lead to
   `xtensa-esp32-elf-g++: fatal error: -fuse-linker-plugin, but liblto_plugin.so not found`
 * looking around found [this ticket](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=48180)
 * tried to set `-fno-use-linker-plugin` via `-Wl` without effect
 * eventually [figured](https://docs.platformio.org/en/latest/scripting/examples/extra_linker_flags.html) how to override `LINKFLAGS`
 * linking ok 🥳 

`platform.ini`:

```ini
extra_scripts =
	pre:disable_lto.py
```

`disable_lto.py`:

```python
Import("env")
env.Append(
  LINKFLAGS=[
      "-fno-use-linker-plugin",
  ]
)
```

llamaonaskateboard, 2022-02-20:
@gretel That also works. In another of my comments above there’s a `platform =` line to point to this repo that has LTO already disabled

gretel, 2022-02-20:
> @gretel That also works. In another of my comments above there’s a `platform =` line to point to this repo that has LTO already disabled

thanks again; working on anupstram so trying to be less intrusive as possible - keep the `platform` as is.

Jason2866, 2022-03-15:
Updated to release MacOS 12.3 and it compiles without any error with my custom Platformio setup (which can use lto).

kilian0304, 2022-03-15:
Thanks. Compiled successfully

VojtechBartoska, 2022-03-23:
I'm closing this issue as you found a solution, labelling to be used For reference.

closed 2022-03-23 by VojtechBartoska

llamaonaskateboard, 2022-03-30:
I can also confirm that everything is fine on the release MacOS 12.3 after reversing the workarounds.
Guess we can just assume the betas had a weird dylib behaviour.