espressif/arduino-esp32 #6270: ESP32 AP Mode Local IP

Fideus, 2022-02-12:
### Board

ESP32

### Device Description

Custom Board

### Hardware Configuration

not relevant

### Version

v1.0.6

### IDE Name

Arduino

### Operating System

Windows 10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

Hello all,

I got a Problem Setting up an Access Point. there seems to be an DHCP server starting at Address x.x.x.241. This will cause an Problem if setting up the local IP in this range, like the example below.
Is there a option that this would be possible?
For my actual project the local IP Address could be changed by an user, and so this could cause problems.
Actual it works if I limit the local IP to x.x.x.240 But it would me more flexible if this isn't necessarry

### Sketch

```cpp
#include <WiFi.h>

IPAddress local_IP(192,168,1,254);
IPAddress gateway(192,168,1,9);
IPAddress subnet(255,255,255,0);

void setup()
{
  Serial.begin(115200);
  Serial.println();

  Serial.print("Setting soft-AP configuration ... ");
  Serial.println(WiFi.softAPConfig(local_IP, gateway, subnet) ? "Ready" : "Failed!");

  Serial.print("Setting soft-AP ... ");
  Serial.println(WiFi.softAP("ESPsoftAP_01") ? "Ready" : "Failed!");

  Serial.print("Soft-AP IP address = ");
  Serial.println(WiFi.softAPIP());
}

void loop() {}
```

### Debug Message

```plain
Setting soft-AP configuration ... Failed!
Setting soft-AP ... Ready
Soft-AP IP address = 192.168.1.254
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2022-02-16:
@Fideus 

Regarding your specific question about using an IP address within DHCP Server range, you may want to set DHCP server parameters to add x.x.x.254 as a Reserved IP Address. It may be possible depending on your router options or on your Linux DHCP Server (I don't have any information about your network or DHCP server).

Or you may want to set ESP Static IP out of DHCP range, but still in the range within the same netmask.

Anyway, **this is not related to ESP32 setup or ESP Arduino**.

## Observation:

The return (```false```) from ```softAPConfig()```, although still sets the Static IP, is being fixed for the release of Arduino Core 2.0.3. 
I see that you reported this issue for the Core 1.0.6.

If you want, you can try to apply the PR #6294 to your local machine with 1.0.6.

Fideus, 2022-02-16:
Hello SuGlider,

Thank you for your help.

It isn't only the false return what is the problem. I also can't connect to the ESP32 with my device.
My device which trying to connect to the esp can't resolve an IP address.
Regarding your PR #6294 it seems that the DHCP server is stopped when the Local_Ip is within the DHCP Range.

Is there a way to exclude the Local_Ip from the DHCP server which is running on the ESP32? Or reconfiguring the DHCP range?

With the AccesPoint mode on the ESP there is no external router

SuGlider, 2022-02-17:
@Fideus 
I found the problem with your configuration.
Looking the code: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiGeneric.cpp#L139-L148
It is clear that when a Static IP address is assigned to ESP32 in AP mode, it will set a DHCP range from IPaddr + 1 to IPaddr + 11.

Because your example assigns x.x.x.254, it is already at the end of network addressing space and it fails with ```DHCPS Set Lease Failed! 0x5001``` message (in order to see it, turn  "Core Debug Level:" to "Verbose").

Try assigning static IP as 192.168.1.240 (instead of .254) and it will work fine.

> Is there a way to exclude the Local_Ip from the DHCP server which is running on the ESP32? Or reconfiguring the DHCP range?

Local IP is the DHCP IP address it self. Range is not configurable at this time and, as said, it goes from IPaddr+1 to IPaddr + 11.

SuGlider, 2022-02-17:
@Fideus 
PR #6296 will make your example to work.
Using 192.168.1.254 as IP Address for the AP Server will move DHCP range to 192.168.1.1 -- 192.168.1.11

I think this covers you request and solves this issue.
Please let me know.

PR #6294 (2022-02-16): Fixes softAPConfig() return
## Summary
This PR fixes the return of Static IP setup using ```softAPConfig()``` function.
```softAPConfig()``` returns always ```false``` even when the operation is successful.
This occurs because of a failure return code while trying to stop DHCP Client Service.

## Impact
None, just fixes the function.

## Related links
Fixes #6270
PR #6296 (2022-03-02): Fixes DHCP Server Lease Range for any AP Server Static IP Address
## Summary
There is an issue that prevents DHCP server to start if WiFi AP Mode is used with Static IP that is too close to the end of subnet range. Lease adresses range will overflow NetMask preventing DHCP Server to start, issuing error message ```DHCPS Set Lease Failed! 0x5001```.

This PR fixes it and continues to consider the same previously coded assumptions about netmasking and DHCP leasing range size for AP mode.

## Impact
None.

## Related links
Fixes #6270
closed 2022-03-02 by me-no-dev