espressif/arduino-esp32 #6271: Setting DHCP IP-range start for softAP

ch-fb, 2022-02-12:
### Related area

WiFi softAP DHCP

### Hardware specification

ESP32

### Is your feature request related to a problem?

When several Wi-Fi stations are connected to an ESP's softAP and the ESP reboots, it can happen that some stations hold on to their old DHCP lease. However, the DHCP server does not know and re-assigns such still-in-use IP addresses. As a workaround for the problem, the DHCP server could use a different IP range after each restart (i.e. cycling through several ranges), but this requires the possibility of configuring the DHCP server.

### Describe the solution you'd like

Possibility of configuring the DHCP server:

*) Adding IPAddress argument to softAPConfig:
bool softAPConfig(IPAddress local_ip, IPAddress gateway, IPAddress subnet, IPAddress dhcpstart);

(can be an optional argument using a default value, or the current 3-argument version softAPConfig(local_ip,gateway,subnet) calls the extended 4-argument version)

*) This requires adding an argument to set_esp_interface_ip:
esp_err_t set_esp_interface_ip(esp_interface_t interface, IPAddress local_ip=IPAddress(), IPAddress gateway=IPAddress(), IPAddress subnet=IPAddress(), IPAddress dhcpstart=IPAddress());

An implementation could look like this:

  dhcps_lease_t lease;
  lease.enable = true;
  lease.start_ip.addr = static_cast<uint32_t>(dhcpstart);
  lease.end_ip.addr   = static_cast<uint32_t>(dhcpstart) + (10 << 24);

instead of
  lease.start_ip.addr = static_cast<uint32_t>(local_ip) +  (1 << 24);
  lease.end_ip.addr   = static_cast<uint32_t>(local_ip) + (11 << 24);
  
This extension would help me (and probably some others) a lot.
Thanks for all your great work,
Christian

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2022-02-13:
This does not look like something that should be fixed on the ESP32 SoftAP side. Why did the clients hold their lease if the AP was rebooted and they had to reconnect again? This does not sound at all OK and not how networking works. To "hold" a lease, a client must first tell the DHCP server that it intends to do so and the DHCP server needs to respond that it has accepted the same lease.

SuGlider, 2022-02-19:
@ch-fb 
I can confirm that I could reproduce the issue. 

How to reproduce:
I used C3 as SoftAP (IP x.x.x.1), ESP32 as STA, and PC Win10 as STA.
When C3 starts DCHP is Ready. Then I connected ESP32 with IP x.x.x.2 followed by  Win10 with IP Addr x.x.x.3.
Everything fine.

I disconnect Win10 and reset AP (C3). Keep ESP32 STA on.
Start reconnection of Win10 and Turn on AP (C3).
Result is that Win10 now acquired IP x.x.x.2 and ESP32 STA after loosing connection to AP tries to acquire an IP and fail.

### This is not easy to reproduce. I had to try it several times to see the issue in action.

ESP32 STA Log:
```
  4343][D][WiFiGeneric.cpp:842] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  4365][V][WiFiGeneric.cpp:305] _arduino_event_cb(): STA Got New IP:192.168.1.2
[  4365][D][WiFiGeneric.cpp:842] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  4368][D][WiFiGeneric.cpp:896] _eventCallback(): STA IP: 192.168.1.2, MASK: 255.255.255.0, GW: 192.168.1.1
.
Connected to WiFi network with IP Address: 192.168.1.2
[120038][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: ESP32-Access-Point, BSSID: 84:f7:03:60:17:61, Reason: 6
[120039][D][WiFiGeneric.cpp:842] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[120047][W][WiFiGeneric.cpp:863] _eventCallback(): Reason: 6 - NOT_AUTHED
[240040][V][WiFiGeneric.cpp:309] _arduino_event_cb(): STA IP Lost
[240041][D][WiFiGeneric.cpp:842] _eventCallback(): Arduino Event: 9 - STA_LOST_IP
```

SuGlider, 2022-02-19:
A possible solution to this issue is to detect the STA_LOST_IP and restart WiFi STA.
Try using SYSTEM_EVENT_STA_LOST_IP.

Example:
``` cpp
// call back function when IP is Lost by Station -- needs to reconnect to AP
void WiFiStationLostIP(WiFiEvent_t event, WiFiEventInfo_t info)
{
  WiFi.end();
  delay(150); // wait a bit for the necessary processing
  WiFi.begin(ssid, password);
}

void setup()
{
  Serial.begin(115200);
  // Set a callback function when the event STA_LOST_IP happens
  WiFi.onEvent(WiFiStationLostIP, SYSTEM_EVENT_STA_LOST_IP);
  // Start WiFi in STA Mode
  WiFi.begin(ssid, password);
}
```

SuGlider, 2022-02-19:
I tested the LOST IP EVENT callback and it works as expected, reconnecting the STA again when its IP is used by another device, right after AP reset/reboot.

This is the STA log with the suggestion I made:

There are two steps in the LOG: 
1- ESP32 STA connects first time with IP 192.168.4.2 to ESP32-C3 AP. Then I start Win10 connection to AP, at the same time that I reset the C3. At this time I see AP message leasing IP 192.168.4.2 to Win10, which is previuos STA IP Address.
2- STA gets diconnected and can't get previous IP Address back. Then it goes to STA_LOST_IP state and calls ```WiFiStationLosIP()```, reconnecting with a new IP 192.168.4.4.
 
```
STA_LOST_IP event callback is set.
[    54][D][WiFiGeneric.cpp:850] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   149][V][WiFiGeneric.cpp:283] _arduino_event_cb(): STA Started
[   150][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[   151][D][WiFiGeneric.cpp:850] _eventCallback(): Arduino Event: 2 - STA_START
Connecting
....[  2248][V][WiFiGeneric.cpp:295] _arduino_event_cb(): STA Connected: SSID: ESP32-Access-Point, BSSID: 84:f7:03:60:17:61, Channel: 1, Auth: OPEN
[  2250][D][WiFiGeneric.cpp:850] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  2276][V][WiFiGeneric.cpp:305] _arduino_event_cb(): STA Got New IP:192.168.4.2
[  2276][D][WiFiGeneric.cpp:850] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  2279][D][WiFiGeneric.cpp:904] _eventCallback(): STA IP: 192.168.4.2, MASK: 255.255.255.0, GW: 192.168.4.1
.
Connected to WiFi network with IP Address: 192.168.4.2
[ 42882][V][WiFiGeneric.cpp:300] _arduino_event_cb(): STA Disconnected: SSID: ESP32-Access-Point, BSSID: 84:f7:03:60:17:61, Reason: 6
[ 42883][D][WiFiGeneric.cpp:850] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 42891][W][WiFiGeneric.cpp:871] _eventCallback(): Reason: 6 - NOT_AUTHED
[162884][V][WiFiGeneric.cpp:309] _arduino_event_cb(): STA IP Lost
[162885][D][WiFiGeneric.cpp:850] _eventCallback(): Arduino Event: 9 - STA_LOST_IP
[162886][D][STA_AP_TESTE.ino:41] WiFiStationLostIP(): Enter into LOST IP Event Call Back...
[164963][V][WiFiGeneric.cpp:295] _arduino_event_cb(): STA Connected: SSID: ESP32-Access-Point, BSSID: 84:f7:03:60:17:61, Channel: 1, Auth: OPEN
[164965][D][WiFiGeneric.cpp:850] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[165005][V][WiFiGeneric.cpp:305] _arduino_event_cb(): STA Got New IP:192.168.4.4
[165005][D][WiFiGeneric.cpp:850] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[165008][D][WiFiGeneric.cpp:904] _eventCallback(): STA IP: 192.168.4.4, MASK: 255.255.255.0, GW: 192.168.4.1

```

SuGlider, 2022-02-19:
@ch-fb 

Please let me know if this solution with WiFi Event Callbck solves the issue.

ch-fb, 2022-03-07:
Hi, sorry for replying so late.
I am afraid this does not solve the problem. My STAs are laptops (windos/linux/macOS) and phones, not other ESPs. I can reproduce the problem relatively frequently and I found discussion of the problem on the net, so others had the problem too. 

Reboting an ESP32 is very fast, so it's AP is gone for just a very short time, too short for other STAs to even notice. And after the reboot, ESPs DHCP server does not know about them anymore. My workaround with cycling through different IP ranges works quite nicely in practice and would not mean much to be implemented in arduino-esp32. 

I don't know whether a DHCP server has the possibility to force all STAs to renew their leases. If so, this could be the solution, but that one is part of the SDK.

VojtechBartoska, 2022-03-23:
@SuGlider Can you please take a look on last comment and validate if already merged PR fix this fully? Thanks!

SuGlider, 2022-05-12:
@ch-fb - The PR #6731 implements the feature you have requested. 
It shall be merged for the Arduino Core 2.0.4 future release, but you can apply it to your project if you want to do so.

PR #6731 (2022-05-16): Adds DHCP Range Setup to APMode
-----------
## Description of Change
This PR adds an optional parameter to `WiFi.softAPConfig()` in order to set the initial address of the DHCP Leasing Range.
It has no impact over sketches that do not set this parameter because it will work as the default previous way. 

The PR also improves IP verification for errors with subnet, collision with DHCP range and setting IPs out of the subnet range.

## Tests scenarios
Tested on ESP32, ESP32-S2, ESP32-C3 and ESP32-S3.
Tested with different configurations for IPs, including using different net masks.

## Related l [... cut]