espressif/arduino-esp32 #6274: unable to compile for ESP32 using Arduino IDE

exiledyorkie, 2022-02-12:
### Board

ESP32 Dev Module

### Device Description

Just the board, nothing else attatched.
The issue is compiling, prior to the board being tethered to the laptop.

### Hardware Configuration

Nothing

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Hi
I'm trying to use the ESP8266Audio library with an ESP32 dev 
board.
I've used the library for an ESP8266 board which worked fine but now I want to use and ESP32 board. It won't compile for any board, if I remove the ESP8266Audio library, then all is fine. I've tried partitions to increase space and changed the board manager multiple times but nothing works. Any idea's please?

I posted on the librarys github and was recomended posting here, due to it being a compiler bug.

### Sketch

```cpp
Just removing the ESP8266Audio library enables a compile. So it shouldn't be the sketch itself?

#define BLYNK_PRINT Serial

#include <Arduino.h>
//#include <ESP8266WiFi.h>
#include <WiFi.h>
#include <WiFiClient.h> //maybe this one?
#include <DNSServer.h>
//#include <ESP8266WebServer.h>
#include <WebServer.h>
#include <WiFiManager.h>
//#include <BlynkSimpleEsp8266.h>
#include <BlynkSimpleEsp32.h>
//#include <Servo.h>
#include <ESP32Servo.h>
//additional test as not compiling
#include <FS.h>
#include "SPIFFS.h"
#include <DNSServer.h>
//#include <ArduinoJson.h>
//

// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
char auth[] = "<insert blynk token here>";
int xVal, yVal, headSpeed, soundCnt;
bool startup, headLeft, headRight;
double rampUpHead, rampUpBody;

// Your WiFi credentials.
// Set password to "" for open networks.
char ssid[] = "YourNetworkName";
char pass[] = "YourPassword";

Servo leftServo, rightServo, headServo;
const int leftServoPin = 5;
const int rightServoPin = 6;
const int headServoPin = 7;
const int NEUTRAL = 1500;
const int CONTROLLER_NEUTRAL = 511;
const int NEUTRAL_ZONE = 20;
const int MIN_CONTROLLER = 0;
const int MAX_CONTROLLER = 1024;

WiFiServer server(80);

BLYNK_WRITE(V0)
{
  xVal = param[0].asInt();
  yVal = param[1].asInt();
  
  startup = true;
}

BLYNK_WRITE(V2)
{
  headLeft = param.asInt();
}

BLYNK_WRITE(V3)
{
  headRight = param.asInt();
}

BLYNK_WRITE(V4)
{
  headSpeed = param.asInt();
}

void setup()
{
  // Debug console
  Serial.begin(9600);
  startup = false;
  headSpeed = 0;
  headLeft = false;
  headRight = false;

  Blynk.config(auth);
  WiFiManager wifiManager;
  bool result = wifiManager.autoConnect("Baby_R2D2","MasterLuke");
  Blynk.connect();
}

void loop()
{
  int leftMotor, rightMotor;
  //leg motors
  int modifiedX = map(xVal, MIN_CONTROLLER, MAX_CONTROLLER, 0, 1000) - 500;
  int modifiedY = map(yVal, MIN_CONTROLLER, MAX_CONTROLLER, 0, 1000) - 500;

  if (modifiedX < 0) {
    leftMotor = NEUTRAL + modifiedY;
    rightMotor = NEUTRAL - ((modifiedY + 1) / abs(modifiedY + 1)) * 500;
  }
  else if (modifiedX > 0) {
    leftMotor = NEUTRAL + ((modifiedY + 1) / abs(modifiedY + 1)) * 500;
    rightMotor = NEUTRAL - modifiedY;
  }
  else {
    leftMotor = NEUTRAL + modifiedY;
    rightMotor = NEUTRAL - modifiedY;
  }

  if(headLeft || headRight) {
    int multiplier;
    if(headLeft) {
      multiplier = 1;
    }
    else {
      multiplier = -1;
    }
    
    headServo.attach(headServoPin);
    headServo.writeMicroseconds(NEUTRAL + map(headSpeed, 0, 1024, 0, 500) * multiplier * rampUpHead);

    if(rampUpHead < 1.0) {
      rampUpHead += 0.1;
    }
  }
  else {
    rampUpHead = 0.0;
    headServo.detach();
  }

  if (startup) {
    if(leftMotor == NEUTRAL && rightMotor == NEUTRAL) {
      rampUpBody = 0.0;
      leftServo.detach();
      rightServo.detach();
    }
    else {
      if(!leftServo.attached()) {
        leftServo.attach(leftServoPin);
      }
      if(!rightServo.attached()) {
        rightServo.attach(rightServoPin);
      }
      leftServo.writeMicroseconds(leftMotor);
      rightServo.writeMicroseconds(rightMotor);
    }
  }
  
  delay(50);
  Blynk.run();
}
```
```

### Debug Message

```plain
Arduino: 1.8.19 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None"

C:\Users\cpmag\Documents\arduino\libraries\ESP8266Audio\src\AudioFileSourceICYStream.cpp:23: warning: "_GNU_SOURCE" redefined

#define _GNU_SOURCE

: note: this is the location of the previous definition

In file included from C:\Users\cpmag\Documents\arduino\libraries\ESP8266Audio\src\AudioGeneratorMIDI.cpp:65:

C:\Users\cpmag\Documents\arduino\libraries\ESP8266Audio\src\libtinysoundfont/tsf.h: In function 'void tsf_channel_midi_control(tsf*, int, int, int)':

C:\Users\cpmag\Documents\arduino\libraries\ESP8266Audio\src\libtinysoundfont/tsf.h:2100:1: error: insn does not satisfy its constraints:

}

^

(insn 858 343 344 51 (set (reg:SF 19 f0 [407])

    (mem/u/c:SF (symbol_ref/u:SI ("*.LC248") [flags 0x2]) [0  S4 A32])) "C:\Users\cpmag\Documents\arduino\libraries\ESP8266Audio\src\libtinysoundfont/tsf.h":2053 47 {movsf_internal}

 (nil))

during RTL pass: postreload

C:\Users\cpmag\Documents\arduino\libraries\ESP8266Audio\src\libtinysoundfont/tsf.h:2100:1: internal compiler error: in extract_constrain_insn, at recog.c:2210

Please submit a full bug report,

with preprocessed source if appropriate.

See https://gcc.gnu.org/bugs/ for instructions.

Multiple libraries were found for "SD.h"

Used: C:\Users\cpmag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\SD

Not used: C:\Program Files (x86)\Arduino\libraries\SD

Multiple libraries were found for "WiFi.h"

Used: C:\Users\cpmag\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.2\libraries\WiFi

Not used: C:\Program Files (x86)\Arduino\libraries\WiFi

exit status 1

Error compiling for board ESP32 Dev Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-02-13:
This has nothing to do with this repository :) Such issues should be fixed by the library maintainer (if he wants to support ESP32 in the same lib)

closed 2022-02-13 by me-no-dev

me-no-dev, 2022-02-13:
@earlephilhower maybe then you can help diagnose/fix this at https://github.com/espressif/crosstool-NG ? Was your library supposed to work? Name suggests ESP8266 :)

earlephilhower, 2022-02-13:
The library actually built and ran on all earlier ESP32  toolchain releases (say from ~7 months ago??).  It even has code to do audio-out using the ULP processor, in fact.

Since it seems to be a logic bug in the Xtensa compiler bits, most likely this is related to the specific GCC version you're building.  Did you change it recently?  I'll see what I can see in the repo you listed.

me-no-dev, 2022-02-13:
versions before 2.0.0 used older 5.2.0 toolchain. We are on 8.4 now. (And I heard we will go even higher soon)