espressif/arduino-esp32 #6275: SoftAP in WIFI_MODE_APSTA can't take password on ESP32-S2

tresler, 2022-02-13:
### Board

ESP32-S2

### Device Description

ESP32-S2

### Hardware Configuration

19,20 to usb

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Linux (Ubuntu 21.10)

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am trying example code for WIFI_MODE_APSTA. A can set this mode, I can connect with password to my home wifi and I can set softAP config and SSID but only without password. If I try it with password (WiFi.softAP(soft_ap_ssid, soft_ap_password)), ESP use default name without encryption. 

### Sketch

```cpp
#include <WiFi.h>
#include <ESPAsyncWebServer.h>

IPAddress local_IP(192,168,0,1);
IPAddress gateway(192,168,0,1);
IPAddress subnet(255,255,255,0);
 
const char* wifi_network_ssid = "asdas";
const char* wifi_network_password =  "asdsa";
 
const char *soft_ap_ssid = "ESP32";
const char *soft_ap_password = "passcode";
 
AsyncWebServer server(80);

void setup() {
 
  Serial.begin(115200);
  
  delay(5000);
  
  WiFi.mode(WIFI_MODE_APSTA);
  
  delay(2000);

  WiFi.softAPConfig(local_IP, gateway, subnet);
 
  WiFi.softAP(soft_ap_ssid);                      //-> function
  //WiFi.softAP(soft_ap_ssid, soft_ap_password);      //-> doesn't function
  
  delay(2000);
 
  WiFi.begin(wifi_network_ssid, wifi_network_password);

 
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.println("Connecting to WiFi..");
  }
 
  Serial.print("ESP32 IP as soft AP: ");
  Serial.println(WiFi.softAPIP());
  Serial.print("ESP32 SSID as soft AP: ");
  Serial.println(WiFi.softAPSSID());
 
  Serial.print("ESP32 IP on the WiFi network: ");
  Serial.println(WiFi.localIP());
 
 
  server.on("/hello", HTTP_GET, [](AsyncWebServerRequest * request) {
 
    if (ON_STA_FILTER(request)) {
      request->send(200, "text/plain", "Hello from STA");
      return;
 
    } else if (ON_AP_FILTER(request)) {
      request->send(200, "text/plain", "Hello from AP");
      return;
    }
 
    request->send(200, "text/plain", "Hello from undefined");
  });
 
  server.begin();
 
}
 
void loop() {}
```

### Debug Message

```plain
Connecting to WiFi..
Connecting to WiFi..
ESP32 IP as soft AP: 192.168.0.1
ESP32 SSID as soft AP: ESP3232
ESP32 IP on the WiFi network: 10.0.1.32¨
```

### Other Steps to Reproduce

I also trying different order of setup.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lordfitzy, 2022-04-10:
I'm seeing the same issue.  In my case default name SSID assigned is ESP_EA5359.  I'm using a ESP32-CAM clone.

lordfitzy, 2022-04-10:
After persistent Googling I found the fix.  You have to set the password long enough.  A 4-char password is fubar.  10-char password works great.  Not sure what the cutoff is but problem is solved.

lbernstone, 2022-04-10:
wpa2 requires an 8 character password.

lordfitzy, 2022-04-10:
Thx for the help!  I did find a fix and it's exactly what you said.

closed 2022-04-11 by VojtechBartoska