espressif/arduino-esp32 #6276: bootloader for esp32-C3

hzhh110, 2022-02-13:
### Related area

ESP32-C3

### Hardware specification

esp32-c3-devkitm-1

### Is your feature request related to a problem?

ESP32-C3 is good, I want to know how to get the variable BOARD_FLASH_MODE and the variable _${__get_board_f_flash(__env__)}?

` FLASH_EXTRA_IMAGES=[
        ("0x0000", join(FRAMEWORK_DIR, "tools", "sdk", "esp32c3", "bin", "bootloader_${BOARD_FLASH_MODE}_${__get_board_f_flash(__env__)}.bin")),
        ("0x8000", join(env.subst("$BUILD_DIR"), "partitions.bin")),
        ("0xe000", join(FRAMEWORK_DIR, "tools", "partitions", "boot_app0.bin"))
    ]`

### Describe the solution you'd like

[image]

### Additional context

[image]

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

hzhh110, 2022-02-13:
For VSCode to program
Building in release mode
Retrieving maximum program size .pio/build/esp32-c3-devkitm-1/firmware.elf
Checking size .pio/build/esp32-c3-devkitm-1/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  15.7% (used 51456 bytes from 327680 bytes)
Flash: [======    ]  62.8% (used 1233806 bytes from 1966080 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: /dev/cu.usbserial-1470
Uploading .pio/build/esp32-c3-devkitm-1/firmware.bin
esptool.py v3.2
Serial port /dev/cu.usbserial-1470
Connecting....
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 84:f7:03:3a:e0:0c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB

hzhh110, 2022-02-13:
[image]

hzhh110, 2022-02-13:
After I use vscode + platfromio to program, after running, press the reset button to output:
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x420
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x236c
SHA-256 comparison failed:
Calculated: ccb0d00bac7e84e1d90a12e4f75f4ab6c5f7e71bb209afd5819c4c9557a6db71
Expected: c9cf160580940ec7801c73b16423824e72ad12055c732e83ce66332240af42a7
Attempting to boot anyway...
entry 0x403ce000

me-no-dev, 2022-02-13:
I am not sure what is the issue here? Last log looks fine, but I do not see you using Arduino in any part of it.

Jason2866, 2022-02-13:
All bootloaders are here https://github.com/espressif/arduino-esp32/tree/master/tools/sdk/esp32c3/bin. Which one you need depends on what flash mode you have used to compile your programm.

hzhh110, 2022-02-17:
It's because flash_download_tool cannot select flash_size, and "flash_size = keep" defaults to 2MB and needs to be changed to "flash_size = 4MB"

0hotpotman0, 2022-03-21:
Hi all
I have a similar question, I burn the BLE at firmware in my board and the firmware has success burn into the board, and then I use the UART tool sent AT command, there is not any response, please help 

[image]

I also tried the factory firmware, it did not work as well. I have checked the cable and tried different UART PINs, all are followed the official guide. 

[image]

best regards

Jason2866, 2022-03-21:
@0hotpotman0  Probably the same as @hzhh110 had. The bootpartition needs to be patched to align with your your boards flash size. If it is wrong device wont boot and you will see nothing via serial console. Use esptool.py to flash your firmware. If you specify the flash size it will modify the bootpartion.bin accordingly

me-no-dev, 2022-03-23:
@Jason2866 @0hotpotman0 is trying to flash AT firmware that has nothing to do with Arduino

Jason2866, 2022-03-23:
@me-no-dev True, but the magic header entrys needs to be correct for every firmware.

VojtechBartoska, 2022-04-11:
is this still relevant?

VojtechBartoska, 2022-05-03:
can this be consider as solved? @hzhh110

VojtechBartoska, 2022-05-18:
Closing as expired, if needed you can reopen the issue.

Thanks for understanding.

closed 2022-05-18 by VojtechBartoska

fatpunkybearass, 2024-09-12:
你好，请问可以分享给我v3.9.0的下载器吗，我找遍全网找不到，有一个别人写的脚本只有这个版本的下载器才可以使用，谢谢

VojtechBartoska, 2024-09-13:
This repository use only English.