espressif/arduino-esp32 #6283: Ethernet doesn't work with IPV6

copercini, 2022-02-14:
### Board

ESP32 dev module + LAN8720A

### Device Description

ESP32 dev module + LAN8720A

### Hardware Configuration

I think the hardware is fine, LAN8720A is working flawless with IPV4

### Version

v2.0.2

### IDE Name

Arduino IDE

### Operating System

Ubuntu 20.04

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

It seems that IPV6 is not working with ethernet interface, I'm trying to add `ETH.enableIpV6();` inside `ARDUINO_EVENT_ETH_CONNECTED` event, but it seems to produce no results, and `ARDUINO_EVENT_ETH_GOT_IP6` is never called (check attached code). It's working fine with IPV4

I also checked with Wi-FI STA interface and IPV6 works as expected

### Sketch

```cpp
#include <ETH.h>

static bool eth_connected = false;

void WiFiEvent(WiFiEvent_t event)
{
  Serial.printf("Event: %d\n", event);
  switch (event) {
    case ARDUINO_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case ARDUINO_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");

      ////I'm trying to enable IPV6 here
      delay(10);
      ETH.enableIpV6();
      delay(10);
      /////

      break;
    case ARDUINO_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      break;
    case ARDUINO_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case ARDUINO_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;

    ////Expecting this IPV6 event, but it never happens
    case ARDUINO_EVENT_ETH_GOT_IP6:
      Serial.println("Ethernet IPv6 is preferred");
      break;
    ////

    default:
      break;
  }
}

void testClient(const char * host, uint16_t port)
{
  Serial.print("\nconnecting to ");
  Serial.println(host);

  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}

void setup()
{
  Serial.begin(115200);
  WiFi.onEvent(WiFiEvent);
  ETH.begin(1, 32, 2, 4, ETH_PHY_LAN8720); //ETH_PHY_ADDR, ETH_PHY_POWER, ETH_PHY_MDC, ETH_PHY_MDIO, ETH_PHY_TYPE
}

void loop()
{
  if (eth_connected) {
    testClient("google.com", 80);
  }
  delay(10000);
}
```

### Debug Message

```plain
Event: 18
ETH Started
Event: 20
ETH Connected
Event: 22
ETH MAC: 24:0A:C4:E3:26:4B, IPv4: 192.168.31.52, FULL_DUPLEX, 100Mbps
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2022-02-14:
`ARDUINO_EVENT_ETH_GOT_IP6` would mean that an address is assigned. How are you expecting to get your ipv6 address? SLAAC and DHCPv6 are not enabled in sdkconfig.

copercini, 2022-02-15:
@lbernstone Thanks for your reply, it's a bit weird because using Wi-Fi in the exactly same sdkconfig it's able to get at least a local IPV6, and event `ARDUINO_EVENT_WIFI_STA_GOT_IP6` is called when it happens. I've tested with this example https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiIPv6/WiFiIPv6.ino

I'm also testing using IDF v4.4, the event appears but the IPV6 doesn't work using ethernet, only using Wi-Fi. Enabling SLAAC and DHCPv6 there, it gets a public IPV6, but I'm also not able to access ESP32 webserver from it

So far:
Arduino 2.0.2:
WiFi IPv4 works and I'm able to access webserver locally
WiFi IPv6 works and I'm able to access webserver locally
Ethernet IPv4 works and I'm able to access webserver locally
Ethernet IPv6 doesn't even call the event and doesn't work

IDF v4.4:
WiFi IPv4 works and I'm able to access webserver locally
WiFi IPv6 works and I'm able to access webserver locally
Ethernet IPv4 works and I'm able to access webserver locally
Ethernet IPv6 calls the event with an IPv6, but I'm not able to access webserver from it

me-no-dev, 2022-02-15:
so in both cases IPv6 does not work? With any option in menuconfig? That is rather strange... it used to work before :) Maybe regression in ETH in IDF?

copercini, 2022-02-15:
yes, in both cases it doesn't work, so I think it's an IDF bug, not in arduino layer. I'll open an issue there (and close this one?)

me-no-dev, 2022-02-15:
maybe leave it open to be closed when fixed there? It is still possible that something in the API has changed and requires changes here, but I doubt it.

ewpa, 2022-02-15:
Confirming I had this issue for some time and I applied a fix in commit e205fedbe8522a5fbf84f5853c214c930bc4be42 for my fork of ewpa/esp-lwip.  @copercini perhaps try to recreate the issue in my fork of ewpa/esp32-arduino under branch 2.0.2+dev and test the effectiveness of the below fix?

_Always process Neighbor Solicitation messages.  This fixes IPv6 Neighbor Discovery not finding the ESP32 on Ethernet interfaces._

VojtechBartoska, 2022-07-14:
still valid @copercini?

VojtechBartoska, 2022-07-26:
This is supposed to be fixed, can you please retest under v2.0.4 @copercini? Thanks.

VojtechBartoska, 2022-08-23:
Closing as solved, if needed please reopen the issue.

closed 2022-08-23 by VojtechBartoska