espressif/arduino-esp32 #6286: Mbedtls ignoring MBEDTLS_SSL_MAX_CONTENT_LEN or MBEDTLS_SSL_IN_CONTENT_LEN

igor-d-n, 2022-02-14:
### Board

ESP32

### Device Description

ALL esp32 devices

### Hardware Configuration

empty

### Version

v2.0.2

### IDE Name

Visual Studio and vMicro plugin

### Operating System

Windows 10

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

512000

### Description

// all below has NO any effect for memory usage by TLS connections
#define CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN 4096
#define MBEDTLS_SSL_MAX_FRAGMENT_LENGTH 4096
#define MBEDTLS_SSL_MAX_CONTENT_LEN 4096
#define MBEDTLS_SSL_DTLS_MAX_BUFFERING 8192
#define MBEDTLS_SSL_IN_CONTENT_LEN 4096
#define MBEDTLS_SSL_OUT_CONTENT_LEN 4096

### Sketch

```cpp
Just try to do so:

#define CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN 4096
#define MBEDTLS_SSL_MAX_FRAGMENT_LENGTH 4096
#define MBEDTLS_SSL_MAX_CONTENT_LEN 4096
#define MBEDTLS_SSL_DTLS_MAX_BUFFERING 8192
#define MBEDTLS_SSL_IN_CONTENT_LEN 4096
#define MBEDTLS_SSL_OUT_CONTENT_LEN 4096

and then do ANY tls connection. Memory consumption always the same
```

### Debug Message

```plain
not needed
```

### Other Steps to Reproduce

just read above

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2022-02-15:
This is because mbedtls is already precompiled. The defines are to show you how it was configured. What is the issue with the current settings?

closed 2022-02-15 by me-no-dev

igor-d-n, 2022-02-15:
Issue is that mbedtls can't work) It consumes too much memory without possibility to lower it. In bearssl memory buffer is setting dynamically. In mbedtls it's setting only by defines and that is very bad.  May be it's possible to include it like sources instead of precompiled?

me-no-dev, 2022-02-15:
mbedtls works fine on our end. there are many SSL examples out there. If you have a particular case where you do not have enough memory, you have options, like use the lib-builder to prepare your own precompile IDF libs with whatever settings you want or use Arduino as IDF component https://github.com/espressif/esp32-arduino-lib-builder

igor-d-n, 2022-02-16:
> mbedtls works fine on our end. there are many SSL examples out there. If you have a particular case where you do not have enough memory, you have options, like use the lib-builder to prepare your [...]

Thanks, it works but with 2.0.1 version now I'm getting errors on linking. Tried to use 2.0.2 but there is a problem with OTA update. Can you suggest something?
```
ld.exe: C:\Users\igord\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.1\tools\sdk\esp32\lib\libmbedcrypto.a(bignum.c.obj):(.literal.mpi_miller_rabin+0x0): undefined reference to __wrap_mbedtls_mpi_exp_mod

Error linking for board ESP32 Dev Module
Build failed for project 'BLE_server_multiconnect'
 
ld.exe: C:\Users\igord\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.1\tools\sdk\esp32\lib\libmbedcrypto.a(bignum.c.obj): in function mpi_miller_rabin
bignum.c*:2796: undefined reference to __wrap_mbedtls_mpi_exp_mod
 
ld.exe: C:\Users\igord\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.1\tools\sdk\esp32\lib\libmbedcrypto.a(rsa_internal.c.obj): in function mbedtls_rsa_deduce_primes
rsa_internal.c*:165: undefined reference to __wrap_mbedtls_mpi_exp_mod
 
ld.exe: C:\Users\igord\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.1\tools\sdk\esp32\lib\libmbedcrypto.a(dhm.c.obj): in function dhm_make_common
dhm.c*:235: undefined reference to __wrap_mbedtls_mpi_exp_mod
 
ld.exe: C:\Users\igord\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.1\tools\sdk\esp32\lib\libmbedcrypto.a(dhm.c.obj): in function mbedtls_dhm_calc_secret
dhm.c*:465: undefined reference to __wrap_mbedtls_mpi_exp_mod
ld.exe: C:\Users\igord\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.1\tools\sdk\esp32\lib\libmbedcrypto.a(dhm.c.obj):\home\test\esp32-arduino-lib-builder\build\..\esp-idf\components\mbedtls\mbedtls\library\dhm.c:425: more undefined references to __wrap_mbedtls_mpi_exp_mod follow
 
collect2.exe*: error: ld returned 1 exit status
```