espressif/arduino-esp32 #6293: OTA from https url broken in 2.0.2 ?

igor-d-n, 2022-02-15:
### Board

ESP32

### Device Description

any

### Hardware Configuration

empty dev pcb

### Version

v2.0.2

### IDE Name

Visual Studio and vMicro plugin

### Operating System

Windows 10

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

512000

### Description

In version 2.0.2 OTA update from https source is broken? Tried many variants and example. Then added verbose loggin.
Strange that in 2.0.1 it worked good. Then i switched to 2.0.2 and it's not working now.

### Sketch

```cpp
....
WiFiClientSecure client;

			debugLog("OTA", "https start");
			client.setCACert(rootCACertificate);
			//client.setInsecure(); // чтобы не требовало отпечатка или CA серта
			//debugLog("OTA", "client connect %d", client.connect("myhost.xx", 443));
			
			char buf[200];
			int errCode = client.lastError(buf, 200);
			if (errCode) {
				debugLog("OTA", "client err. code=%d msg=%s", errCode, buf);
			}
			
			ret = httpUpdate.update(client, this->updateURL());
....
```

### Debug Message

```plain
[ 30844][V][HTTPClient.cpp:247] beginInternal(): url: https://myhost.xx/fwlatest
[ 30857][D][HTTPClient.cpp:298] beginInternal(): protocol: https, host: myhost.xx port: 443 url: /fwlatest
[ 33074][D][HTTPClient.cpp:595] sendRequest(): request type: 'GET' redirCount: 0

[ 33074][V][ssl_client.cpp:61] start_ssl_client(): Free internal heap before TLS 149840
[ 33078][V][ssl_client.cpp:67] start_ssl_client(): Starting socket
[ 35795][V][ssl_client.cpp:143] start_ssl_client(): Seeding the random number generator
[ 35796][V][ssl_client.cpp:152] start_ssl_client(): Setting up the SSL/TLS structure...
[ 35799][V][ssl_client.cpp:175] start_ssl_client(): Loading CA cert
[ 35810][V][ssl_client.cpp:244] start_ssl_client(): Setting hostname for TLS session...
[ 35813][V][ssl_client.cpp:259] start_ssl_client(): Performing the SSL/TLS handshake...
[ 36509][V][ssl_client.cpp:280] start_ssl_client(): Verifying peer X.509 certificate...
[ 36510][V][ssl_client.cpp:288] start_ssl_client(): Certificate verified.
[ 36513][V][ssl_client.cpp:303] start_ssl_client(): Free internal heap after TLS 122780
[ 36520][D][HTTPClient.cpp:1141] connect():  connected to myhost.xx:443
[ 36528][E][ssl_client.cpp:339] data_to_read(): RET: -26880
[ 36533][E][ssl_client.cpp:339] data_to_read(): RET: -26880
[ 36538][E][ssl_client.cpp:339] data_to_read(): RET: -26880
[ 36544][E][ssl_client.cpp:339] data_to_read(): RET: -26880
[ 36549][V][ssl_client.cpp:351] send_ssl_data(): Writing HTTP request with 585 bytes...
[ 36559][E][ssl_client.cpp:339] data_to_read(): RET: -26880
[ 36562][E][ssl_client.cpp:339] data_to_read(): RET: -26880
[ 36567][E][ssl_client.cpp:339] data_to_read(): RET: -26880
[ 36573][E][ssl_client.cpp:339] data_to_read(): RET: -26880
[ 36578][E][ssl_client.cpp:339] data_to_read(): RET: -26880
[ 36593][E][ssl_client.cpp:339] data_to_read(): RET: -26880
[ 36593][E][ssl_client.cpp:339] data_to_read(): RET: -26880
[ 36594][E][ssl_client.cpp:339] data_to_read(): RET: -26880
[ 36609][E][ssl_client.cpp:339] data_to_read(): RET: -26880
[ 36609][E][ssl_client.cpp:339] data_to_read(): RET: -26880
[ 36610][E][ssl_client.cpp:339] data_to_read(): RET: -26880
[ 36625][E][ssl_client.cpp:339] data_to_read(): RET: -28928
[ 36625][E][ssl_client.cpp:36] _handle_error(): [data_to_read():343]: (-28928) SSL - Bad input parameters to function
[ 36632][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
[ 36638][D][HTTPClient.cpp:619] sendRequest(): sendRequest code=-5

[ 36643][W][HTTPClient.cpp:1437] returnError(): error(-5): connection lost
[ 36650][E][HTTPUpdate.cpp:231] handleUpdate(): HTTP error: connection lost
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

igor-d-n, 2022-02-17:
May be it's now an error. Find out that I can't set MFL for https connection in OTA. It must work "out of a box" or it can be set somehow?

VojtechBartoska, 2022-04-11:
Hello, can you please retest this on v2.0.3-rc1?

DmytroKorniienko, 2022-05-04:
@VojtechBartoska Same issue was with 2.0.2 for me, but successfully have been resolved with 2.0.3 (last release)
19:05:52.961 > cam_hal: EV-EOF-OVF
19:05:53.108 > cam_hal: EV-EOF-OVF
19:05:53.338 > cam_hal: EV-EOF-OVF
19:05:53.376 > cam_hal: EV-EOF-OVF
19:05:53.435 > Update Success: 1165408B

Thank you.

VojtechBartoska, 2022-05-04:
thanks for your testing @DmytroKorniienko! Closing as solved, if needed please reopen the issue.

closed 2022-05-04 by VojtechBartoska